springfall2008/batpred #1266: Battery reaches 4% by 21:30 each night

78davidcross, 2024-06-28:
**Describe the bug**
Info as requested by Trefor on Facebook Messenger.

Plan should be discharging excess battery to approx. 4% by\around 23:30 but its reaching 4% consistently by approx. 21:30 each evening.

**Expected behavior**
Battery should be supplying house load through to 23:30 with any excess left over exported by 23:30 with the battery sitting at approx. 4% remaining.

**Predbat version**

8.2.0 on Main

**Environment details**

- Inverter and battery setup: Gen 2 5Kw GivE, 1qty 9.5 Gen 2 Battery
- Standard HAOS installer or Docker: Docker
- Anything else?  On IOG so standard charge window starts at 23:30

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
[image]
[image]
[image]
[appdaemon1.log](https://github.com/user-attachments/files/16036809/appdaemon1.log)
[appdaemon2.log](https://github.com/user-attachments/files/16036810/appdaemon2.log)
[appdaemon3.log](https://github.com/user-attachments/files/16036811/appdaemon3.log)
[appdaemon.log](https://github.com/user-attachments/files/16036812/appdaemon.log)
[PredBat Saved Settings YAML.txt](https://github.com/user-attachments/files/16036815/PredBat.Saved.Settings.YAML.txt)

springfall2008, 2024-06-30:
Can you re-test on v8.2.1 release please as 'main' was a bit unstable for a while?

78davidcross, 2024-06-30:
Have been on 8.2.1 since around 12:30. Battery is now at 15% and is unlikely to last. I've stopped discharge. Battery temp was reported at 80 degrees.
[image]

78davidcross, 2024-07-01:
Hello. Am now running 8.2.1.  Today's plan is attached.

I'm not convinced it's going to achieve 4% at 23:30. It continues to discharge and I'm confident it's going to be empty long before 23:30.
[image]

gcoan, 2024-07-01:
Have you got combine_discharge_slots on?

Suggest you turn it off. 

If predbat is discharging too much you can use select.predbat_manual_idle to force it to idle eco mode at any slots you want, but it does look like it's going to discharge too far

78davidcross, 2024-07-01:
Combine discharge slots has always been off.

At 18:00 it started discharging again. Now at 17% which isn't going to cut it. I've set it Control Charge now. I can't trust it to discharge correctly.

I know I can set manual discharges but the point of Predbat is that I shouldn't need to be doing that daily and I am having to check on it every few hours.

I'll wait for Trefor to have some time to look at it but for now discharge is turned off.

gcoan, 2024-07-01:
There is an 8.2.2 which includes some more bug fixes from Trefor, might be worth trying that.

I don't see any similar bulk discharging going on with my own setup.  Would probably help Trefor if you can post the logfile corresponding to the above

78davidcross, 2024-07-01:
Yeah I saw 8.2.2 yesterday and am running it.  The version is now shown at the top of the charge plan too.  

By 19:30 today the battery was empty.  Log attached for this afternoon.

I've been chatting to Trefor separately so he's aware that there's something odd about what's happening here.

Thanks for trying to assist.

I'll keep in Charge only for now and will manually export whatever is left late in the evening in the coming days,
[appdaemon.log](https://github.com/user-attachments/files/16058145/appdaemon.log)

78davidcross, 2024-07-02:
Might be relevant, might not, but I've noticed the High Rate Slot doesn't turn off.  HA history shows its been On for weeks\months.  Is that normal?

The Low Rate Slot goes on and off as expected.

gcoan, 2024-07-02:
My high rate slot has only turned on and off a couple of times in June, once for a day and once for 10 minutes.  But like you I'm on a fixed 15p export tariff so maybe expect it to be constant as there's no 'low rate' export periods

78davidcross, 2024-07-10:
Hello,

Am running 8.2.4 now but the issue remains.

Predbat is discharging the battery at full rate yet the plan thinks the SOC will only fall by 1 or 2% it seems across an hour of discharge whereas it falls a lot lot more than that.

What's the relationship between the Limit column and the SOC % column?  Limit shows 58% in the included screenshot and SOC % at 23:30 is 60%. But why? The battery hits the 58% limit by 18:52 or thereabouts and then that limit is lowered and discharge continues yet the SOC % doesn't change.

I've been trailing Willy Reckitt's IOG Full Discharge automation to see if that handles the discharge to 4% by 23:30 whilst maintaining enough power in the battery to cover the house load in the evening and it works perfectly.

For now I've been allowing Predbat to manage daytime operations with a Best SOC Min of 7 to prevent it discharging the entire battery and us using the grid in the evening. Then late afternoon Predbat is put into Monitor mode and the IOG Full Discharge from Willy takes over.
[image]

gcoan, 2024-07-10:
> What's the relationship between the Limit column and the SOC % column? Limit shows 58% in the included screenshot and SOC % at 23:30 is 60%. But why? The battery hits the 58% limit by 18:52 or thereabouts and then that limit is lowered and discharge continues yet the SOC % doesn't change.

The limit is the level that Predbat is discharging the battery to, and the SoC is the expected charge % that the battery will be at the end of each slot.

Your problem is that the battery is discharging faster than Predbat thinks it is,  I had a look at your predbat settings and I think the problem is:

```
- entity: input_number.predbat_battery_rate_max_scaling_discharge
  friendly_name: Battery rate max scaling discharge
  icon: mdi:multiplication
  max: 1.0
  min: 0
  name: battery_rate_max_scaling_discharge
  step: 0.01
  type: input_number
  unit: '*'
  value: 0.05
```

From the documentation https://springfall2008.github.io/batpred/customisation/#scaling-and-weight-options

"input_number.predbat_battery_rate_max_scaling_discharge is a percentage factor to adjust your maximum discharge rate from that reported by the inverter. For GE inverters the discharge rate is reported as the max AC rate and thus is fairly accurate. You can try computing your discharge curve and check recommendations for changing this figure in the logfile."

78davidcross, 2024-07-10:
The value in the entity is what is reported in the log, or leave on Auto which is what it's set to.

gcoan, 2024-07-10:
> The value in the entity is what is reported in the log, or leave on Auto which is what it's set to.

The value in the entity is 0.05,  so this is telling Predbat that in the plan it should scale the inverter's reported discharge rate to 5% of what the inverter says it is.

Normally this is set to 1, i.e. treat what then inverter says is the discharge rates as being correct.  You might change it to 0.99 or 0.98 to make a minor adjustment to the reported discharge rate, but setting it to 5% is why the battery is discharging much faster than Predbat thinks it should be.

Set it back to the default 1

78davidcross, 2024-07-15:
I set input_number.predbat_battery_rate_max_scaling_discharge to 1 as suggested on Saturday and that seems to have fixed the issue. Predbat now discharges correctly with roughly 4% left by 23:30.

Thanks for looking at this, issue appears resolved now.

closed 2024-07-15 by 78davidcross