springfall2008/batpred #1273: Updated from 8.1.1 to 8.2.1 - but HA says update is still available

PianSom, 2024-06-30:
Not a big issue this one ...

HA Container user. 

Updated using ui on HA yesterday evening from 8.1.1 to 8.2.1. Update appears to work ok, and correct version is reported eg on Predbat plan page

But HA gives a Settings page flag that a Predbat update is available, and reports I am still on 8.1.1

Have tried a HA restart.

PianSom, 2024-06-30:
Updated to 8.2.2. Issue persists

SwiftRR, 2024-06-30:
I updated directly from 8.11 to 8.22 this evening and the update is being shown in status in HA. So no issue for me here.
[image]

Rob

PianSom, 2024-07-01:
Issue resolved by appdaemon container restart

PianSom, 2024-07-03:
Same problem when updating to 8.2.3. Same solution - restart docker appdaemon container

springfall2008, 2024-07-03:
I suspect its just not updating its state quickly enough

PianSom, 2024-07-04:
I just updated to 8.2.4 with the same issues. Here are the log entries showing from the AD container for the last few days.

<details><summary>log</summary>
<p>

2024-07-02T20:36:29.605090945Z Update complete
2024-07-02T20:37:42.926240241Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v7.22.5/apps/predbat/predbat.py
2024-07-02T20:37:42.926262779Z Got data, writing to /conf/apps/predbat/predbat.py.v7.22.5
2024-07-03T05:46:33.591046023Z Update complete
2024-07-03T05:47:46.351298089Z Predbat files are installed correctly for version v8.2.3
2024-07-03T07:36:55.783469149Z Predbat files are installed correctly for version v8.2.3
2024-07-04T10:19:30.031898960Z Update complete
2024-07-04T10:20:21.506125739Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/predbat.py
2024-07-04T10:20:21.506158394Z Got data, writing to /conf/apps/predbat/predbat.py.v8.2.4
2024-07-04T10:20:21.506164224Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/config.py
2024-07-04T10:20:21.506168610Z Got data, writing to /conf/apps/predbat/config.py.v8.2.4
2024-07-04T10:20:21.506172773Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/prediction.py
2024-07-04T10:20:21.506176768Z Got data, writing to /conf/apps/predbat/prediction.py.v8.2.4
2024-07-04T10:20:21.506181473Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/utils.py
2024-07-04T10:20:21.506185948Z Got data, writing to /conf/apps/predbat/utils.py.v8.2.4
2024-07-04T10:20:21.506190030Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/inverter.py
2024-07-04T10:20:21.506193920Z Got data, writing to /conf/apps/predbat/inverter.py.v8.2.4
2024-07-04T10:20:21.506197650Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/ha.py
2024-07-04T10:20:21.506201473Z Got data, writing to /conf/apps/predbat/ha.py.v8.2.4
2024-07-04T10:20:21.506205436Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/download.py
2024-07-04T10:20:21.506209527Z Got data, writing to /conf/apps/predbat/download.py.v8.2.4
2024-07-04T10:20:21.506227174Z Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.2.4/apps/predbat/unit_test.py
2024-07-04T10:20:21.506231216Z Got data, writing to /conf/apps/predbat/unit_test.py.v8.2.4
2024-07-04T10:20:21.506234593Z Predbat files are installed correctly for version v8.2.4
2024-07-04T10:21:17.835328725Z Predbat files are installed correctly for version v8.2.4

      

</p>
</details>

SwiftRR, 2024-07-04:
I am running standalone predbat add-on. HAOS on a raspberry pi4 using a SSD
Core 2024.6.4
Supervisor 2024.06.2

I tried installing predbat v8.2.4 and it took ages and crashed HA.
I managed to get back on after a few minutes. My status reports look fine with no errors.
In my Predbat add-on log, I see the extract below.
So not a smooth upgraded for me this time.

Notice line 8:
`2024-07-04 13:12:52.407778: Warn: unable to find /config/appdaemon.yaml skipping checks as maybe outside AppDaemon`
I presume that this is so that common code can be used between AppDaemon and standalone varieties of predbat.

Also notice the historical day warnings that are repeated every 5 minutes with minor differences between entries. They won't go away. 
```
Predbat files are installed correctly for version v8.2.4
**** Starting Standalone Predbat ****
2024-07-04 13:12:52.201207: Loading apps.yaml
2024-07-04 13:12:52.402903: Info: Connected to Home Assistant at http://supervisor/core
2024-07-04 13:12:52.403095: Creating task: <coroutine object HAInterface.socketLoop at 0x7f83782f10>
2024-07-04 13:12:52.404026: Info: Web Socket task started
2024-07-04 13:12:52.405141: Info: Start socket for url http://supervisor/core/api/websocket
2024-07-04 13:12:52.407778: Warn: unable to find /config/appdaemon.yaml skipping checks as maybe outside AppDaemon
2024-07-04 13:12:52.442576: Info: Web Socket active
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
2024-07-04 13:13:45.962316: Warn: Historical day 2 has 35 minutes of gap in the data, filled from 9.01 kWh to make new average 9.23 kWh (percent 98%)
2024-07-04 13:13:45.970524: Warn: Historical day 4 has 30 minutes of gap in the data, filled from 8.0 kWh to make new average 8.17 kWh (percent 98%)
2024-07-04 13:13:45.986162: Warn: Historical day 5 has 90 minutes of gap in the data, filled from 7.6 kWh to make new average 8.11 kWh (percent 94%)
2024-07-04 13:13:46.005109: Warn: Historical day 6 has 100 minutes of gap in the data, filled from 6.2 kWh to make new average 6.66 kWh (percent 93%)
2024-07-04 13:13:46.037910: Warn: Historical day 7 has 285 minutes of gap in the data, filled from 7.6 kWh to make new average 9.48 kWh (percent 80%)
2024-07-04 13:16:22.619310: Info: record_status Idle

```

screig, 2024-07-04:
I get this error

`Error: Exception raised unsupported operand type(s) for +=:`

[image]

gcoan, 2024-07-04:
> I get this error
> 
> 
> 
> `Error: Exception raised unsupported operand type(s) for +=:`
> 

Is this an upgrade, a new installation?

If its an upgrade that fails after upgrading, try restarting the Appdaemon or Predbat add-on and that may well clear the error

springfall2008, 2024-07-13:
I think this is now fixed, let me know if you have any new issues

closed 2024-07-13 by springfall2008

cbrinks17, 2024-08-16:
Standalone HA on Home Assistant Green. No containers or anything complicated. I am still getting this with every update. Restarting AppDemon clears the update notification, but that is somewhat tedious to do every time.