springfall2008/batpred #1279: Unable to update to anything later than v8.0.0

PhilPJL, 2024-07-02:
**Bug**

I have a notification to update from v8.0.0 to v8.2.3 (this has been happening since v8.0.0)

Press install.  The installation appears to proceed but when it finishes it still says v8.0.0 installed and 'update available'.  Pressing install again has no effect.

The Status value for Predbat now displays exception details.  See 3rd screenshot.

I can downgrade Predbat back to v8.0.0 to get rid of the exception.

**Expected behavior**

To install successfully.

**Predbat version**

Predbat Core v8.0.0
Predbat v8.2.3?

**Environment details**

GivEnergy 5kW inverter + 9kWh battery.

Standard HAOS.

**Screenshots**

[image]
[image]
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-07-02:
There's been a few people that have had this fail to work after upgrading past v8, myself included.

Restart Appdaemon/Predbat add-on as appropriate after upgrading and it should clear the error and predbat works OK

PhilPJL, 2024-07-03:
Thanks, I forgot about the restart vs reboot.  Reboot + restart appdaemon seems to have fixed things.

gcoan, 2024-07-03:
Glad you are sorted

I don't know why these v8 upgrades are having these behaviours requiring an add-on restart.  Prior versions have upgraded seamlesly.