springfall2008/batpred #1282: discharge-stop service feature request

MalcolmSpencer, 2024-07-04:
**Is your feature request related to a problem? Yes
I can't use the service API integration to control the discharge start and stop with the Myenergi Libbi inverter.
This is because the Libbi inverter has a separate service for charge and discharge

Charge - select.myenergi_libbi_24042899_charge_from_grid - True/False
Discharge - select.myenergi_libbi_24042899_operating_mode - Normal/Export

But Predbat currently uses the charge stop service to also stop the discharge as per the code at 

https://github.com/springfall2008/batpred/blob/93bc6ac998940430516b8d318c16e6f5a62311de/apps/predbat/inverter.py#L1836C1-L1846C80

**Describe the solution you'd like**
Add a discharge-stop service

  discharge_stop_service:
    service: select.select_option
    entity_id: "select.myenergi_libbi_24042899_operating_mode"
    option: "Normal"

Modify the inverter code to check if the discharge stop service is defined in the apps.yaml file
If discharge stop is defined use that if it is not defined then use charge stop as today.
This will ensure existing configs aren't broken. 

**Describe alternatives you've considered**
Create an automation outside of predbat to control the discharge

**Additional context**
Add any other context or screenshots about the feature request here.

springfall2008, 2024-07-06:
Is it okay to call the discharge stop service even if the inverter isn't discharging? If so this is easy to add, but if need to happen only when discharge is active that is harder as I don't really track state for this.

MalcolmSpencer, 2024-07-06:
Hi,
Yes its fine to call the discharge stop service if the inverter isn't discharging.

springfall2008, 2024-07-14:
Fix on main for testing

springfall2008, 2024-08-06:
This is done now

closed 2024-08-06 by springfall2008