springfall2008/batpred #1283: Clarification about switch.predbat_iboost_gas

mogons57, 2024-07-04:
**Is your feature request related to a problem? Please describe.**
It is not clear in the documentation when switch.predbat_iboost_gas is on what electricity rates are being referred to in 
"When enabled will control the solar diverter to only operate when electric rates are lower than gas rates."

**Describe the solution you'd like**
I'm trying to setup things so that when I have spare solar and my export rate is lower than the gas import rate my iboost (actually an EDDI but essentially the same) is triggered to start diverting power.

**Describe alternatives you've considered**
This does not seem to be working (and I've played with the gas price weighting to try to force this to work) so I'm wondering if the price being referred to is the electricity import rate. 

**Additional context**
I can see arguments for using either the electricity import rate or the export rate to trigger iboost diverting (depending on what you are trying to do) so might an extra switch be included to decide which rate the gas is being compared with?

springfall2008, 2024-07-08:
I've re-worked the iboost now, new version on 'main' for testing which includes export rate filtering and export rate higher than gas option

mogons57, 2024-07-08:
Just had a look at the new code (and not sure I understand all of it) but If I've not misread things there is only one sensor to use to drive the iboost settings i.e. **binary_sensor.predbat_iboost_active**.  

My current automation to change the control mode of the iboost/eddi [off or divert] is:

alias: Solar Diverter (Eddi)
description: Start/stop solar diverter based on Predbat determined slots
trigger:
  - platform: state
    entity_id:
      - binary_sensor.predbat_iboost_active
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: binary_sensor.predbat_iboost_active
            state: "True"
        sequence:
          - service: select.select_option
            metadata: {}
            data:
              option: Normal
            target:
              entity_id: select.myenergi_eddi_#####_operating_mode
      - conditions:
          - condition: state
            entity_id: binary_sensor.predbat_iboost_active
            state: "False"
        sequence:
          - service: select.select_option
            data:
              option: Stopped
            target:
              entity_id: select.myenergi_eddi_#####_operating_mode
mode: single

but is there another sensor I can use to tell me whether I should be setting divert (just use excess solar) or boost (heat at full rate) mode i.e. control to iboost to [off, divert, boost]?

springfall2008, 2024-07-13:
Good point, I probably should add that.

springfall2008, 2024-07-13:
Adding on 'main' attributes 'solar' and 'full' to the iboost_active sensor for this purpose

closed 2024-07-13 by springfall2008