springfall2008/batpred #1284: Not calculating best plan after 8.1.0

Philhews, 2024-07-05:
**Describe the bug**
Tomorrows agile rates are negative in the afternoon, and the solar forecast is not enough to max out my inverter so freeze discharging before and using some negative grid to charge later is the best cost. This is the plan that is selected on 8.1.0

However when upgrading past that version (anything up to latest) the plan no longer utilises the AC negative rates and just does solar charge with a demonstrably worse 'best' cost value.

**Expected behavior**
Best plan should be selected on newer versions

**Predbat version**
8.1.0

**Environment details**

Hybrid 3.6 Gen2, 9.5kWh battery
RPI4 with HAOS and standalone predbat

**Screenshots**
Plan on 8.1.0
[image]
Cost on 8.1.0
[image]
Battery graph on 8.1.0
[image]

Plan on 8.1.1
[image]
Cost on 8.1.1
[image]
Battery graph on 8.1.1
[image]

**Log file**
Update from 8.1.0 to 8.1.1 happened at 20:53 so log from before then is working as I would expect, and immediately after is the problem
[predbat.log](https://github.com/user-attachments/files/16113382/predbat.log)

Philhews, 2024-07-06:
I'll see how that affects the current plans, obviously the rates are different now so it's not a direct comparison.

Did something change between the two versions mentioned that would make these settings more/less important in the calculation to result in the differences I was seeing?

Philhews, 2024-07-06:
8.1.1 still tending towards less charging, requires more solar to refill the battery rather than using grid at 7 or 9p

8.1.0 plan
[image]

8.1.1 plan
[image]

Log file, update to .1 at 23.02
[predbat.log](https://github.com/user-attachments/files/16117499/predbat.log)

springfall2008, 2024-07-13:
> I'll see how that affects the current plans, obviously the rates are different now so it's not a direct comparison.
> 
> Did something change between the two versions mentioned that would make these settings more/less important in the calculation to result in the differences I was seeing?

Yes, there were bugs in the code which mean it wasn't always applying losses correctly.

closed 2025-01-01 by springfall2008