springfall2008/batpred #1285: HIgh and low thresholds not being calculated properly

mogons57, 2024-07-06:
**Describe the bug**
If the "high rate threshold" and "low rate threshold" are set to zero (which the documentation says are the defaults) then the two thresholds, shown on the energy rates plot, become the same for me at 14p (I'm on Octopus flux). When both thresholds are set to 1, I still have the same problem.

Testing a few other values for these parameters....
setting high set to 1.05, and low to 0.95 gives me thresholds of high 21.2p and low 14.4p - seems OK
setting high set to 1.50, and low to 0.50 gives me thresholds of high 22.1p and low 11.2p - seems OK
setting high set to 1.95, and low to 0.05 gives me thresholds of high 28.8p and low  2.2p - seems OK
setting high set to 1.05, and low to 1.00 gives me thresholds of high 21.2p and low 14.4p - seems OK
setting high set to 1.20, and low to 1.00 gives me thresholds of high 22.4p and low 14.4p - seems OK

However
setting high set to 1.00, and low to 0.95 gives me thresholds of high 14.0p and low 14.0p - Not right
setting high set to 1.00, and low to 0.85 gives me thresholds of high 14.0p and low 14.0p - Not right
setting high set to 1.00, and low to 0.00 gives me thresholds of high 14.0p and low 14.0p - Not right
setting high set to 0.00, and low to 0.95 gives me thresholds of high 14.0p and low 14.0p - Not right

It looks like setting a high value of 0 or 1 is causing the calculation not to work.

The thresholds had been working fine but they might have gone odd when the prices changed recently, or possible when a predbat update was done.

**Expected behavior**
Values of calculated thresholds that are not the same.

**Predbat version**
8.2.4

**Environment details**
- Givenergy AC3 and 9.5 battery
- Standard HAOS install

**Screenshots**
[image]

springfall2008, 2024-07-06:
Why do you say the values are wrong? The thresholds are import below and export above thresholds?

mogons57, 2024-07-07:
It was the calculation of the thresholds that did not seem to work properly for me if the "rate high threshold" was set to either 0 or 1. However I tried to replicate the problem again today and things seem to be working fine now with the calculated thresholds looking right for any rate values I set. Something must have got confused in my system, but as things are now working I guess case closed until/if it happens again.

closed 2024-07-07 by mogons57