springfall2008/batpred #1291: Predicted load changing throughout the day.

Philhews, 2024-07-07:
**Describe the bug**
The predicted load for today changes throughout the course of the day

**Expected behavior**
Predicted load is based on historical data, given that data isn't changing, the predicted load for today shouldn't change. 

**Predbat version**

Any

**Screenshots**
[image]

gcoan, 2024-07-07:
I think you are misunderstanding the sensor.

All of the Predbat sensors are 'today's value up to end of plan' so as the day goes on the plan will roll forward and the sensor will include the additional time period.

It's not a 'today's projection' value.

Also, if you have switch.predbat_calculate_inday_adjustment turned on https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options Predbat will adjust your projected load based on your current load usage today - i.e. if you have used a lot of energy already today it will project you are going to use more. Useful if you have different patterns such as weekends or electric heating (ASHP)

Philhews, 2024-07-07:
I'm not sure your explanation has cleared it up for me, if it's not projecting today what is it projecting? 
This graph has a 'predicted' value, this is what then links to the previous sensor I posted. If it's not a predicted usage value for today then I'd suggest it shouldn't be on an example graph named predicted and graphed across today!
[image]

gcoan, 2024-07-07:
Sorry @Philhews this is a 'today' sensor, my mistake.

Looking at it on my own HA, and looking at the code, there is something weird going on, both for this and the load energy actual, they start off at a high value and then decrease as the day goes on:
[image]

Although the attributes of the forecast and actual over the day in 5 minute slots look OK, the sensor value I can't work out what it contains.

And I also noticed that neither of these are described in the output data in the docs https://springfall2008.github.io/batpred/output-data/

Philhews, 2024-07-07:
Load energy actual seems to start the day as what it thinks you'll be on at the end of the day, then as you go through the day it becomes your actual so far plus the predicted for the remainder of the day, it filters out ac charge periods and maybe car stuff but I think the actual makes sense to me, at least I understand why it would change (up or down).

gcoan, 2024-07-08:
Yes I could see that sort of logic in the code, but (notwithstanding its not described in the documentation), but I don't quite follow why the initial (midnight) value is double my daily load

springfall2008, 2024-07-13:
I think its dropping if you used less than you predicted and you have days_previous set to (or including) 1 which means as your actual changes the predictions change?

Philhews, 2024-07-13:
I've got days previous as 2-8, not using 1 at all.

Philhews, 2024-07-31:
So throughout the day the log reports changing historical load values, which shouldn't be possible running for days 2-8 previous, but would explain why the original predicted value for the day changes

2024-07-31 06:00:03.543331: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.82, 8.51, 11.92, 6.95, 12.13, 9.07, 8.77] - min 6.82
2024-07-31 08:00:03.571728: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.66, 8.08, 11.69, 7.83, 11.54, 9.21, 9.06] - min 6.66
2024-07-31 10:00:23.531567: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.64, 8.36, 9.9, 9.35, 11.42, 8.82, 9.39] - min 6.64
2024-07-31 12:00:03.842195: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.71, 8.47, 7.92, 11.0, 11.38, 9.09, 9.56] - min 6.71
2024-07-31 14:00:04.367878: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.92, 8.48, 7.7, 11.11, 10.51, 9.97, 9.34] - min 6.92
2024-07-31 14:55:45.879399: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.84, 8.57, 7.58, 11.2, 9.55, 11.04, 9.05] - min 6.84
2024-07-31 16:00:03.953726: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [6.84, 8.01, 8.03, 11.23, 8.61, 11.41, 9.5] - min 6.84
2024-07-31 18:00:04.192843: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [7.23, 6.72, 8.54, 11.09, 8.59, 11.16, 10.27] - min 6.72

Philhews, 2024-07-31:
Is it not actually doing midnight to midnight? So 2 days previous isn't just all of Monday when we're anywhere on Wednesday but actually from 48 hours ago to 24hours ago? (or 72 to 48, but you get the idea).

gcoan, 2024-10-23:
@Philhews I have added documentation to explain how the sensor works and what it contains

closed 2025-01-01 by springfall2008