springfall2008/batpred #1292: Predbat Planning broken apparently due to inverter/Appdaemon clocks discrepancy

jrwm47, 2024-07-07:
Predbat planning and control appears to have halted due to a huge and increasing reported discrepancy between inverter and AppDaemon times. Appdaemon time is correct. Inverter time is way out. Do I have a hardware RTC problem? 

After several weeks of very careful setting up in Monitor mode,  a week ago I finally gave Predbat control of my GE AIO. Initially it appeared to be working extremely well. But today it has started showing an SOC (59%)  [see screenshot below] in the plan which is completely different from that reported by GivEnergy and WonderWatt (control switched off), both reporting (95%).

The log file [below] reports a 1563.5 minute inverter/Appdaemon time discrepancy and says:

2024-07-07 16:46:54.345839: Warn: Set inverter 0 mode Eco via REST failed
2024-07-07 16:46:54.389117: Info: record_status Warn: Inverter 0 REST failed to setBatteryMode. 

**Predbat version**
1.1.7. I have deliberately not upgraded to 1.1.8.

**Environment details**
GivEnergy 13.5kWh All-In-One _**without**_ Gateway; CTs hardwired to inverter by installer 
HAOS on RasPi 4 on very reliable LAN
3.6kW PV with Solis inverter
eddi and harvi
Octopus Agile

- Anything else?
Yes. Several things:

1) Ever since my AIO was installed, GE platform and app have reported completely crazy Energy numbers - *,000s of kWh per day! Power numbers however look ok. I have reported the problem, with screenshots, numerous times to GE support but they have gone silent on me.

2) I struggled for days to get the GivTCP/Predbat pipeline to work. Only when I switched to the GivTCP_Beta Add-on did GivTCP/predbat start reading the inverter serial number and predbat planning started.

3) I used your most recent integrated Add-on installation sequence, without an explicit Appdaemon installation. But Appdaemon clock appears to be running and checked by predbat.

4) I only enabled iBoost for our eddi yesterday and I suspect that was when the predbat planning problem started. Could this be relevant?   

**YAML**
# ------------------------------------------------------------------
# JM v1.0 1/7/24
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  
  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
[... 467 lines cut ...]
2024-07-07 17:16:46.758748: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-07-07 17:16:57.893760: Warn: Set inverter 0 mode Eco via REST failed
2024-07-07 17:16:57.938488: Info: record_status Warn: Inverter 0 REST failed to setBatteryMode
2024-07-07 17:20:05.156174: Warn: Historical day 2 has 330 minutes of gap in the data, filled from 24.33 kWh to make new average 31.56 kWh (percent 77%)
2024-07-07 17:20:05.172692: Warn: Historical day 3 has 220 minutes of gap in the data, filled from 14.86 kWh to make new average 17.54 kWh (percent 85%)
2024-07-07 17:20:06.582473: Warn: Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:20:06.582219+01:00 this is -1598.48 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:20:06.636865: Info: record_status Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:20:06.582219+01:00 this is -1598.48 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:20:06.637055: Warn: Inverter control auto restart trigger: Clock skew >=10 minutes
2024-07-07 17:20:06.637142: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-07-07 17:20:23.519813: Warn: Set inverter 0 mode Eco via REST failed
2024-07-07 17:20:23.560145: Info: record_status Warn: Inverter 0 REST failed to setBatteryMode
2024-07-07 17:25:04.750198: Warn: Historical day 2 has 335 minutes of gap in the data, filled from 24.28 kWh to make new average 31.64 kWh (percent 77%)
2024-07-07 17:25:04.764735: Warn: Historical day 3 has 220 minutes of gap in the data, filled from 14.91 kWh to make new average 17.6 kWh (percent 85%)
2024-07-07 17:25:06.413383: Warn: Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:25:06.413126+01:00 this is -1603.48 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:25:06.467376: Info: record_status Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:25:06.413126+01:00 this is -1603.48 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:25:06.467606: Warn: Inverter control auto restart trigger: Clock skew >=10 minutes
2024-07-07 17:25:06.467725: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-07-07 17:25:22.745612: Warn: Set inverter 0 mode Eco via REST failed
2024-07-07 17:25:22.791716: Info: record_status Warn: Inverter 0 REST failed to setBatteryMode
2024-07-07 17:26:44.156384: Warn: Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:26:44.156082+01:00 this is -1605.12 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:26:44.215271: Info: record_status Invertor time is 2024-07-06 13:41:38+00:00 AppDaemon time 2024-07-07 17:26:44.156082+01:00 this is -1605.12 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter or fixing AppDaemon time zone
2024-07-07 17:26:44.216535: Warn: Inverter control auto restart trigger: Clock skew >=10 minutes
2024-07-07 17:26:44.216696: Info: auto_restart not defined in apps.yaml, Predbat can't auto-restart inverter control
2024-07-07 17:26:55.130833: Warn: Set inverter 0 mode Eco via REST failed
2024-07-07 17:26:55.174285: Info: record_status Warn: Inverter 0 REST failed to setBatteryMode

gcoan, 2024-07-07:
The message 'appdaemon time' is misleading, it actually refers to the time of the computer that Predbat is running on.  I have fixed this in the next PR I will push through.

So what Predbat is reporting is that there is a mismatch between the clock on your Pi running Predbat and the time it is getting back from GivTCP.

So check firstly the clock on the Pi, if this is correct then it points to one of two issues:

1. The inverter time is wrong. You should be able to sync this from the portal.
2. GivTCP not being able to communicate reliably with the AIO.  It might be a physical/wifi network connection issue or it might be a bug in GivTCP with the AIO

jrwm47, 2024-07-08:
Thanks for your comments and advice.

1. I think the Pi4/HA time is correct as it agrees with all other clocks around here.
The inverter clock was more than 2 days behind and the discrepancy was rapidly increasing.

1.1 I tried resetting it through the GE portal (Time & Date - SEND) but it didn't appear to do anything.

1.2 So next I did a soft reset of the inverter (Restart - SEND) and then restarted GivTCP_Beta and predbat in Monitor mode. Initially everything seemed to be ok but the inverter clock time now appears to be completely stuck and predbat soon stopped updating the plan (SOC% hasn't changed since the restart).

It seems I either have a dead RTC in the inverter or something very screwy with its datetime registers. I don't look forward to persuading GE Support of the problem. I realise this isn't predbat issue but do you know the simplest way to get internal clock readings out of my GE AIO?

2.  The GivTCP problem could be comms but the CTs (ID1, ID2) are hard-wired and LAN mesh wifi signal at the inverter antenna is strong (-61dBm). I suspect a bug in the GivTCP or GE firmware. Again, do you know a simple way to get Energy values out of the AIO?

Cheers

closed 2025-01-01 by springfall2008