springfall2008/batpred #1293: Invalid value for number.givtcp_chXXX_charge_target_soc_2

deanjesper, 2024-07-07:
**Describe the bug**

Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:182
integration: MQTT (documentation, issues)
First occurred: 21:35:46 (64 occurrences)
Last logged: 21:45:32

Invalid value for number.givtcp_chXXX_charge_target_soc_2: 0 (range 4.0 - 100.0)

**Expected behavior**

Nor sure

**Predbat version**

8.2.4

**Environment details**

GivTCP and AIO

**Screenshots**

N/A

**Log file**

See above

gcoan, 2024-07-07:
@deanjesper can you add the logfile please and probably apps.yaml in case there is a config issue?

springfall2008, 2024-07-08:
Somehow you must have reserve not set to 4 or above, check 'battery_min_soc' is not set to 0 in apps.yaml

deanjesper, 2024-07-09:
> Somehow you must have reserve not set to 4 or above, check 'battery_min_soc' is not set to 0 in apps.yaml

Sorry haven't had a chance to look into this properly yet, I couldn't see battery_min_soc in my apps.yaml. I'll post it later. Best SOC Keep is set to 0.5 and Set Reserve Min is 4% on the entities. Interestingly if I change Set Reserve Min to 5% I see this error:

Logger: homeassistant.helpers.service
Source: helpers/service.py:301
First occurred: 08:12:11 (2 occurrences)
Last logged: 08:12:44

Referenced entities input_number.predbat_set_reserve_min are missing or not currently available
Referenced entities input_number.predbat_best_soc_keep are missing or not currently available

gcoan, 2024-07-09:
These are both input_number fields https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options and don't appear in apps.yaml

Very strange.  Think we need a broader look at the logfile and apps.yaml configuration

deanjesper, 2024-07-09:
I found the issue, for some reason the charge 2 percentage was set to 0 in the GivEnergy portal so I've set it to 100% and the error has gone!

[image]

gcoan, 2024-07-09:
Very strange.  Glad you sorted it out.

If its OK can we close this issue now?

closed 2024-07-14 by springfall2008