springfall2008/batpred #1303: No feed in scenario how to?

lucode, 2024-07-13:
I haven't installed Predbat yet because I did not find any information about the usage in a No-feed-in scenario. I'm not allowed to feed into the grid where I live only a few spikes are allowed. I only want to control the EV charging and the minimum Battery SOC to prevent unnecessary discharging the battery, if there is no chance to get sufficient solar power the next day.

If Predbat is usable for me I would like to see some information about that use case in the docs.
Another use case not mentioned in the docs is about fixed day and night tariffs. 

I really only get known about Predbat by change when the Gary goes solar Youtube video popped into my stream. I really did some extended search for a Home-Assistant module as Predbat but as it is nowhere mentioned on the HA forum I didn't get any findings about it on Google etc. search.  It would help if you consider opening up a community post in the HA Forum, otherwise, a lot of HA users miss that great product.  

Note: I don't have a Facebook account.

gcoan, 2024-07-13:
> I haven't installed Predbat yet because I did not find any information about the usage in a No-feed-in scenario. I'm not allowed to feed into the grid where I live only a few spikes are allowed. I [...]

Predbat should still be able to do what you want, to only control your battery charging. Just don't setup an export tariff and set select.predbat_mode to 'Control charge'

https://springfall2008.github.io/batpred/customisation/#predbat-mode

This will charge your battery when necessary but not do any forced discharge/export of the battery to the grid

> Another use case not mentioned in the docs is about fixed day and night tariffs.

Not sure what the problem is, there's comprehensive details of how to setup import (and export) tariffs in the documentation https://springfall2008.github.io/batpred/energy-rates/

Several options as to how to do this, but if you're not an Octopus user in the UK then you can just define the start time, end time and price of each import band in the rates_import settings in apps.yaml

gcoan, 2024-07-13:
I've added a clarification to the documentation about setting Predbat mode to 'Control charge' if you have a 'no export' restriction from your energy provider

lucode, 2024-07-17:
Cool, thank you I do consider to write a HA Forum post, how I got it running, after I have a bit more experience - If this in the sense of dev team? I still consider Predbat a geeky thing, so I'm not sure if it would be to early to get not so skilled people installing it.

closed 2024-07-17 by lucode