springfall2008/batpred #1304: SOC setpoint keeps fluctuating

tmh88, 2024-07-14:
**Describe the bug**
SOC setpoint keeps fluctuating instead of been constant

**Expected behavior**
A clear and concise description of what you expected to happen.

**Predbat version**

8.2.4

**Environment details**

- Sunsynk Inverter and Batteries
- Standard HAOS installer

**Screenshots**
[image]

**apps.yaml**
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.sunsynk_load_power_daily_2
  import_today:
    - sensor.smart_meter_electricity_import_today
  export_today:
    - sensor.smart_meter_electricity_total_export_daily
  pv_today:
    - sensor.sunsynk_total_pv_power_daily

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
[... 398 lines cut ...]
  #iboost_energy_scaling: 1.0
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'

tmh88, 2024-07-14:
[logs.txt](https://github.com/user-attachments/files/16216657/logs.txt)

tmh88, 2024-07-14:
[apps.txt](https://github.com/user-attachments/files/16216660/apps.txt)

springfall2008, 2024-07-14:
Okay I can see the issue here, the current reserve is reading as 1.6% when it should be 1.6kWh. Looking why...

springfall2008, 2024-07-14:
Your apps.yaml does not set 'reserve' but it sets 'battery_min_soc' to a value in kWh when it should be a value in %.

tmh88, 2024-07-14:
> Your apps.yaml does not set 'reserve' but it sets 'battery_min_soc' to a value in kWh when it should be a value in %.

Thats sorted it thanks

tmh88, 2024-07-15:
@springfall2008 

[image]

tmh88, 2024-07-15:
@springfall2008 
grid charging window finished at 8:30am

[logs 15-07-24.txt](https://github.com/user-attachments/files/16238540/logs.15-07-24.txt)

tmh88, 2024-07-15:
@springfall2008 
I have done some more testing tonight, i had made a few indents errors in my config which I have corrected.
I have noticed that the charge current doesn’t always update, even with force charging, discharge current seems fine. by toggling on/off switch.predbat_set_charge_freeze or switch.predbat_set_discharge_freeze_only. not sure if this is coincidence or not. haven’t been able to reproduce the fault but currently is giving me a charge current setpoint so see how it goes overnight.

Config and logs below
[apps.txt](https://github.com/user-attachments/files/16240990/apps.txt)

[logs after config corrections 15-07-24.txt](https://github.com/user-attachments/files/16240992/logs.after.config.corrections.15-07-24.txt)

closed 2025-01-01 by springfall2008