springfall2008/batpred #1310: Installed 8.3.0 from HA update option & install zapped apps.yaml

gcoan, 2024-07-15:
**Describe the bug**
I was on Predbat 8.2.4, normally I use the predbat select.update control to upgrade but as I was getting a nagging error that HACS thought I was on Predbat 7.x.x I used the HA update option to upgrade to 8.3.0

Got the error about template apps.yaml, restarted Appdaemon but error didn't clear.

On investigation I have found that the entire batpred directory has been replaced by the upgrade process, removing the backup apps.yaml file I had and also replacing my apps.yaml file with the standard template one.

**Expected behavior**
Clicking the upgrade option in HA should behave just the same as upgrading via the select.update control, only the predbat code is changed, anything else in the directory and especially the apps.yaml file is not touched.

**Predbat version**
8.2.4 going to 8.3.0

**Environment details**
HAOS 12.4, HA 2024.7.2

**Screenshots**
Directory contents afterwards:
[image]

config subdirectory, can see that apps.yaml was replaced at the same time as the other .py files:
[image]

Logfile:
[appdaemon.log](https://github.com/user-attachments/files/16239632/appdaemon.log)

Nothing relevant in the HA core log so not included

springfall2008, 2024-07-20:
I find HACS is buggy, if you install a second time the old one is restored (its actually renamed). Not sure I can find this I'm afraid, I'd suggest not using HACS

gcoan, 2024-07-20:
@springfall2008 I wasn't using HACS to upgrade (although have never had a problem with it upgrading before), I was using the HA upgrade option on the Settings menu that Predbat had published, saying that the upgrade was available.

I don't know what it did with the existing batpred folder within /config/appdaemon, it looked like it completely deleted it before installing a standard template one. As if it re-ran the vanilla install again rather than an upgrade?

springfall2008, 2024-08-05:
This is odd, its normally only if apps.yaml is missing it downloads a new one.

gcoan, 2024-08-10:
I'll close this and re-open if it occurs again

closed 2024-08-10 by gcoan