springfall2008/batpred #1311: Exception raised 'Prediction' object has no attribute 'iboost_running_solar'

Jorthax, 2024-07-16:
**Describe the bug**
Since installing 8.3.0, my system has reported this error message only - Exception raised 'Prediction' object has no attribute 'iboost_running_solar'

**Expected behavior**
Status should be "Idle" or other normal status instead of this error.

**Predbat version**
Patched from 8.2.4 to 8.3.0

**Environment details**

Givenergy setup with NO iBoost
Predbat pre-install in AppDaemon. Current version: 1.0.11

**Screenshots**
N/A

**Log file**
Predbat files are installed correctly for version v8.3.0
2024-07-16 10:41:08.022969 WARNING pred_bat: ------------------------------------------------------------
2024-07-16 10:41:08.023522 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-07-16 10:41:08.024172 WARNING pred_bat: Worker Ags: {'id': 'aa2c644bac194713b7be5bd89f579705', 'name': 'pred_bat', 'objectid': 'c35cc65a9ebb4dff9523badac3a6cc10', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7fb620ded0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-07-16 10:41:08.024863 WARNING pred_bat: ------------------------------------------------------------
2024-07-16 10:41:08.028638 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/config/apps/predbat.py", line 10899, in update_time_loop
    raise e
  File "/config/apps/predbat.py", line 10893, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/apps/predbat.py", line 9788, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 7959, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/predbat.py", line 7755, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 2105, in run_prediction
    self.iboost_running_solar = pred.iboost_running_solar
                                ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Prediction' object has no attribute 'iboost_running_solar'

2024-07-16 10:41:08.029251 WARNING pred_bat: ------------------------------------------------------------
2024-07-16 10:41:19.108575 WARNING pred_bat: ------------------------------------------------------------
2024-07-16 10:41:19.108954 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2024-07-16 10:41:19.109515 WARNING pred_bat: Worker Ags: {'id': '715be3b8c2dc45d8b28eded1190587ef', 'name': 'pred_bat', 'objectid': 'c35cc65a9ebb4dff9523badac3a6cc10', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7fb620ded0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2024-07-16 10:41:19.110008 WARNING pred_bat: ------------------------------------------------------------
2024-07-16 10:41:19.113237 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/config/apps/predbat.py", line 10949, in run_time_loop
    raise e
  File "/config/apps/predbat.py", line 10944, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/apps/predbat.py", line 9788, in update_pred
    recompute = self.calculate_plan(recompute=recompute)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 7959, in calculate_plan
    self.calculate_yesterday()
  File "/config/apps/predbat.py", line 7755, in calculate_yesterday
    ) = self.run_prediction(charge_limit_best, charge_window_best, [], [], False, end_record=(24 * 60), save="yesterday")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/apps/predbat.py", line 2105, in run_prediction
    self.iboost_running_solar = pred.iboost_running_solar
                                ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Prediction' object has no attribute 'iboost_running_solar'

2024-07-16 10:41:19.113789 WARNING pred_bat: ------------------------------------------------------------

duckfullstop, 2024-07-17:
For what it's worth, a full restart of both _HA_ and _appdaemon_ seemed to fix this one for me (presumably some entities needed regenerating?)

Jorthax, 2024-07-18:
I am sure I had done this - but now it appears to be working correctly - Thank you.

closed 2024-07-18 by Jorthax

LynchMicro, 2024-08-01:
Sorry if this is a stupid question but I have tried google and appdaemon reference docs and can't work out how to restart appdaemon. I've done a full reboot of the system running HA but guess this hasn't restarting appdaemon in the correct way as I still have the same error as @Jorthax

LynchMicro, 2024-08-01:
Nevermind, it would seem I just needed to wait a bit longer +/- restart HA again.

gcoan, 2024-08-02:
> Nevermind, it would seem I just needed to wait a bit longer +/- restart HA again.

Appdaemon runs as a separate server to HA so if you just restart HA, the default option will not restart any add-on's.

You can either click the 'more optiopns' on restart and do a full restart of HA and teh add-ons or you can navigate to nthe addons in HA, click Appdaemon and just restart that addon