springfall2008/batpred #1312: Failed to start after installing V8.3.0 from HA update

lucode, 2024-07-16:
**Describe the bug**
As this is a brand new setup I can't get all the details of the process. But I'm quite sure that in the previous version my setup did start, but still with errors. It did create a dashboard yaml file with showed some values. 

From the logs I can see that the app is not happy with the provided value **soc_kw**.  (guess it is asking for kWh?)

```
2024-07-16 17:20:57.901524: Warn: Inverter 0: Using inverter type SF - not all features are available
2024-07-16 17:20:57.903206: Warn: Return bad float value multiplus_battery_soc_kwh from soc_kw using default 0.0
2024-07-16 17:20:57.937280: Info: record_status Warn: Return bad float value multiplus_battery_soc_kwh from soc_kw
2024-07-16 17:20:58.942352: Error: Exception raised 0
2024-07-16 17:20:58.948000: Error: Traceback (most recent call last):

```
**Expected behavior**
As I did change the template for the **soc_kw** entity to output an integer I can't explain what still should be wrong and anyway with the older version predbat did start despite this error.

**Predbat version**

V8.3.0

**Environment details**

- Inverter and battery setup
I do have a Victron Inverter and am in  the process to get known predbat and the conf file. Is use the SOFAR template and the choosen inverter type is _SF_.

- Standard HAOS installer or Docker
Addon installer 
- Anything else?
I did post the current apps.yaml file here: https://github.com/HA-projects/predbat/blob/main/Inverters/Victron/victron_multiplus.yaml

**Screenshots**

**Log file**

```
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[17:20:50] INFO: Configuring timezone (Europe/Paris)...
Predbat files are installed correctly for version v8.3.0
**** Starting Standalone Predbat ****
2024-07-16 17:20:51.359843: Loading apps.yaml
s6-rc: info: service base-addon-log-level successfully started
2024-07-16 17:20:51.561387: Info: Connected to Home Assistant at http://supervisor/core
2024-07-16 17:20:51.561546: Creating task: <coroutine object HAInterface.socketLoop at 0x7f036b0a1d90>
2024-07-16 17:20:51.563051: Info: Start socket for url http://supervisor/core/api/websocket
2024-07-16 17:20:51.564717: Info: Web Socket task started
2024-07-16 17:20:51.565025: Warn: unable to find /config/appdaemon.yaml skipping checks as maybe outside AppDaemon
2024-07-16 17:20:51.592957: Info: Web Socket active
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Watching ['./download.py', './ha.py', './predbat.py', './inverter.py', './unit_test.py', './utils.py', './apps.yaml', './hass.py', './prediction.py', './config.py'] for changes
2024-07-16 17:20:57.901524: Warn: Inverter 0: Using inverter type SF - not all features are available
2024-07-16 17:20:57.903206: Warn: Return bad float value multiplus_battery_soc_kwh from soc_kw using default 0.0
2024-07-16 17:20:57.937280: Info: record_status Warn: Return bad float value multiplus_battery_soc_kwh from soc_kw
2024-07-16 17:20:58.942352: Error: Exception raised 0
2024-07-16 17:20:58.948000: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10893, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9776, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9290, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
  File "/config/inverter.py", line 444, in find_charge_curve
    soc_kwh_data[0],
KeyError: 0
```

gcoan, 2024-07-16:
It looks like your apps.yaml is pointing to an invalid sensor name:

```yaml

soc_kw:
    - multiplus_battery_soc_kwh
```

It should be configured to point to a sensor name that contains the current battery soc value.

Maybe sensor.multiplus_battery_soc_kwh ?

lucode, 2024-07-17:
Ok, thanks a lot, the error message "Return bad float value multiplus_battery_soc_kwh" did puzzle me. Somehow I must have deleted the sensor. part.

closed 2024-07-17 by lucode