springfall2008/batpred #1313: Sunsynk inverter charge current 0A after grid charge slot

tmh88, 2024-07-16:
**Describe the bug**
After grid charge or during solar production (non-scheduled grid charge slot) the battery current drops to 0A not allowing the battery to charge from solar.

**Expected behavior**
I would expect the battery charge current to remain to allow charging of the battery via solar.

The sunsynk inverters charge from solar and discharge from battery along as the battery SOC > SOC setpoint for that time slot.

Both gird & solar charging of the battery is controlled by the max charge current and only charge from grid if grid charge is checked and battery SOC is < SOC setpoint.

Discharging is controlled by max discharge current & workmode selected. Limit to Home, and don't discharge battery to grid. Only if "Selling First" is selected then the battery will export to the grid.

I have followed the example yaml file in the repo.

**Predbat version**

8.3.0

**Environment details**

- Sunsynk Inverter and battery
- Standard HAOS installer
- app.yaml
[apps.txt](https://github.com/user-attachments/files/16254888/apps.txt)

**Screenshots**
[image]

[image]

[image]

**Log file**
[Logs 16-07-24 8-30.txt](https://github.com/user-attachments/files/16254889/Logs.16-07-24.8-30.txt)

springfall2008, 2024-07-17:
So you are saying to start/stop a grid charge you have to change the target SOC to minimum for charge from solar only and to the target for grid charging?

springfall2008, 2024-07-17:
Please test the version on 'main' which should have a fix for this issue

tmh88, 2024-07-17:
> So you are saying to start/stop a grid charge you have to change the target SOC to minimum for charge from solar only and to the target for grid charging?

in short yes, there is also a check box for the time slot to enable/disable grid charging but i can write an automation based on the output binary_sensor.predbat_charging

tmh88, 2024-07-17:
> Please test the version on 'main' which should have a fix for this issue

thank you, looks promising so far will leave it on test and let you know
[image]

if this works, i can write up and upload my configs and automations if that helps?

springfall2008, 2024-07-20:
> thank you, looks promising so far will leave it on test and let you know [...]

Yes please

tmh88, 2024-07-21:
@springfall2008 log file as discussed
[Log 21-07-24.txt](https://github.com/user-attachments/files/16323592/Log.21-07-24.txt)

springfall2008, 2024-07-21:
Not sure I see why, the discharge rate is set back to full at 8:30 when charging stops?

2024-07-21 08:30:18.082555: Switch sensor.predbat_SK_0_scheduled_charge_enable is now off
2024-07-21 08:30:18.082694: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
2024-07-21 08:30:18.082730: Inverter 0 Turning off scheduled charge
2024-07-21 08:30:18.082803: Inverter 0 Call service template charge_stop_service = {'service': 'input_select.select_option', 'entity_id': 'input_select.test_sunsynk_work_mode', 'option': 'Limited to Home'}
2024-07-21 08:30:18.082856: Inverter 0 Call service input_select/select_option with data {'entity_id': 'input_select.test_sunsynk_work_mode', 'option': 'Limited to Home'}
2024-07-21 08:30:18.097567: Inverter 0 current discharge rate is 0.0 and new target is 6444
2024-07-21 08:30:23.123127: Inverter 0 Wrote 6444 to discharge_rate, successfully now 6444
2024-07-21 08:30:28.154124: Inverter 0 Wrote 120.0 to timed_discharge_current, successfully now 120.0
2024-07-21 08:30:28.154347: Not setting charging SOC as we are not within the window (now 07-21 08:30:00 target set_soc_minutes 30 charge start time 07-22 01:30:00)
2024-07-21 08:30:28.154417: Inverter 0 Current Target SOC is 10 already at target
2024-07-21 08:30:28.177420: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes

Then after that you restart Predbat (no sure why):

2024-07-21 08:30:28.177499: iBoost model today updated to 0.0
2024-07-21 08:30:28.219728: Completed run status Idle
2024-07-21 08:30:29.469701: Info: record_status Idle
2024-07-21 08:30:29.481902: Saved current settings to /config/predbat_config.json
2024-07-21 08:34:03.731458: Stopping Predbat
2024-07-21 08:34:03.731670: Predbat terminating
2024-07-21 08:34:03.731875: Predbat terminated
2024-07-21 08:34:03.815786: Info: Web socket stopping
2024-07-21 08:34:03.817448: Info: Web socket stopping
2024-07-21 08:34:35.671295: Loading apps.yaml
2024-07-21 08:34:35.698699: Predbat: Startup predbat

springfall2008, 2024-07-21:
So what's happening is once you reach your charged SOC it turns off discharge to hold the battery at that level and then turns it back on when the charging period ends

tmh88, 2024-07-21:
> So what's happening is once you reach your charged SOC it turns off discharge to hold the battery at that level and then turns it back on when the charging period ends

The target soc was 10% and my batteries never dropped below 55% so I would expect it to discharge the batteries until 10%. Then hold at setpoint ?

[image]

tmh88, 2024-07-21:
Battery soc wouldn't have dropped below 72% if predbat was controlling the inverter, currently just using test entities whilst testing

springfall2008, 2024-07-21:
Oh I see, I know why I believe its a setting that perhaps shouldn't be that way, let me try a fix on main

springfall2008, 2024-07-21:
Fix on main for testing

tmh88, 2024-07-21:
@springfall2008 Thankyou, do i need to change to the sunsynk branch? dosent seem to be pulled into the main branch?

did you get anywhere with allowing the charge current to change when charging from solar?

springfall2008, 2024-08-03:
did the fix work?

tmh88, 2024-08-03:
> did the fix work?

Hi @springfall2008 yes the fix worked, been running for about a week or so now. I've also uploaded my configs for other if you want to add it to the documents.

Thanks for all your help and hard work

closed 2024-08-05 by springfall2008