springfall2008/batpred #1316: Pause mode not working on latest AIO firmware

dochtie, 2024-07-18:
**Describe the bug**
After updating to D0.612-A0.612 and BMS10 predbat is no longer able to control pause mode.

I am also unable to control pause mode directly via GivTCP so not really a predbat issue.

Raising for awareness and if anyone knows a temp workaround. As far as I know you can’t roll back a GE firmware update.

**Expected behavior**
Pause mode should work via GIVTCP. Pause mode works as normal through GE cloud portal.

**Predbat version**

8.3.0

**Environment details**

D0.612-A0.612 and BMS10 
AIO 

**Screenshots**

**Log file**
[image]

gcoan, 2024-07-18:
Are you on the main GivTCP release or the beta?  I believe the beta has better aio support

dochtie, 2024-07-18:
> Are you on the main GivTCP release or the beta? I believe the beta has better aio support

I’m on the main. Is the beta pretty stable do you know?

dochtie, 2024-07-18:
@gcoan I gave it a try and it works on the beta (via GivTCP anyway). Thanks for pointing me in that direction!

springfall2008, 2024-07-20:
Not a predbat issue as such, closing

closed 2024-07-20 by springfall2008