springfall2008/batpred #1318: No apps.yaml

masongary, 2024-07-18:
I've installed predbat but get no apps.yaml to edit. The code is running and clearly finding the configutration file but I'm unable to locate it to configure to my inverter and remove the template line - where can I find it?

*Add-on: predbat
Bootstrap Predbat
Startup
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------

gcoan, 2024-07-18:
Strange.

That log doesn't show how far the add-on has got, only that its started.

Details of the directory that Predbat code and the template apps.yaml is installed in is in the install documentation https://springfall2008.github.io/batpred/install/#predbat-add-on-install

Can you see what's in there?

If there are several .py files but no apps.yaml you can just copy it across from the appropriate template file for your inverter - see https://springfall2008.github.io/batpred/install/#inverter-control-integration-install

masongary, 2024-07-19:
Sadly I don’t get any folder or .py files. The log would suggest it’s running something but then stops as the template mode is still on

gcoan, 2024-07-19:
There's no predbat subfolder under /addon_configs and you have followed the instructions for your editor to access different folders on the HA server https://springfall2008.github.io/batpred/install/#editing-configuration-files-in-home-assistant ?

Can you share the logfile please.  Might have to try removing the addon and reinstalling it if its not installed properly

masongary, 2024-07-19:
Yeah believe I’ve done everything there just the directory and file does not appear. 

I’ve just uninstalled, restarted ha and then added the repository and installed again but same problem. Here’s the log file once it boots. I my clearly finds a configuration file but I just don’t see it….

s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[20:46:47] INFO: Configuring timezone (Europe/London)...
Predbat files are installed correctly for version v8.3.0
**** Starting Standalone Predbat ****
2024-07-18 20:46:47.995540: Loading apps.yaml
s6-rc: info: service base-addon-log-level successfully started
2024-07-18 20:46:48.444931: Info: Connected to Home Assistant at http://supervisor/core
2024-07-18 20:46:48.445337: Creating task: <coroutine object HAInterface.socketLoop at 0x7fb3e761f0>
2024-07-18 20:46:48.447136: Info: Web Socket task started
2024-07-18 20:46:48.447891: Warn: unable to find /config/appdaemon.yaml skipping checks as maybe outside AppDaemon
2024-07-18 20:46:48.453528: Info: Start socket for url http://supervisor/core/api/websocket
2024-07-18 20:46:48.542356: Warn: Regular expression argument: geserial unable to match re:sensor.givtcp_(.+)_soc_kwh, now will disable
2024-07-18 20:46:48.545403: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
2024-07-18 20:46:48.556531: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-07-18 20:46:48.559506: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-07-18 20:46:48.562547: Warn: Regular expression argument: octopus_intelligent_slot unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_intelligent_dispatching), now will disable
2024-07-18 20:46:48.565640: Warn: Regular expression argument: octopus_ready_time unable to match re:(time.octopus_energy([0-9a-z_]+|)_intelligent_ready_time), now will disable
2024-07-18 20:46:48.568678: Warn: Regular expression argument: octopus_charge_limit unable to match re:(number.octopus_energy([0-9a-z_]+|)_intelligent_charge_limit), now will disable
2024-07-18 20:46:48.572665: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-07-18 20:46:48.588069: Info: Web Socket active
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-07-18 20:46:51.635871: Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
2024-07-18 20:46:51.678577: Info: record_status Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
Watching ['./download.py', './prediction.py', './hass.py', './unit_test.py', './apps.yaml', './ha.py', './utils.py', './inverter.py', './config.py', './predbat.py'] for changes
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service init-predbat: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service init-predbat successfully stopped
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service init-predbat: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service init-predbat successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
[08:01:20] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: a9177b14665e9a596a4ae960156519b2a7866accf801ff7bd5262b811f52203298155cbe1e741c66967fe54a5834efd61ca945b2f2bd3c49

-----------------------------------------------------------
 Add-on: predbat
Bootstrap Predbat
Startup
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
[... 10 lines cut ...]
Predbat files are installed correctly for version v8.3.0
**** Starting Standalone Predbat ****
2024-07-19 08:01:29.358001: Loading apps.yaml
2024-07-19 08:01:30.597775: Info: Connected to Home Assistant at http://supervisor/core
2024-07-19 08:01:30.598039: Creating task: <coroutine object HAInterface.socketLoop at 0x7faf3a61f0>
[... cut ...]
s6-rc: info: service legacy-services successfully started
2024-07-19 08:01:33.460424: Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
2024-07-19 08:01:33.501588: Info: record_status Error: You still have a template configuration, please edit apps.yaml or restart AppDaemon if you just updated with HACS
Watching ['./download.py', './prediction.py', './hass.py', './unit_test.py', './apps.yaml', './ha.py', './utils.py', './inverter.py', './config.py', './predbat.py'] for changes

masongary, 2024-07-19:
I found it - apologies I just found the line about exposing the root to see the config files

masongary, 2024-07-19:
Thank you for your help

closed 2024-07-19 by masongary

gcoan, 2024-07-19:
> I found it - apologies I just found the line about exposing the root to see the config files

RTFM !

Glad you're sorted now.  I have tried to cover everything in the documentation.

Let us know if you need any further help

Cheers