springfall2008/batpred #1322: how to update to 7.22 when it is not in the dropdown list

Jingbin-GitHub, 2024-07-22:
Sorry, newbie and this is probably a silly question. I am on 7.20.4 and the latest is 8.3.1. I need to get to 7.22.5 first but it is not in the drop-down list. How to get to 7.22.5?

[image]

gcoan, 2024-07-23:
You can manually copy the 7.22.5 code from github, unzip it, place the files in your apps directory replacing all the .py files and restart appdaemon 

https://github.com/springfall2008/batpred/releases/tag/v7.22.5

Jingbin-GitHub, 2024-07-23:
thanks, i am on 7.22.5 now. Can I update to the latest from the drop-down list or I need to use the same method to get to 8.0.0 first?

gcoan, 2024-07-23:
I'd recommend you upgrade to 7.22.5 first using the method above and then once predbat is running OK you can use the predbat update control to upgrade to any future predbat version. That's what the release descriptions recommend.

You might be OK to follow a similar manual download process to go to any version 8.0.0+ version, but the inbuilt predbat update selector wont work as from v7.22.5 predbat has been split into multiple files and the older predbat versions are not aware of this change

closed 2024-08-18 by springfall2008