springfall2008/batpred #1323: Plan Discharging At Loss

nbullus, 2024-07-22:
**Describe the bug**
Battery is being discharged at a loss. See attachment showing plan discharging at a loss.

**Expected behavior**
Battery should not discharge back to the grid at higher cost than what it cost to charge it up.

**Predbat version**

8.3.1

**Environment details**

- Solis 5kw Inverter + GivEnergy AIO Battery
- Standard HAOS installer
- Settings:
   predbat_metric_battery_value_scaling = 1
   predbat_metric_min_improvement_discharge = 0.1

**Screenshots**

See attached document containing charts and plan.

**Log file**

See attached document containing Predbat log.

[Plan, Charts, Log.docx](https://github.com/user-attachments/files/16340144/Plan.Charts.Log.docx)

cyberkryten, 2024-08-01:
Predbat doesn't compute the cost of each kWh in the battery and track it.  It plans best times to charge/discharge to maximise returns.  So, if the battery is full and there's money to be made discharging then it will do so if that power isn't needed.  

It is possible that it force charged at an expensive time, thinking the house would need more power than it did, in which case the issue is more that it charged incorrectly than it then exported that power.

Can you add log/debug output and the plan - that's more useful than a word doc which I'm not going to open I'm afraid.

nbullus, 2024-08-01:
The attached comment contains the Predbat log and plan already. If you wan't to review the contents in a different application then just cut/paste as required. I don't have the contents in any other format now.

closed 2025-01-01 by springfall2008