springfall2008/batpred #1325: Adding support for a Fronius inverter

knackerbrot, 2024-07-23:
I am trying to run predbat/batpred in monitor mode so I can investigate adding support for my inverter/battery (Fronius Symo Gen24 with BYD HVM battery)

I've modified the apps.yaml but can't seem to get the addon to load. 

apps.yaml and relevent logs are attached. 
[apps-yaml.txt](https://github.com/user-attachments/files/16348177/apps-yaml.txt)
[predbad.log](https://github.com/user-attachments/files/16348178/predbad.log)

I've tried several different minor changes to the apps.yaml. I know I don't have a forecast from solcast set up yet, I'm just aiming to load the addon properly so I can investigate further. Apologies - I'm pretty clueless generally, 

Happy to provide more context if required - thank you

gcoan, 2024-07-23:
you are missing a definition of inverter_type in the apps.yaml which tells Predbat what type of inverter you have.

Either start off with one of the existing inverter types that predbat knows about (that behaves similarly to your inverter), and customise from there, or define it as being a completely bespoke inverter type

see https://springfall2008.github.io/batpred/other-inverters/ - the last bit to define a custom inverter type which is what it looks like what you are doing?

knackerbrot, 2024-07-24:
thanks! yep - it's my intention to define a custom inverter type. 

your advice was helpful: predbat runs and generates predbat_dashboard.yaml once I add `  inverter_type: "SF"` to apps.yaml.

I can now continue to investigate. 

Someone has already developed a tool that interacts with this inverter type via MODBUS that seems to be working: https://github.com/atbore-phx/sbam

The sbam tool has limited functionality so I'm hoping to transfer the MODBUS logic over from sbam to predbat.

Any further advice would be appreciated.

springfall2008, 2024-08-06:
I think you need to split the work into two steps:

1. Get your inverter working in Home Assistant and confirm you can control it.
2. Configure Predbat to work with your inverter via the Home Assistant Controls

Its not clear to me if you have step 1 yet?

knackerbrot, 2024-08-06:
Ok thanks. 

The inverter is being successfully controlled via homeassistant but it's currently through sbam. 

I will firstly focus on understanding exactly how sbam is managing to control the battery charging - appreciate the advice.

knackerbrot, 2025-10-28:
Yes, as far as I understand the communication between batpred and Fronius is what's required. Theoretically, the fact that sbam manages to communicate with Fronius should mean that the solution 'just' needs to be transcribed to work with batpred. 

My own situation is currently that I'm busy with other things and the likely financial benefits of an effective weather/usage/etc. informed battery management system would be limited (for my particular case).

closed 2026-08-28 by chalfontchubby