springfall2008/batpred #1326: New setup does not show any usable data to control my inverter or EV charger

lucode, 2024-07-23:
**Describe the bug**
I have read much of the documentation and spent some time understanding the app. It works now and outputs reasonable prediction data, but I can't find any entities I could use to control my inverter or EV charger.

The entities for the Car charger are as follows:
[image]
The value "EV charging slot" is getting synchronized with my actual charging time, but does not predict anything.

The Inverter is from Victron and I used the SF inverter type for my setup.
Link to my apps.yaml:
https://github.com/HA-projects/predbat/blob/main/Inverters/Victron/victron_multiplus_v2.yaml

**Expected behaviour**
I tried the mode SOC-only and Control charge, but I can't see any difference. In the docs a value is mentioned "target charge percentage" and a charge window, which are both not present in my setup.

> In Control SOC-only mode Predbat will adjust the target charge percentage (SOC target) according to the Best plan, but the charge window will not be modified.

It is very unclear to me how to use this app in this reagrd, Or it is meant to use Prediction Best values to control the SOC of my battery? 

**Predbat version**
8.3.1

**Environment details**

- Inverter Victron as SF and battery setup via Victron and Helpers
- Standard HAOS installer 
- EV charger via a Switch and Renault API, plus power meter

**Screenshots**
[image]

**Log file**
```
Startup
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.3
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[17:41:47] INFO: Configuring timezone (Europe/Paris)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Predbat files are installed correctly for version v8.3.1
**** Starting Standalone Predbat ****
2024-07-23 17:41:48.925493: Loading apps.yaml
2024-07-23 17:41:49.032679: Info: Connected to Home Assistant at http://supervisor/core
2024-07-23 17:41:49.034715: Creating task: <coroutine object HAInterface.socketLoop at 0x7f9a23742f10>
2024-07-23 17:41:49.036429: Info: Start socket for url http://supervisor/core/api/websocket
2024-07-23 17:41:49.036848: Info: Web Socket task started
2024-07-23 17:41:49.037605: Warn: unable to find /config/appdaemon.yaml skipping checks as maybe outside AppDaemon
2024-07-23 17:41:49.088028: Info: Web Socket active
Watching ['./download.py', './ha.py', './predbat.py', './inverter.py', './unit_test.py', './utils.py', './apps.yaml', './hass.py', './prediction.py', './config.py'] for changes
```

springfall2008, 2024-08-05:
I'm not sure I understand the question, but SOC only mode is when you have configured a fixed charge window and you just want Predbat to decide the charge %. I'd recommend control Charge/Discharge initially.

In your apps.yaml you can't control your system as you hard-wired the controls, these are meant to point to HA switches/numbers that Predbat can modify.

There are various control methods that can be selected in the inverter configuration section, either using these controls directly but also be using a service API to HA.

How can you control your inverter in HA currently, what do the controls look like?

```
charge_limit:
   - 100
  reserve:
    - 20
  scheduled_charge_enable:
   - off
  scheduled_discharge_enable:
   - off
  charge_start_time:
   - off
  charge_end_time:
   - off
  discharge_start_time:
   - off
  discharge_end_time:
   - off
  ```

lucode, 2024-08-06:
> How can you control your inverter in HA currently, what do the controls look like?

Thank you for the hints. I start to understand now. So it means that "charge_limit" is a sensor, controlled by Predbat to set my SOC Target? And, charge_start/stop_time are setting the schedule to reach the SOC Target?
And what about the service API (charge_start_service - Should be set to a service that is called when charging starts)?
Are those if I don't want to use the  charge_start/stop_time  entities to control my charge?

Frankly, the list of the inverter controls should be more precise about the entities. At least they should be divided in two categories, which makes it clear which ones are input sensors or hard-coded and which are entities (variables) that are controlled by Predbat.
https://springfall2008.github.io/batpred/apps-yaml/#home-assistant-entity-inverter-control
 ```
 # Entity name for used to set the SOC target for the battery in percentage (AC charge target)    
  charge_limit:
   - sensor.victron_predbat_soc_target
  reserve:
    - 20
  scheduled_charge_enable:
   - off
  scheduled_discharge_enable:
   - off
  charge_start_time:
   - sensor.victron_predbat_charge_start_time
  charge_end_time:
   - sensor.victron_predbat_charge_stop_time
```

springfall2008, 2024-08-07:
Basically yes, although sensor. are read only entities so you can't use those for controls, they have to be a select. for times I believe.

If you don't have some of these then the inverter options can disable them. The service API is called if you have has_service_api enabled.

closed 2026-08-28 by chalfontchubby