springfall2008/batpred #1327: SolarEdge "Hold Charge" continues to charge battery from grid

Noodleyman, 2024-07-24:
**Describe the bug**
I've raised this a few times before via general discussions, but rarely had the chance to capture useful information related to the problem. However, today I was able to capture the system doing something it should not be. The issue relates to the SolarEdge system and when PredBat wants to Pause charging (hold charge).  Hold Charge sets the SolarEdge mode to be "Charge from Solar Power and Grid" which continues to pull power from the grid. When it does this there have been instances where it will continue to walk up the "pause charge" target SOC to 100% and thus enter a cycle of non-stop charging from grid until the battery is full.

When I catch it doing this, I set it to read only and manually configure the "Maximize Self Consumption" mode. 

**Expected behavior**
Hold Charge should not continue to charge from grid

**Predbat version**
8.3.1 (Have had this issue for quite a long time, so is not new to this version)

**Environment details**
- Solar Edge SE8000H, with 10kWh home battery and home backup module
- Standard HAOS installer running on an rPi4

**Screenshots**
The Plan:
[image]

The actual state of the PV system:
[image]

PredBat status 
[image]

SolarEdge Modbus Multi history showing status change:
[image]

Input Variables:
[image]

Switches:
[image]

**apps.yaml**
```
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # XXX: This is a configuration template, delete this line once you edit your configuration
  # template: True

  # Timezone to work in
  timezone: Europe/London

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  load_today:
    - sensor.solar_house_consumption_kwh
    #- sensor.solaredge_i1_ac_energy
    #- sensor.solaredge_i1_ac_energy_kwh
  import_today:
    - sensor.solaredge_m1_imported
    #- sensor.solaredge_m1_imported_kwh
  export_today:
    - sensor.solaredge_m1_exported
    #- sensor.solaredge_m1_exported_kwh
  pv_today:
    - sensor.solar_panel_production_kwh   # This is a custom sensor - see Readme on how to create or comment out
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  # Inverter type stubbed to SX4 to allow execution - SE is an invalid type in current head version
[... 269 lines cut ...]
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0
```

Noodleyman, 2024-07-24:
Adding log as a 2nd post due to content length

**Log file**
```
024-07-24 08:57:22.209557 INFO AppDaemon: New client Admin Client connected
2024-07-24 08:55:58.962328 INFO pred_bat: Saved current settings to /config/predbat_config.json
2024-07-24 08:55:58.915741 INFO pred_bat: Info: record_status Hold charging target 62%-59%
2024-07-24 08:55:58.868587 INFO pred_bat: Completed run status Hold charging
2024-07-24 08:55:58.706236 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-07-24 08:55:58.627117 INFO pred_bat: Inverter 0 Wrote 4.0 to reserve, successfully now 4
2024-07-24 08:55:50.053928 INFO pred_bat: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-07-24 08:55:46.993299 INFO pred_bat: Inverter 0 Current Reserve is 60.0 % and new target is 4.0 %
2024-07-24 08:55:46.978091 INFO pred_bat: Adjust reserve to default as SOC 62 % is above target 59 % or charging active
2024-07-24 08:55:46.962574 INFO pred_bat: Adjust idle time computed idle is 08:55:00-23:59:00
2024-07-24 08:55:46.954241 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-07-24 08:55:46.946046 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00
2024-07-24 08:55:46.930696 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 07-25 18:35:00 - 07-25 19:00:00
2024-07-24 08:55:46.916459 INFO pred_bat: Next discharge window will be: 2024-07-25 18:35:00+01:00 - 2024-07-25 19:01:00+01:00 at reserve 74
2024-07-24 08:55:46.899931 INFO pred_bat: Adjust idle time computed idle is 08:55:00-23:59:00
2024-07-24 08:55:46.892478 INFO pred_bat: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-07-24 08:55:45.772843 INFO pred_bat: Inverter 0 Call service select/select_option with data {'entity_id': 'select.solaredge_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2024-07-24 08:55:45.769209 INFO pred_bat: Inverter 0 Call service template charge_start_service = {'service': 'select.select_option', 'entity_id': 'select.solaredge_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2024-07-24 08:55:45.765773 INFO pred_bat: Charge window will be: 2024-07-24 08:30:00+01:00 - 2024-07-24 09:00:00+01:00 - current soc 62 target 59
2024-07-24 08:55:45.724977 INFO pred_bat: Low import rate windows [0: 07-24 08:30:00 - 07-24 09:00:00 @ 22.1, 1: 07-24 09:00:00 - 07-24 09:30:00 @ 22.1, 2: 07-24 09:30:00 - 07-24 10:00:00 @ 22.1, 3: 07-24 10:00:00 - 07-24 10:30:00 @ 22.1, 4: 07-24 10:30:00 - 07-24 11:00:00 @ 22.1, 5: 07-24 11:00:00 - 07-24 11:30:00 @ 22.1, 6: 07-24 11:30:00 - 07-24 12:00:00 @ 22.1, 7: 07-24 12:00:00 - 07-24 12:30:00 @ 22.1, 8: 07-24 12:30:00 - 07-24 13:00:00 @ 22.1, 9: 07-24 13:00:00 - 07-24 13:30:00 @ 22.1, 10: 07-24 13:30:00 - 07-24 14:00:00 @ 22.1, 11: 07-24 14:00:00 - 07-24 14:30:00 @ 22.1, 12: 07-24 14:30:00 - 07-24 15:00:00 @ 22.1, 13: 07-24 15:00:00 - 07-24 15:30:00 @ 22.1, 14: 07-24 15:30:00 - 07-24 16:00:00 @ 22.1, 15: 07-24 19:00:00 - 07-24 19:30:00 @ 22.1, 16: 07-24 19:30:00 - 07-24 20:00:00 @ 22.1, 17: 07-24 20:00:00 - 07-24 20:30:00 @ 22.1, 18: 07-24 20:30:00 - 07-24 21:00:00 @ 22.1, 19: 07-24 21:00:00 - 07-24 21:30:00 @ 22.1, 20: 07-24 21:30:00 - 07-24 22:00:00 @ 22.1, 21: 07-24 22:00:00 - 07-24 22:30:00 @ 22.1, 22: 07-24 22:30:00 - 07-24 23:00:00 @ 22.1, 23: 07-24 23:00:00 - 07-24 23:30:00 @ 22.1, 24: 07-24 23:30:00 - 07-25 00:00:00 @ 22.1, 25: 07-25 00:00:00 - 07-25 00:30:00 @ 22.1, 26: 07-25 00:30:00 - 07-25 01:00:00 @ 22.1, 27: 07-25 01:00:00 - 07-25 01:30:00 @ 22.1, 28: 07-25 01:30:00 - 07-25 02:00:00 @ 22.1, 29: 07-25 02:00:00 - 07-25 02:30:00 @ 13.26, 30: 07-25 02:30:00 - 07-25 03:00:00 @ 13.26, 31: 07-25 03:00:00 - 07-25 03:30:00 @ 13.26, 32: 07-25 03:30:00 - 07-25 04:00:00 @ 13.26, 33: 07-25 04:00:00 - 07-25 04:30:00 @ 13.26, 34: 07-25 04:30:00 - 07-25 05:00:00 @ 13.26, 35: 07-25 05:00:00 - 07-25 05:30:00 @ 22.1, 36: 07-25 05:30:00 - 07-25 06:00:00 @ 22.1, 37: 07-25 06:00:00 - 07-25 06:30:00 @ 22.1, 38: 07-25 06:30:00 - 07-25 07:00:00 @ 22.1, 39: 07-25 07:00:00 - 07-25 07:30:00 @ 22.1, 40: 07-25 07:30:00 - 07-25 08:00:00 @ 22.1, 41: 07-25 08:00:00 - 07-25 08:30:00 @ 22.1, 42: 07-25 08:30:00 - 07-25 09:00:00 @ 22.1, 43: 07-25 09:00:00 - 07-25 09:30:00 @ 22.1, 44: 07-25 09:30:00 - 07-25 10:00:00 @ 22.1, 45: 07-25 10:00:00 - 07-25 10:30:00 @ 22.1, 46: 07-25 10:30:00 - 07-25 11:00:00 @ 22.1, 47: 07-25 11:00:00 - 07-25 11:30:00 @ 22.1, 48: 07-25 11:30:00 - 07-25 12:00:00 @ 22.1, 49: 07-25 12:00:00 - 07-25 12:30:00 @ 22.1, 50: 07-25 12:30:00 - 07-25 13:00:00 @ 22.1, 51: 07-25 13:00:00 - 07-25 13:30:00 @ 22.1, 52: 07-25 13:30:00 - 07-25 14:00:00 @ 22.1, 53: 07-25 14:00:00 - 07-25 14:30:00 @ 22.1, 54: 07-25 14:30:00 - 07-25 15:00:00 @ 22.1, 55: 07-25 15:00:00 - 07-25 15:30:00 @ 22.1, 56: 07-25 15:30:00 - 07-25 16:00:00 @ 22.1, 57: 07-25 19:00:00 - 07-25 19:30:00 @ 22.1, 58: 07-25 19:30:00 - 07-25 20:00:00 @ 22.1, 59: 07-25 20:00:00 - 07-25 20:30:00 @ 22.1, 60: 07-25 20:30:00 - 07-25 21:00:00 @ 22.1, 61: 07-25 21:00:00 - 07-25 21:30:00 @ 22.1, 62: 07-25 21:30:00 - 07-25 22:00:00 @ 22.1, 63: 07-25 22:00:00 - 07-25 22:30:00 @ 22.1, 64: 07-25 22:30:00 - 07-25 23:00:00 @ 22.1, 65: 07-25 23:00:00 - 07-25 23:30:00 @ 22.1, 66: 07-25 23:30:00 - 07-26 00:00:00 @ 22.1, 67: 07-26 00:00:00 - 07-26 00:30:00 @ 22.1, 68: 07-26 00:30:0 [... cut ...]
2024-07-24 08:50:00.196265 INFO pred_bat: Predbat /config/apps/predbat.py version v8.3.1 currently running, latest version is v8.3.1 latest beta v8.3.1
2024-07-24 08:50:00.193554 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 100.0 minutes
2024-07-24 08:50:00.134091 INFO pred_bat: Saved current settings to /config/predbat_config.json
2024-07-24 08:50:00.130663 INFO pred_bat: --------------- PredBat - update at 2024-07-24 08:50:00+01:00 with clock skew 0 minutes, minutes now 530
```

Noodleyman, 2024-07-24:
I'm assuming it is related to the "Warn: Failed to decode response" error, but I can't find any other errors or useful references in any other logs around the same time, so unsure what causes that or how to gather more details about what is causing it.

cyberkryten, 2024-08-01:
I've only got one of those Warn responses at it's at the same point of execution:

```
2024-07-31 19:00:21.037162 INFO pred_bat: Inverter 0 Current Reserve is 49.0 % and new target is 10.0 %                                                                                                                                                                                                             
2024-07-31 19:00:24.052280 INFO pred_bat: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value                                                                                                                                                                                                          
2024-07-31 19:00:32.686981 INFO pred_bat: Inverter 0 Wrote 10.0 to reserve, successfully now 10   

```

However, I'm guessing it wasn't an issue or was covered by the retries at history shows that the Reserve did get changed to 10% at 19:00:32, so whilst there may be a bug in there (most likely actually the inverter failing to cope!) it didn't have any effect (on my system at least).  Does HA history show the value changed?

If it is "Holding Charge" then it has to get the power for the house from the grid, otherwise it would have to discharge the battery to power the house - it's battery or grid overnight.  

If that is the intent (ie don't charge or discharge the battery) then I guess it could set the "Solar Power Only" mode (effectively Off as far as the battery is concerned) but I don't think that mode is supported by Predbat so I guess the Grid Charge is the closest mode that does what is intended?

As for setting Discharge rate, that works fine on my SE8000, it holds zero (or whatever other value is set) until the next overnight inverter reset - before I started using Predbat I had an automation to reset it to the required value.  I know it works as I've been very closely monitoring (due to other issues).

Noodleyman, 2024-08-01:
Morning @cyberkryten 

Thanks for sharing your thoughts. You are correct abou tthe value of 0 for discharge rate. I reached out on the Modbus github community and validated that 0 is valid.

Interesting to see you have the same 500 error. I'm going to have to try and setup some tests to trigger the issue again and capture more details I think. The issue I have is replicating it this time of year with the way I have it setup is difficult. So I use  set of automations to override Predbat so I can to Charge From Clipped mode when the forecast > 40kWh, that is almost every day this time of year thus the windows of time where it can manage the plans and want to hold is minimal. I might turn those off and keep an eye on it next week.

Based on the definitions  in the PreddBat documentation:
_Hold charging - A type of charge where the target SoC % is the same as the current SoC %, effectively the same as a charge freeze (but without being explicitly selected)._

_Freeze discharging - The battery is in Discharge mode, the same as Idle (Eco) mode, but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is a shortfall of solar power to meet house load, the battery will discharge._

Based upon those descriptions, I would anticipate:

- Inverter set to "Maximize Self Consumption" mode
- number.solaredge_i1_storage_charge_limit gets set to 0
- number.solaredge_i1_storage_discharge_limit gets set to 5000

From the log I shared, I can see it did something along those lines but in reverse, although at the time I had my 0 to 1 workaround in place, so value is 1 not 0.
` 08:55:40.273500 INFO pred_bat: Inverter 0 SOC: 6.02 kW 62 % Current charge rate 5000.0 w Current discharge rate 1.0 w Current power 4998.0 w Current voltage 52.0`

Discharge rate set to 1 / 0,  Charge rate = 5000. but since it also set "Charge from Solar Power and Grid" mode, it will charge the battery at 5kWh rate, and since it is in charge mode it is also covering any shortfall in power needs from the grid (PV generation - usage = shortfall). 

On SolarEdge equipment, if it wants to use the  "Charge from Solar Power and Grid" mode for holds/freezes, it must set the number.solaredge_i1_storage_charge_limit gets set to 0 (or 1) and number.solaredge_i1_storage_discharge_limit  to 5000.

If it wants to use the "Maximize Self Consumption" mode, then the number.solaredge_i1_storage_charge_limit gets set to 0 which will prioritize exports over charging.

I don't think the way it is working currently is correct, whic is why when a hold is attempting to implement it gets stuck in charging mode until it decides it wants an altenrate mode which will change the mode type

cyberkryten, 2024-08-01:
But why would charge limit = 0 for "Hold **charging**" which is defined as "A type of charge where..."?   

It isn't a full "Hold" (both charge/discharge set to zero) so some charging is expected, but it isn't entirely clear where from and the SoC being the same contradicts that and suggests no charging.  

I suspect it comes from managing the car charging cycle, so the intent is for the battery level to remain constant whilst house and car charging comes from the grid, so it is "Battery Hold, Car Charge from Grid" - if that is the case, then your suggestion works and from looking at inverter.py it does seem that the intent is to set current charge limits to allow that

In my logs, that's only occurred when SOC<Target:

`2024-07-31 20:45:08.606368 INFO pred_bat: Current SOC 94% is less than Target SOC 100. Grid charging enabled with charge current set to 65.00
`

65A @ 230V would be 14.95kW, so something isn't right!  This may not actually work with SE.

If you grep for "Current SOC" in logs, see what you have as all the messages about the control have that text.

Noodleyman, 2024-08-01:
I'm making a bunch of assumptions, some or most are likely incorrect. :) 

I assume charge limit = 0, beacuse it has reached or exceeded the SoC target, so no need to put more power into the battery for now. I can see this was the case from the log I shared earlier:
`2024-07-24 08:55:46.978091 INFO pred_bat: Adjust reserve to default as SOC 62 % is above target 59 % or charging active`

It does seem to be using the RESERVE option though to set the hold targets, which does make sense, but that would also not allow discharge from the battery. 
```
2024-07-24 08:55:46.978091 INFO pred_bat: Adjust reserve to default as SOC 62 % is above target 59 % or charging active
2024-07-24 08:55:46.993299 INFO pred_bat: Inverter 0 Current Reserve is 60.0 % and new target is 4.0 %
```
Although, whilst it is doing the above stuff it is still in the "Charge from Solar Power and Grid" mode, thus will continue to charge the battery from grid untl the battery is full (or mode changed, whichever comes first) at the solaredge_i1_storage_charge_limit  rate of 5kWh.

I assume that a reserve set at the target hold SoC%, with "Charge from Solar Power and Grid" mode enabled and number.solaredge_i1_storage_charge_limit gets set to 0 would also work as a hold.

Everything you've described with regards to car charging cycle seems logical to me.  From my earlier log and searching for "Current SOC"
`2024-07-24 08:55:45.765773 INFO pred_bat: Charge window will be: 2024-07-24 08:30:00+01:00 - 2024-07-24 09:00:00+01:00 - current soc 62 target 59`

I'll have a poke around the logs for  "Current SOC". I am currently using the UI to view the logs, via _:5051/aui/index.html#/logs_. Is there a better method to access them for further history?

Noodleyman, 2024-08-01:
adding some extra history covering the earlier logs time period:
[image]

The 0 to 1 was my workaround kicking in, (since removed)
[image]

cyberkryten, 2024-08-01:
I see your inverter resets regularly and then sets the charge/discharge values to 11400 when it comes back online too - I spoke to SE about that and 'it just does that'!

For the logs, I SSH direct to the HA instance, but if you're not comfortable with that stick with how you're doing it.

I'm not really sure of the logic that gets used for Hold Charge during the day, will see if I can figure anything out when the car next charges up.

I decided that the SE "Charge from Clipped" was no use to me as it is the opposite and will never use the battery during the daytime which meant when demand is more than solar (I have AC) it ends up using the grid for that.

Noodleyman, 2024-08-01:
It does indeed do the resets. I just assumed it's because it supports oversizing and 7.5kW peak. even if you set it higher it does not seem to violate the system limits so that's good :)

I'm fine with the SSH of the logs, I'm just not sure where the appropriate logs are stored so went with the UI option. 

I dont have an EV, so no car chaging requirements here, thus why I have those switches disabled in config.

The charge from clipped is vital for me, I am in a very rural area with bad infrastructure, thus have a 3.68kWh export rate limit. but I have oversized my system to a total of 36 panels, pulling about 15kWp. I use a lot of power in the day for a variety of servers being powered up (and portable AC unit) compared to a normal home user. given the export limit I ensure I have at least 10% battery in the morning (gotta cover that coffee making time!), and once at 10% flip to charge from clipped. it will cover the home usage, then export first. once exprting at 3.68 the rest goes into the battery which will be full by about 11AM-12AM as i've been generating about 11kWh or so. The system limits itself then. need to scrape out that extra bit of export in the morning. once battery hits 100% the read only flag gets removed and gives control back to PredBat.  If I had a higher export limit I would be doing it differently.  As I also have the backup module, I've setup weather events to trigger dynamic reserve of battery as well. right now we've a yellow thunderstorm alert, so it's set the reserve to 50% of the battery in case of a power cut. it's working well.

Noodleyman, 2024-08-20:
I've noted a repeat of the issue today and managed to catch it whilst it was happening.  I am wondering if I am perhaps just being too quick and it needs to wait for another processing cycle before adapting and changing the mode. 

plan from 10:25 which says limit to 79%
[image]

Predbat Status history:
[image]

Storage command mode history:
[image]

Log attached, too large to share inline. 
[Uploading predbat log.txt…]()

I note at 10:06 it had an error,
```
2024-08-20 10:05:48.607642 INFO pred_bat: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/services/number/set_value
2024-08-20 10:06:12.877519 INFO pred_bat: Warn: Inverter 0 Trying to write 0 to discharge_rate didn't complete got 5000
2024-08-20 10:06:12.926927 INFO pred_bat: Info: record_status Warn: Inverter 0 write to discharge_rate failed
```

It also put it into "Charge from Solar Power and Grid" mode. 

The thing that stands out with a "hold charging" mode, is that it instead sets the discharge rate = 0. Shouldn't that be the charge rate instead?

example: 10:11, predbat status is set to "hold charging"

Log shows:
`Inverter 0 SOC: 6.75 kW 70 % Current charge rate 5000.0 w Current discharge rate 0.0 w Current power 4999.0 w Current voltage 52.0`

which will actually not hold charging at all, instead it will prevent power being consumed from the battery.

am I just reading this all wrong? it seems like the issue is that it is setting disharge rate instead of charge rate for holds.

cyberkryten, 2024-08-20:
I think the wording is a little confusing if you read "Hold Charging" in isolation, but it references "Freeze Charging" and I think that explains the intent better:

`
Freeze charging - The battery is charging but the current battery level (SoC) is is frozen (held). Think of it as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar power above house load, the excess solar will be used to charge the battery,
Hold charging - A type of charge where the target SoC % is the same as the current SoC %, effectively the same as a charge freeze (but without being explicitly selected).`

The intent is for the battery to remain unused with house load satisfied from solar or grid so the SoC remains the same or increases if Solar>Load.

This is achieved by stopping discharge from the battery by setting the discharge limit to zero.

Noodleyman, 2024-08-20:
Kind of makes sense, although doesn't quite fit with what it is actually doing. I'll break it down based on the phrasing because it helps me understand if I am being a muppet :) .

**The battery is charging but the current battery level (SoC) is is frozen (held).**
It did put the battery into "Charge From Solar Power and Grid" mode. We do also see in the logs, Note: Inverter does not support discharge freeze - disabled, Note: Inverter does not support charge freeze - disabled. thus it shouldn't be trying to do these freeze methods.

**The grid or solar covers any house load.**
In this case, the solar was covering with 4kW excess.

**If there is a shortfall of Solar power to meet house load, the excess house load is met from grid import**
In this case, there was not a shortfall, there was an excess of around 4kWh.

**but if there is excess Solar power above house load, the excess solar will be used to charge the battery**
This was the case, and it was doing this, however because it also was in "Charge From Solar Power and Grid" mode, the shortfall between the 4Kwh into battery, and it's max charge rate of 5kWh was met from the grid. so it was pulling 1kWh from the grid as well.

If it was in either "Charge from Solar Power Only" mode, or "Maximize Self Consumption" mode, then it would have worked as per above, only putting the excess into the battery, the 4kWh

Noodleyman, 2024-10-11:
Thought I would follow up on this thread as I ran into this issue a few more times recently. I put in a new automation to workaround the issue, and that is now working correctly for me.  Sharing in case anybody else finds themselves wanting to use it:

```
alias: Solar - Workaround PredBat Hold Charge Bug
description: ""
trigger:
  - platform: state
    entity_id:
      - predbat.status
    to: Hold charging
    for:
      hours: 0
      minutes: 0
      seconds: 10
condition:
  - condition: state
    entity_id: select.solaredge_i1_storage_command_mode
    state: Charge from Solar Power and Grid
action:
  - action: select.select_option
    metadata: {}
    data:
      option: Maximize Self Consumption
    target:
      entity_id: select.solaredge_i1_storage_command_mode
mode: single
```

The above got triggered a couple of times today and did the job it was meant to :)

Noodleyman, 2024-10-15:
replacing previous shared code with a much better, and more fit for purpose set of automations which is doing a better job.

```
alias: Solar - Workaround PredBat Hold Charge Bug - rate to 10w
description: Set discharge rate to 10w when in HOLD mode
trigger:
  - platform: state
    entity_id:
      - predbat.status
    to: Hold charging
    for:
      hours: 0
      minutes: 0
      seconds: 5
  - platform: state
    entity_id:
      - number.solaredge_i1_storage_charge_limit
    to: "5000"
    for:
      hours: 0
      minutes: 0
      seconds: 8
  - platform: state
    entity_id:
      - number.solaredge_i1_storage_discharge_limit
    to: "5000"
    for:
      hours: 0
      minutes: 0
      seconds: 8
condition:
  - condition: state
    entity_id: select.solaredge_i1_storage_command_mode
    state: Charge from Solar Power and Grid
  - condition: state
    entity_id: predbat.status
    state: Hold charging
action:
  - action: number.set_value
    metadata: {}
    data:
      value: "10"
    target:
      entity_id: number.solaredge_i1_storage_discharge_limit
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - action: number.set_value
    metadata: {}
    data:
      value: "10"
    target:
      entity_id: number.solaredge_i1_storage_charge_limit
mode: single
```

```
alias: Solar - Workaround PredBat Hold Charge Bug - Restore Normal Values
description: Restore discharge rate to 5000w
trigger:
  - platform: state
    entity_id:
      - predbat.status
    from: Hold charging
    for:
      hours: 0
      minutes: 0
      seconds: 5
condition: []
action:
  - action: number.set_value
    metadata: {}
    data:
      value: "5000"
    target:
      entity_id: number.solaredge_i1_storage_discharge_limit
  - action: number.set_value
    metadata: {}
    data:
      value: "5000"
    target:
      entity_id: number.solaredge_i1_storage_charge_limit
mode: single
```

springfall2008, 2025-01-01:
As this ticket is old I'm going to close it, please re-open if the issue is there on latest.

gcoan, 2025-01-04:
let me know how it goes and I'll add it to the predbat documentation as a working example if everyone is happy with it

cyberkryten, 2025-01-08:
Just a note for anyone with multiple batteries, the SE Inverter itself does things with the individual battery settings so trying to do anything yourself is pretty much pointless.  The SE docs say different things in different places - some say they will either charge/discharge in sequence one battery after another but in my setup this doesn't happen.

Other places say they will all discharge together, the reality is they mostly discharge together, but that it's a bit random!  The batteries get out of step by as much as 20% at times and only a full charge will equalise that (a full discharge will not, as a 5% backup reserve can be met by one battery discharged completely, one at 5% and one at 10%).

As an example, I currently have two batteries at 100% and one at 97% and the sensor.solaredgemodbus_b1_peak_charge_power and sensor. solaredgemodbus_b2_peak_charge_power values are both zero as they are full, whereas sensor. solaredgemodbus_b3_peak_charge_power is at the normal figure of 5000

Because of this, with multiple batteries, you also cannot set a charge curve because Predbat only sees total charge and not the individual charge levels of each battery.

Noodleyman, 2025-01-08:
@cyberkryten, this is realllllly useful info. I'm considering a 2nd battery at some point as I want the extra capacity because we have frequent outages and the off grid module. a 2nd battery would tide us over well, and it means we won't need to touch the grid at all in summer.

cyberkryten, 2025-01-08:
I'd add that whilst the Backup Interface needs to be connected directly with RS485 you should absolutely NOT connect multiple batteries to the inverter using cables as the way they process the cabled RS485 commands is too slow to actually work!  I had mine connected like that for months (because cabled is always better than wireless, right!) and eventually traced the issues I had to the batteries only working when connected using the SE Wireless network.  Also, if you wire them, you can only use 9.5kWh instead of 9.7kWh (never got to the bottom of why that's a restriction!).

I wanted more but three is the most SE will support for a home installation, so I'm glad IOG adds daytime slots!

cyberkryten, 2025-01-10:
I know I've seen the Hold/Freeze behaviour before - I recall the charge/discharge rates being set to zero to allow this but I can't find it being used recently and also I don't have a service for it.

The only services I have in my apps.yaml are:

  charge_start_service:
  charge_stop_service:
  discharge_start_service:

I wonder if I've missed a requirement to add extra services for the SE Inverter types?  However, looking at the example YAML it seems to only include those three of the SE example, so doe the other services work for SE inverters?  If so then we've got Docs/Examples issues.

cyberkryten, 2025-01-10:
Yep, I'm missing all the config relating to freeze charge/discharge and don't recognise them at all (and they aren't in any backups).

Also, I don't have a discharge_stop_service defined so guess I need to add that as well

I'll add the freeze options to test and see what chaos ensues - I'll be honest and say I don't really understand when they'll be used!

cyberkryten, 2025-01-10:
Yep, I spotted that and added those too and it just went through a Freeze Charge cycle where it dropped storage_discharge_limit to zero, which it used to do ages ago.

cyberkryten, 2025-03-31:
Hmmm, now having some issues with this.

It seems to be setting charge_rate and discharge_rate both to zero and then never restoring them to the correct values.  I assume it's related to the clipping we're all currently trying to figure out.

However, the result yesterday was extreme clipping the entire day AND massive peak power usage from the grid whenever the AC kicked in, because the batteries were effectively turned off all day (both charge and discharge = 0 is equivalent to OFF)

I think what is happening is that, for some reason, when charge_rate is set to a value, Predbat then copies this into discharge_rate to set it to the same figure, which makes no sense as they shouldn't be tied together.

I've disabled freeze discharge:

`    support_discharge_freeze: False  `

And then manually set the charge_rate to 2000 in an attempt to manage clipping and on the next run, Predbat set discharge_rate to 2000 as well.

No idea why it does this?

```
2025-03-31 10:20:05.449421: Inverter 0 SOC: 22.7kW 78% Current charge rate 2000W Current discharge rate 11400W Current power 1998.0W Current voltage 52.0V
2025-03-31 10:20:06.172126: Inverter 0 current discharge rate is 11400.0W and new target is 2000W
```

cyberkryten, 2025-03-31:
I've confirmed that these two values are tied together by doing a couple of further tests - no matter what the discharge rate is set to, Predbat always overwrites the Discharge Rate value with whatever is currently set in Charge Rate.
 
Unfortunately, this makes any use of charge/discharge rate impossible on my system (SolarEdge) and those values are used to implement Hold/Freeze which is probably why I disabled the Freeze charge/discharge options last summer

I can't spot any obvious issues in the code - I've had a look through inverter.py and execute.py and can't find any cases where any of the discharge rate variables are being populated from charge rate ones, which is the obvious typo that would cause this.

Manually set Charge rate to 4000, Predbat changed Discharge rate to match:

```
2025-03-31 11:15:05.316192: Inverter 0 with soc_max 29.1 kWh nominal_capacity 29.1 kWh battery rate raw 2000.0 w charge rate 2.0 kW discharge rate 2.0 kW battery_rate_min 0.0 w ac limit 8.0 kW export limit 3.68 kW reserve 5.0 % current_reserve 5.0 % temperature 20 c
2025-03-31 11:15:05.316273: Inverter 0 SOC: 24.15kW 83% Current charge rate 2000W Current discharge rate 4000W Current power 1998.0W Current voltage 52.0V
2025-03-31 11:15:05.316462: Found 1 inverters totals: min reserve 1.46 current reserve 1.46 soc_max 29.1 soc 24.153 charge rate 2.0 kW discharge rate 4.0 kW battery_rate_min 0.0 w ac limit 8.0 export limit 3.68 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2025-03-31 11:15:06.041111: Inverter 0 current discharge rate is 4000.0W and new target is 2000W
2025-03-31 11:15:11.255219: Inverter 0 write_and_poll_value: Wrote 2000 to discharge_rate, successfully now 2000
2025-03-31 11:15:18.783697: Inverter 0 with soc_max 29.1 kWh nominal_capacity 29.1 kWh battery rate raw 2000.0 w charge rate 2.0 kW discharge rate 2.0 kW battery_rate_min 0.0 w ac limit 8.0 kW export limit 3.68 kW reserve 5.0 % current_reserve 5.0 % temperature 20 c
2025-03-31 11:15:18.783779: Inverter 0 SOC: 24.15kW 83% Current charge rate 2000W Current discharge rate 2000W Current power 1998.0W Current voltage 52.0V
2025-03-31 11:15:18.783976: Found 1 inverters totals: min reserve 1.46 current reserve 1.46 soc_max 29.1 soc 24.153 charge rate 2.0 kW discharge rate 2.0 kW battery_rate_min 0.0 w ac limit 8.0 export limit 3.68 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
```

Manually set Charge rate to 11400, Predbat changed Discharge rate to match:

```
2025-03-31 11:35:05.484446: Inverter 0 with soc_max 29.1 kWh nominal_capacity 29.1 kWh battery rate raw 11400.0 w charge rate 11.4 kW discharge rate 11.4 kW battery_rate_min 0.0 w ac limit 8.0 kW export limit 3.68 kW reserve 5.0 % current_reserve 5.0 % temperature 20 c
2025-03-31 11:35:05.484527: Inverter 0 SOC: 25.03kW 86% Current charge rate 11400W Current discharge rate 2000W Current power 3576.0W Current voltage 52.0V
2025-03-31 11:35:05.484722: Found 1 inverters totals: min reserve 1.46 current reserve 1.46 soc_max 29.1 soc 25.026 charge rate 11.4 kW discharge rate 2.0 kW battery_rate_min 0.0 w ac limit 8.0 export limit 3.68 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2025-03-31 11:35:06.227065: Inverter 0 current discharge rate is 2000.0W and new target is 11400W
2025-03-31 11:35:11.464322: Inverter 0 write_and_poll_value: Wrote 11400 to discharge_rate, successfully now 11400
2025-03-31 11:35:19.800068: Inverter 0 with soc_max 29.1 kWh nominal_capacity 29.1 kWh battery rate raw 11400.0 w charge rate 11.4 kW discharge rate 11.4 kW battery_rate_min 0.0 w ac limit 8.0 kW export limit 3.68 kW reserve 5.0 % current_reserve 5.0 % temperature 20 c
2025-03-31 11:35:19.800148: Inverter 0 SOC: 25.03kW 86% Current charge rate 11400W Current discharge rate 11400W Current power 3546.0W Current voltage 52.0V
2025-03-31 11:35:19.800345: Found 1 inverters totals: min reserve 1.46 current reserve 1.46 soc_max 29.1 soc 25.026 charge rate 11.4 kW discharge rate 11.4 kW battery_rate_min 0.0 w ac limit 8.0 export limit 3.68 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
```

cyberkryten, 2025-03-31:
After going through the execute.py code some more I found where it is getting the value for the max then traced that back into my config, which has been setup this way for ages, probably a year or more:

```
  battery_rate_max:
    #- sensor.solaredge_b1_max_charge_power
    # Use Inverter limit as can charge faster with three batteries
    - number.solaredgemodbus_i1_storage_charge_limit
  charge_rate:
    - number.solaredgemodbus_i1_storage_charge_limit
  discharge_rate:
    - number.solaredgemodbus_i1_storage_discharge_limit
```

I figured I could just re-use the inverter value instead of setting up a template sensor to provide the max charge rate for three batteries but, um, that doesn't work!  Someone pass the pointy had with the 'D' on the front...

I've now added a template sensor and will re-enable the freeze as I suspect it may work a little better now!

```
  battery_rate_max:
    # 31Mar25 NEW Sensor for total of three batteries Max Charge Power
    - sensor.semodbus_power_batteries_max_charge_power

```

chalfontchubby, 2026-09-09:
Thanks for the detailed digging on this one - the trace back to `battery_rate_max`/`charge_rate` both pointing at `number.solaredgemodbus_i1_storage_charge_limit` was useful.

This has had no activity since March 2025 and a lot has changed in the inverter layer since, so any diagnosis from back then is hard to apply to current code. We're working through a large backlog as volunteers and need to filter what's still live.

If you're still hitting this, let us know and we'll keep it open - ideally with a current version number and a `predbat_debug_*.yaml`. Otherwise we'll close it in about a week (around 2026-09-16).

_Posted by Claude on Rik's behalf._

chalfontchubby, 2026-09-15:
No action since we asked a week ago - closing. Please open a new issue of the symptom reoccurs.

closed 2026-09-15 by chalfontchubby