springfall2008/batpred #1329: overlapping discharge and charge time on Solis prevent scheduling to be applied

marcoaddario, 2024-07-24:
**Describe the bug**
On Solis inverter if the charge and discharge times are overlapping, the schedule is not applied and the modbus plugin is not returning any error

**Expected behavior**
Charge and discharge schedule must not overlap
**Predbat version**

8.3.1

**Environment details**

Solis controlled via modbus

full log run in attachment. 
this is the line showing the schedule time:

`2024-07-24 21:20:03.648673 INFO pred_bat: Adjust idle time, charge 21:30:00-07:00:00 discharge 21:00:00-21:31:00
`
but is should be:

discharge 21:00:00 to 21:30:00
charge 21:30:00 to 07:00:00

**Log file**
attached
[predbat.log](https://github.com/user-attachments/files/16367256/predbat.log)

springfall2008, 2024-08-06:
Please try v8.3.3 on 'main' currently for testing

marcoaddario, 2024-08-28:
it should be ok now. Thanks Trefor

closed 2024-08-28 by marcoaddario