springfall2008/batpred #1330: Entities not found in PV Forecast vs Actual Chart Template

DarrenJCraddock, 2024-07-25:
**Describe the bug**
I have created all the charts using the code from charts template and the apexcharts-card.
They all appear to be working apart from the PV Forecast vs Actual chart which has the following errors:
Entity not available: sensor.givtcp_sa2243g277_pv_power
Entity not available: sensor.givtcp_sa2243g277_energy_today_kwh

These look to be hardcoded to a specific inverter. All other graphs seem to be ok, so are there generic sensor names that should be used instead?

DarrenJCraddock, 2024-07-25:
I've changed for my specific inverter ID, but I'm wondering why this only needs to be done for this chart.

gcoan, 2024-07-25:
yes they should be changed for your specific inverter id

I'll change the charts to make this clearer

gcoan, 2024-08-19:
Closed with PR 1377

closed 2024-08-19 by gcoan