springfall2008/batpred #1331: Sunsynk Docs apps.yaml and automations

tmh88, 2024-07-26:
**apps.yaml example for Sunsynk inverters**

`---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.sunsynk_load_power_daily_2 # load_power_daily_kWh
  import_today:
    - sensor.smart_meter_electricity_import_today # electric_import_daily_kWh
  export_today:
    - sensor.smart_meter_electricity_total_export_daily # electric_export_daily_kWh
  pv_today:
    - sensor.sunsynk_total_pv_power_daily

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  inverter_type: SK

  # **** Start of Outputs to Sunsynk Inverter ****

  charge_start_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode" # Inverter work mode
    option: "Limited to Home"

  charge_stop_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode" # Inverter work mode
    option: "Limited to Home"

[... 355 lines cut ...]

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
  
  iboost_energy_today:
    - sensor.spa_heater_power_daily``

Meatballs1, 2024-08-20:
What are the  `test` values  e.g. `input_number.test_sunsynk_battery_max_charge_current` ?  

I currently have the following templates (slightly different naming for the BMS reported limitation):

```
Max Charge Rate: {{ [5000,[states('number.sunsynk_battery_max_charge_current')|int,states('sensor.sunsynk_battery_charge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
Max Discharge Rate: {{ [5000,[states('number.sunsynk_battery_max_discharge_current')|int, states('sensor.sunsynk_battery_discharge_bms_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
```

But not sure what the `Sunsynk Charge Rate Calc` refers to - are the input numbers just helpers to manually restrict it?

Meatballs1, 2024-08-20:
Ah I think I see, the charge_rate/discharge_rate should be the actual power the inverter can handle, the inverter_limit_charge could just be an input number to apply a limit.

tmh88, 2024-08-20:
Below is the code for the my app.yaml and automations used to integrate my Sunsynk 8.8Kw ECCO inverter, hopefully it helps others. Thanks to @springfall2008 for the code modifications along the way.

## apps.yaml

```yaml
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.sunsynk_load_power_daily_2 # load_power_daily_kWh
  import_today:
    - sensor.smart_meter_electricity_import_today # electric_import_daily_kWh
  export_today:
    - sensor.smart_meter_electricity_total_export_daily # electric_export_daily_kWh
  pv_today:
    - sensor.sunsynk_total_pv_power_daily

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  inverter_type: SK

  # **** Start of Outputs to Sunsynk Inverter ****

  charge_start_service:
    service: select.select_option
    entity_id: "select.sunsynk_work_mode" # Inverter work mode
    option: "Limited to Home"

  charge_stop_service:
    service: select.select_option
[... 490 lines cut ...]
input_number.sunsynk_battery_max_charge_current_limit = Setpoint in HA to set the charge limit in A

```yaml
{{ [8000,[states('input_number.sunsynk_battery_max_charge_current_limit')|int,states('sensor.sunsynk_battery_charge_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
```

### Sunsynk Battery Max Discharge Rate (Sets Discharge Limit)

input_number.sunsynk_battery_max_discharge_current_limit = Setpoint in HA to set the discharge limit in A

```yaml
{{ [8000,[states('input_number.sunsynk_battery_max_discharge_current_limit')|int, states('sensor.sunsynk_battery_discharge_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
```

### Sunsynk Charge Rate Calc (Feedback to Pedbat of actual Charge Limit)

```yaml
{{ [8000,[states('number.sunsynk_battery_max_charge_current')|int,states('sensor.sunsynk_battery_charge_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
```

### Sunsynk Discharge Rate Calc (Feedback to Pedbat of actual Discharge Limit)

```yaml
{{ [8000,[states('number.sunsynk_battery_max_discharge_current')|int,states('sensor.sunsynk_battery_discharge_limit_current')|int]|min * states('sensor.sunsynk_battery_voltage')|float]|min }}
```

tmh88, 2024-08-20:
@springfall2008 have I interpreted this correctly?
in the apps.yaml
inverter_limit_charge: & inverter_limit_discharge: are inputs to predbat to limit the charge and discharge power to the batteries

charge_rate: & discharge_rate: are inputs to predbat to tell predbat the actual charge/discharge power of the inverter

springfall2008, 2024-08-25:
Inverter limit charge/discharge are inputs that defined the maximum rate to set

charge_rate/discharge_rate are controls, Predbat can read them to find the current rate and write them to change the rate

Meatballs1, 2024-09-08:
> Inverter limit charge/discharge are inputs that defined the maximum rate to set
> 
> charge_rate/discharge_rate are controls, Predbat can read them to find the current rate and write them to change the rate

I think charge_rate  is ignored because ooutput_charge_control  is set to `current` for SK?

We have the `number.sunsynk_battery_max_charge_current` which are used in the timed_charge_current/discharge_current settings.

Meatballs1, 2024-09-14:
I had some issues with:

```
Resetting charging SOC as we are not within the window or charge is disabled and inverter_soc_reset is enabled (now 09-14 21:05:00 target set_soc_minutes 30 charge start time 09-14 23:30:00)
```

It kept putting my Timezone SOC back up to 100%, which was causing Idle to fail to cover the load. Disabled it in expert mode and will see how I go.

ryangbedford, 2024-10-16:
I've got a Sunsynk inverter, use Home assistant and have them linked via Solar Assistant, so have a variety of entities in HA ready to use. I've been trying to adapt the files etc above and rename various sensors, but am having trouble with a few. Is it actually possible to use Solar Assistant with my Sunsynk inverter to setup Predbat, or is the Kellerza integration the only way? Thanks.

ryangbedford, 2024-10-17:
With the predbat charge/discharge control automation. For the sunsynk.solar_sell switch, on the actual inverter is this the work_mode setting of selling first? From Solar Assistant, there is no solar_sell switch, but I do have a select entity that I can change to "selling first" or "Zero export to CT" which I think is the same as turning solar_sell on and off?

savethemanual, 2025-02-03:
@ryangbedford How did you get on with this? For discharge, you need to set the work mode to "selling first" as you say. I think the switch.sunsynk.solar_sell is wrong for discharging.

@tmh88 is this your most up to date apps.yaml?

I'm still figuring this out, but shouldn't the automations be?:

      # Enable Export (Discharge Start)
      - conditions:
          - condition: trigger
            id:
              - predbat_discharge_on
        sequence:
          - service: select.select_option
            target:
              entity_id: select.sunsynk_load_limit
            data:
              option: "Allow Export"

      # Disable Export (Discharge Stop)
      - conditions:
          - condition: trigger
            id:
              - predbat_discharge_off
        sequence:
          - service: select.select_option
            target:
              entity_id: select.sunsynk_load_limit
            data:
              option: "Zero Export"

Ba22a, 2025-03-27:
Hi, I also have a Sunsynk inverter, use Home assistant and have them linked via Solar Assistant. Has anyone managed to get the  apps.yaml, the templates and the automations working properly for batpred?

savethemanual, 2025-03-27:
> Hi, I also have a Sunsynk inverter, use Home assistant and have them linked via Solar Assistant. Has anyone managed to get the apps.yaml, the templates and the automations working properly for batpred?

Can you post what you have got already?

Ba22a, 2025-03-27:
What I have is now all over the place! I think it is best I start again! :-) A working Sunsynk through Solar Assistant example would be amazing!

Ba22a, 2025-03-27:
Here are my HA entities:

- sensor.sunsynk_grid_energy_in_cost: 2.31840000000001

- sensor.sunsynk_grid_energy_out_compensation: 0.0015

- update.sunsynk_power_flow_card_update: off

- number.sunsynk_grid_peak_shaving_power: 8000

- number.sunsynk_max_charge_current: 40.0

- number.sunsynk_battery_absorption_charge_voltage: 57.6

- number.sunsynk_capacity_point_3: 10

- number.sunsynk_battery_equalization_charge_voltage: 57.6

- number.sunsynk_voltage_point_6: 49.0

- number.sunsynk_generator_stop_capacity: 100

- number.sunsynk_grid_voltage_high: 273.7

- number.sunsynk_generator_peak_shaving_power: 8000

- number.sunsynk_voltage_point_1: 49.0

- number.sunsynk_battery_float_charge_voltage: 55.2

- number.sunsynk_max_grid_charge_current: 40.0

- number.sunsynk_max_discharge_current: 80.0

- number.sunsynk_voltage_point_2: 49.0

- number.sunsynk_capacity_point_2: 10

- number.sunsynk_capacity_point_4: 10

- number.sunsynk_output_shutdown_capacity: 5

- number.sunsynk_generator_stop_voltage: 54.0

- number.sunsynk_capacity_point_1: 50

- number.sunsynk_generator_start_capacity: 30

- number.sunsynk_voltage_point_3: 49.0

- number.sunsynk_start_battery_discharge_capacity: 50

- number.sunsynk_generator_start_voltage: 49.0

- number.sunsynk_capacity_point_5: 10

- number.sunsynk_grid_voltage_low: 184.0

- number.sunsynk_stop_battery_discharge_capacity: 10

- number.sunsynk_capacity_point_6: 10

- number.sunsynk_voltage_point_5: 49.0

- number.sunsynk_max_sell_power: 8000

- number.sunsynk_grid_frequency_high: 52.0

- number.sunsynk_voltage_point_4: 49.0

- number.sunsynk_max_generator_charge_current: 40.0

- number.sunsynk_grid_frequency_low: 47.0

- number.sunsynk_grid_trickle_feed: 0

- select.sunsynk_grid_charge: Enabled

- select.sunsynk_generator_connected_to_grid_input: Disabled

- select.sunsynk_generator_charge: Disabled

- select.sunsynk_auxiliary_load_output_on_grid_always_on: Disabled

- select.sunsynk_generator_peak_shaving: Disabled

- select.sunsynk_solar_export_when_battery_full: Disabled

- select.sunsynk_work_mode: Zero export to CT

- select.sunsynk_grid_peak_shaving: Disabled

- select.sunsynk_force_generator_on: Disabled

- select.sunsynk_energy_pattern: Load first

- sensor.sunsynk_load_power_essential: -66

- sensor.sunsynk_grid_frequency: 49.96

- sensor.sunsynk_pv_current_1: 0.0

- sensor.sunsynk_grid_power_ct: 0

- sensor.sunsynk_pv_power: 52

- sensor.sunsynk_battery_voltage: 51.5

- sensor.sunsynk_pv_current_2: 0.1

- sensor.sunsynk_serial_number: 2408151661

- sensor.sunsynk_temperature: 39.9

- sensor.sunsynk_load_percentage: 15

- sensor.sunsynk_battery_current: -23.9

- sensor.sunsynk_grid_power: 0

- sensor.sunsynk_pv_voltage_1: 187.2

- sensor.sunsynk_pv_voltage_2: 252.3

- sensor.sunsynk_pv_power_1: 18

- sensor.sunsynk_device_mode: Discharge above 10%

- sensor.sunsynk_grid_voltage: 244.9

- sensor.sunsynk_generator_power: 0

- sensor.sunsynk_ac_output_frequency: 49.96

- sensor.sunsynk_grid_power_ld: -1282

- sensor.sunsynk_load_power_non_essential: 1282

- sensor.sunsynk_ac_output_voltage: 245.0

- sensor.sunsynk_load_power: 1216

- sensor.sunsynk_pv_power_2: 34

- sensor.sunsynk_battery_power: -1227

- sensor.sunsynk_battery_state_of_charge: 25

- sensor.sunsynk_battery_temperature: 17.0

- sensor.sunsynk_response: unknown

- sensor.sunsynk_battery_energy_in: 7.6

- sensor.sunsynk_battery_energy_out: 5.39

- sensor.sunsynk_grid_energy_in: 10.08

- sensor.sunsynk_grid_energy_out: 0.01

- sensor.sunsynk_load_energy: 16.79

- sensor.sunsynk_pv_energy: 9.99

- switch.sunsynk_grid_charge_point_2: off

- switch.sunsynk_grid_charge_point_4: off

- switch.sunsynk_use_timer: on

- switch.sunsynk_gen_charge_point_4: off

- switch.sunsynk_gen_charge_point_5: off

- switch.sunsynk_grid_charge_point_3: off

- switch.sunsynk_grid_charge_point_5: off

- switch.sunsynk_gen_charge_point_6: off

- switch.sunsynk_gen_charge_point_2: off

- switch.sunsynk_grid_charge_point_1: on

- switch.sunsynk_grid_charge_point_6: off

- switch.sunsynk_gen_charge_point_1: off

- switch.sunsynk_gen_charge_point_3: off

- sensor.sunsynk_battery_max_charge_rate: unavailable

- sensor.sunsynk_battery_max_discharge_rate: unavailable

- sensor.sunsynk_charge_rate_calc: unavailable

- sensor.sunsynk_discharge_rate_calc: unavailable

Ba22a, 2025-03-27:
So this is what I have now for the apps.yaml, It is the 4 templates I am struggling to workout the values for:

pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # XXX: Template configuration, delete this line once you have set up for your system
#  template: True

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  inverter_type: "SA"
  num_inverters: 1
  #
  # Controls/status - must by 1 per inverter
  #

  # Max inverter power from battery
  battery_rate_max:
    - 8000
  # Battery capacity in kWh
  soc_max:
    - 10.7

  # Solar Assistant specific parameters (This is for a Growatt inverter configured, change to yours):

  load_today:
    - sensor.sunsynk_load_energy
  import_today:
    - sensor.sunsynk_grid_energy_in
  export_today:
    - sensor.sunsynk_grid_energy_out
  pv_today:
    - sensor.sunsynk_pv_energy
  battery_voltage:
    - sensor.sunsynk_battery_voltage

  # This is disabled by default in the Solax integration so it must be manually enabled.
  inverter_time:
    - sensor.date_time

  # This is disabled by default in the Solax integration so it must be manually enabled.
  battery_power:
    - sensor.sunsynk_battery_power
  pv_power:
    - sensor.sunsynk_pv_power
  soc_percent:
    - sensor.sunsynk_battery_state_of_charge

[... 251 lines cut ...]
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
    - name: 'large'
      minutes: 60
      energy: 1.0
    - name: 'small'
      minutes: 15
      energy: 0.25

  # Nordpool market energy rates
  #futurerate_url: 'https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP'
  #futurerate_adjust_import: False
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0

savethemanual, 2025-03-28:
You need to use the correct apps.yaml template for Sunsynk inverters. It seems you're using Solax...

This is mine:
`# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Template configuration - delete this next line once you edit modified it XXXX:
  #template: True

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.sunsynk_day_load_energy # load_power_daily_kWh
  import_today:
    - sensor.smart_meter_electricity_import_today # electric_import_daily_kWh
  export_today:
    - sensor.sunsynk_day_grid_export # electric_export_daily_kWh
  pv_today:
    - sensor.sunsynk_day_pv_energy

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  inverter_type: SK

  # **** Start of Outputs to Sunsynk Inverter ****

  charge_start_service:
    service: select.select_option
    entity_id: "select.sunsynk_load_limit" # Inverter work mode
[... 526 lines cut ...]

- sensor:
    - name: "sunsynk_battery_max_discharge_rate"
      unit_of_measurement: "w"
      state_class: measurement
      state: >
            {{ [8800,[states('input_number.battery_max_discharge_current_limit')|int,states('number.sunsynk_battery_max_discharge_current')|int]|min
            * states('sensor.sunsynk_battery_voltage')|float]|min }}
            
- sensor:
    - name: "sunsynk_charge_rate_calc"
      unit_of_measurement: "w"
      state_class: measurement
      state: >
            {{ [8800,[states('input_number.battery_max_charge_current_limit')|int,states('number.sunsynk_battery_max_charge_current')|int]|min
            * states('sensor.sunsynk_battery_voltage')|float]|min }}

- sensor:
    - name: "sunsynk_discharge_rate_calc"
      unit_of_measurement: "w"
      state_class: measurement
      state: >
            {{ [8800,[states('input_number.battery_max_discharge_current_limit')|int,states('number.sunsynk_battery_max_discharge_current')|int]|min
            * states('sensor.sunsynk_battery_voltage')|float]|min }}
            `

markvk42, 2025-03-28:
i have a working config now, took some figuring out! what i dint realise for ages is that i had to write automations to control the inverter and point the yaml to them.

Ba22a, 2025-03-28:
I am using Solar Assistant to get the data from SunSynk Inverter to Home Assistant so I have slightly different entities and controls than SunSynk via a modbus bridge
,

markvk42, 2025-03-28:
yes so am i so its exactly the same.

Ba22a, 2025-03-28:
Amazing! Can you share them please, thanks

markvk42, 2025-03-28:
what exactly do you want?

Ba22a, 2025-03-28:
Apps.yaml, automations yaml and template code.

markvk42, 2025-03-28:
dont know how to do that on here, join the predbat fb group and ill send them to you.

chalfontchubby, 2026-08-20:
Closing - this looks resolved by #1357. Please reopen if it's still a problem.

closed 2026-08-20 by chalfontchubby