springfall2008/batpred #1332: Hauwei yaml config issues

UKHarj, 2024-07-27:
I have followed the add on method however I cannot get to the stage where prebat creates a dashboard entry in the log. 

My inverter is the SUN2000 with 10 kWh battery.

2024-07-27 23:00:00.273591: --------------- PredBat - update at 2024-07-27 23:00:00+01:00 with clock skew 0 minutes, minutes now 1380
2024-07-27 23:00:00.273851: Saved current settings to /config/predbat_config.json
2024-07-27 23:00:00.273895: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 4.2 minutes
2024-07-27 23:00:00.273923: Predbat /config/predbat.py version v8.3.1 currently running, latest version is v8.3.1 latest beta v8.3.1
2024-07-27 23:00:00.273945: Autoupdate: Currently up to date
2024-07-27 23:00:00.294476: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-07-27 23:00:00.294738: Predbat mode is set to Monitor
2024-07-27 23:00:00.400916: Found 11520 load_today datapoints going back 0 days
2024-07-27 23:00:00.703475: Car charging hold True threshold 6.0
2024-07-27 23:00:00.703526: Current data so far today: load 0.3 kWh import 0.79 kWh export 0.23 kWh pv 15.53 kWh
2024-07-27 23:00:00.703537: Fetching futurerate data from None
2024-07-27 23:00:00.715791: Standing charge is set to 52.31099999999999 p
2024-07-27 23:00:00.949327: Rate min forward looking: now 13.69 at end of forecast 13.69
2024-07-27 23:00:00.949458: Import rates min 13.69 max 32.22 average 18.29
2024-07-27 23:00:00.954943: Export rates min 15.0 max 15.0 average 15.0
2024-07-27 23:00:00.954986: Rate thresholds (for charge/discharge) are import 31.72p (0.0) export 15.0p (0.0)
2024-07-27 23:00:00.958137: High export rate found rates in range 15.0 to 15.0
2024-07-27 23:00:00.960876: Low Import rate found rates in range 13.69 to 29.27
2024-07-27 23:00:00.999024: Today's energy import 0.79 kWh export 0.23 kWh cost 63.23 p import 66.75 p export -3.52 p carbon 0.0 kg
2024-07-27 23:00:00.999077: Using solcast from inside HA
2024-07-27 23:00:00.999118: PV Data for pv_forecast_today total 28.27 kWh
2024-07-27 23:00:00.999145: PV Data for pv_forecast_tomorrow total 53.66 kWh
2024-07-27 23:00:00.999169: PV Data for pv_forecast_d3 total 54.23 kWh
2024-07-27 23:00:00.999195: PV Data for pv_forecast_d4 total 49.06 kWh
2024-07-27 23:00:01.002418: PV Forecast for today is 28.27 (13.42 10% 45.42 90%) kWh and left today is 0.0 (0.0 10% 0.0 90%) kWh
2024-07-27 23:00:01.012818: PV Forecast for day tomorrow is 53.66 (41.9 10% 55.28 90%) kWh
2024-07-27 23:00:01.023188: PV Forecast for day d2 is 54.23 (16.89 10% 54.61 90%) kWh
2024-07-27 23:00:01.033502: PV Forecast for day d3 is 49.06 (14.31 10% 53.39 90%) kWh
2024-07-27 23:00:01.057865: Historical data totals for days [7] are [0.3] - min 0.3
2024-07-27 23:00:01.058000: Warn: Historical day 7 has 1340 minutes of gap in the data, filled from 0.3 kWh to make new average 4.26 kWh (percent 7%)
2024-07-27 23:00:01.064540: Historical days now [7] weight [1]
2024-07-27 23:00:01.075992: Today's load divergence 100.0 % in-day adjustment 100.0 % damping 0.95x
2024-07-27 23:00:01.076012: Today's predicted so far 0 kWh with 0.0 kWh car/iBoost excluded and 0 kWh import ignored and 0 forecast extra.
2024-07-27 23:00:01.076023: Today's actual load so far 0.93 kWh with 0.0 kWh Car/iBoost excluded and 0.02 kWh import ignored.
2024-07-27 23:00:01.120045: Inverter 0: Type HU Huawei Solar
2024-07-27 23:00:01.120086: Warn: Inverter 0: Using inverter type HU - not all features are available
2024-07-27 23:00:01.120198: Reserve min: 14.0% Battery_min:14.0%
2024-07-27 23:00:01.120236: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 5000.0 w charge rate 5.0 kW discharge rate 5.0 kW battery_rate_min 0.0 w ac limit 6.0 kW export limit 6.0 kW reserve 14.0 % current_reserve 14.0 %
2024-07-27 23:00:01.120340: Inverter 0 scheduled_charge_enable off scheduled_discharge_enable off
2024-07-27 23:00:01.120409: Inverter 0 SOC: 0.0 kW 0 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 0.0 w Current voltage 52.0
2024-07-27 23:00:01.120482: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
2024-07-27 23:00:01.120492: Inverter 0 scheduled_charge_enable set to False
2024-07-27 23:00:01.120521: Inverter 0 scheduled charge enable is False
2024-07-27 23:00:01.120529: Inverter 0 charge windows currently []
2024-07-27 23:00:01.120539: Inverter 0 Charge settings: timed charged is disabled, power 2.6 kW
2024-07-27 23:00:01.120584: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-07-27 23:00:01.120593: Inverter 0 sensor.predbat_HU_0_scheduled_discharge_enable set to False
2024-07-27 23:00:01.120603: Inverter 0 scheduled discharge enable is False
2024-07-27 23:00:01.120619: Inverter 0 discharge windows currently [{'start': 1439, 'end': 1439, 'average': 0}, {'start': 2879, 'end': 2879, 'average': 0}]
2024-07-27 23:00:01.120637: Note: Inverter does not support discharge freeze - disabled
2024-07-27 23:00:01.120643: Note: Inverter does not support charge freeze - disabled
2024-07-27 23:00:01.120649: Note: Inverter does not support reserve - disabling reserve functions
2024-07-27 23:00:01.120673: Found 1 inverters totals: min reserve 1.4 current reserve 1.4 soc_max 10.0 soc 0.0 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 6.0 export limit 6.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-07-27 23:00:01.164129: Base charge    window [  ]
2024-07-27 23:00:01.164218: Base discharge window [ 27-07 23:59:00 - 27-07 23:59:00 @ 0p 100.0%, 28-07 23:59:00 - 28-07 23:59:00 @ 0p 100.0% ]
2024-07-27 23:00:01.164239: Plan was last updated on 2024-07-27 22:55:00+01:00 and is now 5.0 minutes old
2024-07-27 23:00:01.164255: Current discharge window has expired, removing it
2024-07-27 23:00:01.164265: Best charge    window [  ]
2024-07-27 23:00:01.165108: Best discharge window [ 27-07 23:00:00 - 27-07 23:30:00 @ 15.0p 100%, 27-07 23:30:00 - 28-07 00:00:00 @ 15.0p 100%, 28-07 00:00:00 - 28-07 00:30:00 @ 15.0p 100%, 28-07 00:30:00 - 28-07 01:00:00 @ 15.0p 100%, 28-07 01:00:00 - 28-07 01:30:00 @ 15.0p 100%, 28-07 01:30:00 - 28-07 02:00:00 @ 15.0p 100%, 28-07 02:00:00 - 28-07 02:30:00 @ 15.0p 100%, 28-07 02:30:00 - 28-07 03:00:00 @ 15.0p 100%, 28-07 03:00:00 - 28-07 03:30:00 @ 15.0p 100%, 28-07 03:30:00 - 28-07 04:00:00 @ 15.0p 100%, 28-07 04:00:00 - 28-07 04:30:00 @ 15.0p 100%, 28-07 04:30:00 - 28-07 05:00:00 @ 15.0p 100%, 28-07 05:00:00 - 28-07 05:30:00 @ 15.0p 100%, 28-07 05:30:00 - 28-07 06:00:00 @ 15.0p 100%, 28-07 06:00:00 - 28-07 06:30:00 @ 15.0p 100%, 28-07 06:30:00 - 28-07 07:00:00 @ 15.0p 100%, 28-07 07:00:00 - 28-07 07:30:00 @ 15.0p 100%, 28-07 07:30:00 - 28-07 08:00:00 @ 15.0p 100%, 28-07 08:00:00 - 28-07 08:30:00 @ 15.0p 100%, 28-07 08:30:00 - 28-07 09:00:00 @ 15.0p 100%, 28-07 09:00:00 - 28-07 09:30:00 @ 15.0p 100%, 28-07 09:30:00 - 28-07 10:00:00 @ 15.0p 100%, 28-07 10:00:00 - 28-07 10:30:00 @ 15.0p 100%, 28-07 10:30:00 - 28-07 11:00:00 @ 15.0p 100%, 28-07 11:00:00 - 28-07 11:30:00 @ 15.0p 100%, 28-07 11:30:00 - 28-07 12:00:00 @ 15.0p 100%, 28-07 12:00:00 - 28-07 12:30:00 @ 15.0p 100%, 28-07 12:30:00 - 28-07 13:00:00 @ 15.0p 100%, 28-07 13:00:00 - 28-07 13:30:00 @ 15.0p 100%, 28-07 13:30:00 - 28-07 14:00:00 @ 15.0p 100%, 28-07 14:00:00 - 28-07 14:30:00 @ 15.0p 100%, 28-07 14:30:00 - 28-07 15:00:00 @ 15.0p 100%, 28-07 15:00:00 - 28-07 15:30:00 @ 15.0p 100%, 28-07 15:30:00 - 28-07 16:00:00 @ 15.0p 100%, 28-07 16:00:00 - 28-07 16:30:00 @ 15.0p 100%, 28-07 16:30:00 - 28-07 17:00:00 @ 15.0p 100%, 28-07 17:00:00 - 28-07 17:30:00 @ 15.0p 100%, 28-07 17:30:00 - 28-07 18:00:00 @ 15.0p 100%, 28-07 18:00:00 - 28-07 18:30:00 @ 15.0p 100%, 28-07 18:30:00 - 28-07 19:00:00 @ 15.0p 100%, 28-07 19:00:00 - 28-07 19:30:00 @ 15.0p 100%, 28-07 19:30:00 - 28-07 20:00:00 @ 15.0p 100%, 28-07 20:00:00 - 28-07 20:30:00 @ 15.0p 100%, 28-07 20:30:00 - 28-07 21:00:00 @ 15.0p 100%, 28-07 21:00:00 - 28-07 21:30:00 @ 15.0p 100%, 28-07 21:30:00 - 28-07 22:00:00 @ 15.0p 100%, 28-07 22:00:00 - 28-07 22:30:00 @ 15.0p 100%, 28-07 22:30:00 - 28-07 23:00:00 @ 15.0p 100%, 28-07 23:00:00 - 28-07 23:30:00 @ 15.0p 100%, 28-07 23:30:00 - 29-07 00:00:00 @ 15.0p 100%, 29-07 00:00:00 - 29-07 00:30:00 @ 15.0p 100%, 29-07 00:30:00 - 29-07 01:00:00 @ 15.0p 100%, 29-07 01:00:00 - 29-07 01:30:00 @ 15.0p 100%, 29-07 01:30:00 - 29-07 02:00:00 @ 15.0p 100%, 29-07 02:00:00 - 29-07 02:30:00 @ 15.0p 100%, 29-07 02:30:00 - 29-07 03:00:00 @ 15.0 [... cut ...]
  File "/config/inverter.py", line 1133, in adjust_discharge_rate
    self.write_and_poll_value(
  File "/config/inverter.py", line 1240, in write_and_poll_value
    matched = abs(float(current_state) - new_value) <= fuzzy
TypeError: float() argument must be a string or a real number, not 'NoneType'

2024-07-27 23:00:02.236941: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
2024-07-27 23:00:02.237009: Error: float() argument must be a string or a real number, not 'NoneType'

gcoan, 2024-07-28:
what is battery_rate_max_discharge set to in apps.yaml?

the error suggests that this isn't set

UKHarj, 2024-07-28:
I'm sure I have set the battery max discharge to 5000. Below is the latest log and yaml config

2024-07-28 19:55:00.568507: --------------- PredBat - update at 2024-07-28 19:55:00+01:00 with clock skew 0 minutes, minutes now 1195
2024-07-28 19:55:00.568791: Saved current settings to /config/predbat_config.json
2024-07-28 19:55:00.568834: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 3.2 minutes
2024-07-28 19:55:00.568863: Predbat /config/predbat.py version v8.3.1 currently running, latest version is v8.3.1 latest beta v8.3.1
2024-07-28 19:55:00.568890: Autoupdate: Currently up to date
2024-07-28 19:55:00.589954: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-07-28 19:55:00.590140: Predbat mode is set to Monitor
2024-07-28 19:55:00.914270: Found 11520 load_today datapoints going back 0 days
2024-07-28 19:55:01.422402: Car charging hold True threshold 6.0
2024-07-28 19:55:01.422456: Current data so far today: load 24.89 kWh import 3.22 kWh export 28.45 kWh pv 41.14 kWh
2024-07-28 19:55:01.422469: Fetching futurerate data from None
2024-07-28 19:55:01.435036: Standing charge is set to 52.31099999999999 p
2024-07-28 19:55:01.667823: Rate min forward looking: now 11.97 at end of forecast 11.97
2024-07-28 19:55:01.667952: Import rates min 11.97 max 32.69 average 17.47
2024-07-28 19:55:01.673361: Export rates min 15.0 max 15.0 average 15.0
2024-07-28 19:55:01.673384: Rate thresholds (for charge/discharge) are import 32.19p (0.0) export 15.0p (0.0)
2024-07-28 19:55:01.676371: High export rate found rates in range 15.0 to 15.0
2024-07-28 19:55:01.679059: Low Import rate found rates in range 11.97 to 31.12
2024-07-28 19:55:01.717553: Today's energy import 3.22 kWh export 28.45 kWh cost -316.39 p import 110.34 p export -426.73 p carbon 0.0 kg
2024-07-28 19:55:01.717607: Using solcast from inside HA
2024-07-28 19:55:01.717652: PV Data for pv_forecast_today total 54.93 kWh
2024-07-28 19:55:01.717682: PV Data for pv_forecast_tomorrow total 53.19 kWh
2024-07-28 19:55:01.717708: PV Data for pv_forecast_d3 total 48.63 kWh
2024-07-28 19:55:01.717734: PV Data for pv_forecast_d4 total 40.02 kWh
2024-07-28 19:55:01.720891: PV Forecast for today is 54.93 (53.36 10% 54.96 90%) kWh and left today is 0.14 (0.13 10% 0.14 90%) kWh
2024-07-28 19:55:01.731148: PV Forecast for day tomorrow is 53.19 (26.86 10% 53.99 90%) kWh
2024-07-28 19:55:01.741277: PV Forecast for day d2 is 48.63 (19.5 10% 52.92 90%) kWh
2024-07-28 19:55:01.751494: PV Forecast for day d3 is 40.02 (8.45 10% 48.6 90%) kWh
2024-07-28 19:55:01.775425: Historical data totals for days [7] are [15.41] - min 15.41
2024-07-28 19:55:01.775551: Warn: Historical day 7 has 625 minutes of gap in the data, filled from 15.41 kWh to make new average 27.22 kWh (percent 57%)
2024-07-28 19:55:01.779363: Historical days now [7] weight [1]
2024-07-28 19:55:01.790464: Today's load divergence 181.78 % in-day adjustment 177.69 % damping 0.95x
2024-07-28 19:55:01.790495: Today's predicted so far 0 kWh with 0.0 kWh car/iBoost excluded and 0 kWh import ignored and 0 forecast extra.
2024-07-28 19:55:01.790507: Today's actual load so far 14.42 kWh with 12.26 kWh Car/iBoost excluded and 0.19 kWh import ignored.
2024-07-28 19:55:01.833786: Inverter 0: Type HU Huawei Solar
2024-07-28 19:55:01.833825: Warn: Inverter 0: Using inverter type HU - not all features are available
2024-07-28 19:55:01.833921: Reserve min: 14.0% Battery_min:14.0%
2024-07-28 19:55:01.833962: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 5000.0 w charge rate 5.0 kW discharge rate 5.0 kW battery_rate_min 0.0 w ac limit 6.0 kW export limit 6.0 kW reserve 14.0 % current_reserve 14.0 %
2024-07-28 19:55:01.834042: Inverter 0 scheduled_charge_enable off scheduled_discharge_enable off
2024-07-28 19:55:01.834091: Inverter 0 SOC: 0.0 kW 0 % Current charge rate 2600.0 w Current discharge rate 2600.0 w Current power 0.0 w Current voltage 52.0
2024-07-28 19:55:01.834164: Inverter 0 Wrote scheduled_charge_enable to False successfully and got off
2024-07-28 19:55:01.834174: Inverter 0 scheduled_charge_enable set to False
2024-07-28 19:55:01.834202: Inverter 0 scheduled charge enable is False
2024-07-28 19:55:01.834211: Inverter 0 charge windows currently []
2024-07-28 19:55:01.834220: Inverter 0 Charge settings: timed charged is disabled, power 2.6 kW
2024-07-28 19:55:01.834265: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-07-28 19:55:01.834274: Inverter 0 sensor.predbat_HU_0_scheduled_discharge_enable set to False
2024-07-28 19:55:01.834284: Inverter 0 scheduled discharge enable is False
2024-07-28 19:55:01.834300: Inverter 0 discharge windows currently [{'start': 1439, 'end': 1439, 'average': 0}, {'start': 2879, 'end': 2879, 'average': 0}]
2024-07-28 19:55:01.834309: Note: Inverter does not support discharge freeze - disabled
2024-07-28 19:55:01.834316: Note: Inverter does not support charge freeze - disabled
2024-07-28 19:55:01.834321: Note: Inverter does not support reserve - disabling reserve functions
2024-07-28 19:55:01.834345: Found 1 inverters totals: min reserve 1.4 current reserve 1.4 soc_max 10.0 soc 0.0 charge rate 2.6 kW discharge rate 2.6 kW battery_rate_min 0.0 w ac limit 6.0 export limit 6.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-07-28 19:55:01.876920: Base charge    window [  ]
[... cut ...]
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.tasmota_energy_today'
  #iboost_energy_scaling: 1.0 `

JohanAlvedal, 2024-07-29:
Hello 
I also have a Huawei with a 10 kwh battery. I got everything working but you have to make some own sensors, as huawei measures everything in W not kwh as prdebat wants. 

Right now, charge, discharge, etc. don't seem to work.

```
`2024-07-29 19:50:31.495221: Inverter 0 Adjust force discharge to True, change times from 19:30:00 - 21:01:00 to 19:30:00 - 21:01:00
2024-07-29 19:50:31.495238: Adjust idle time, charge 23:59:00-23:59:00 discharge 19:30:00-21:01:00
2024-07-29 19:50:31.495341: Adjust idle time computed idle is 21:01:00-23:59:00
2024-07-29 19:50:31.495391: Inverter 0 Call service template discharge_start_service = huawei_solar/forcible_discharge_soc
2024-07-29 19:50:31.495408: Inverter 0 Call service discharge_start_service with data {'device_id': '6893ee838a0972bd3e50d9e0faa0ba5a', 'target_soc': 92, 'power': 4200}
2024-07-29 19:50:31.500542: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/services/discharge_start_service
2024-07-29 19:50:31.500715: Inverter 0 Current Target SOC is 92 already at target
2024-07-29 19:50:31.558516: Completed run status Discharging
2024-07-29 19:50:31.569479: Info: record_status Discharging target 98%-92%
2024-07-29 19:50:31.580521: Saved current settings to /config/predbat_config.json`
```

UKHarj, 2024-07-29:
Hi 

I have now managed to get the dashboard working. The changed needed was 'battery' to 'batteries' in the default template. I now have the same log issue as JohanAlvedal, discharge and charge don't work when I manually try to override them. 

2024-07-29 19:25:03.429117: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00
2024-07-29 19:25:03.429130: Adjust idle time, charge 22:30:00-06:00:00 discharge 00:00:00-00:00:00
2024-07-29 19:25:03.429230: Adjust idle time computed idle is 19:25:00-22:30:00
2024-07-29 19:25:03.429249: Inverter 0 Call service template charge_stop_service = huawei_solar.stop_forcible_charge
2024-07-29 19:25:03.429260: Inverter 0 Call service charge_stop_service with data {'device_id': '7fda8868d4d563a807adfd5a363777f9'}
2024-07-29 19:25:03.434327: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/services/charge_stop_service
2024-07-29 19:25:03.454148: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-07-29 19:25:03.492786: Completed run status Idle
2024-07-29 19:25:03.503612: Info: record_status Idle
2024-07-29 19:25:03.515368: Saved current settings to /config/predbat_config.json

UKHarj, 2024-07-29:
I've looked into this a little further and feel as though the issue could be with the way the power is specified for the service

2024-07-29 22:02:11.648255: Inverter 0 Call service charge_start_service with data {'device_id': '2669d9aa48778272230354f5fdde33a3', 'target_soc': 100, 'power': 3000}
2024-07-29 22:02:11.653325: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/services/charge_start_service
2024-07-29 22:02:11.653528: Configuring charge window now (now 07-29 22:00:00 target set_window_minutes 30 charge start time 07-29 22:00:00
2024-07-29 22:02:11.653602: Adjust idle time, charge 22:00:00-22:30:00 discharge 00:00:00-00:00:00
2024-07-29 22:02:11.653693: Adjust idle time computed idle is 22:30:00-23:59:00
2024-07-29 22:02:11.653709: Setting ECO mode as no discharge window planned
2024-07-29 22:02:11.653752: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00
2024-07-29 22:02:11.653762: Adjust idle time, charge 22:00:00-22:30:00 discharge 00:00:00-00:00:00
2024-07-29 22:02:11.653817: Adjust idle time computed idle is 22:30:00-23:59:00
2024-07-29 22:02:11.653841: Inverter 0 Current Target SOC is 100 already at target
2024-07-29 22:02:11.653865: Current SOC 85% is less than Target SOC 100. Grid charging enabled with charge current set to 65.00
2024-07-29 22:02:11.713084: Completed run status Charging
2024-07-29 22:02:11.724085: Info: record_status Charging target 85%-100%
2024-07-29 22:02:11.735083: Saved current settings to /config/predbat_config.json
2024-07-29 22:02:11.735414: Creating predbat dashboard at /config/predbat_dashboard.yaml

I think instead of inverter ID the battery ID is required as the service description in the screenshot.
[screenshot.docx](https://github.com/user-attachments/files/16419185/screenshot.docx)

I have switched the battery ID and even lowered the power from 5000 to 3000 (in the UK I'm capped at 3000 for AC charging into the battery). However it still doesnt work, instead of charging the battery, I get charge from the grid into the house, which is a small step forward.

UKHarj, 2024-07-31:
Hi just to summarise. I have tested all 3 services (charge/discharge/stop) in developer tools. The force charge and discharge required a battery device ID, SOC and Power. When I enter these items into developer tools both the force charge and discharge entities give a green tick, and then the stop service works fine too, it need only the device and not SOC or Power. 

So when I update my my yaml with the battery ID device I still get the following error:

2024-07-31 16:08:08.193509: Inverter 0 Call service template charge_stop_service = huawei_solar.stop_forcible_charge
2024-07-31 16:08:08.193521: Inverter 0 Call service charge_stop_service with data {'device_id': '2669d9aa48778272230354f5fdde33a3'}
2024-07-31 16:08:08.198648: Warn: Failed to decode response <Response [404]> from http://supervisor/core/api/services/charge_stop_service

Note I have removed the slash that was is in the service for the Hauwei template yaml (huawei_solar/stop_forcible_charge) I got from github as I thought this may have been the source of the error but unfortunately it wasn't!

JohanAlvedal, 2024-07-31:
Or in time
service: huawei_solar.forcible_discharge
data:
  duration: 60
  power: "1000"
  device_id: 6893ee838a0972bd3e50d9e0faa0ba5a

service: huawei_solar.forcible_charge
data:
  device_id: 6893ee838a0972bd3e50d9e0faa0ba5a
  duration: 60
  power: "1000"

Or SOC target 
service: huawei_solar.forcible_discharge_soc
data:
  device_id: 6893ee838a0972bd3e50d9e0faa0ba5a
  target_soc: 12
  power: "1000"

service: huawei_solar.forcible_charge_soc
data:
  device_id: 6893ee838a0972bd3e50d9e0faa0ba5a
  target_soc: 100
  power: "1000"

springfall2008, 2024-08-06:
Fix on 'main' and being tested

UKHarj, 2024-08-06:
Hi Trefor, its harjinder dool from the predbat facebook page. I tested overnight from main and predbat charged and held charge over night on some agile import rates which were lower than my export rate, can't see an errors in the log so I think the fix in main is working.

springfall2008, 2024-08-06:
Great, new release has been made

closed 2024-08-06 by springfall2008