springfall2008/batpred #1334: Freeze charge not freezing

naltsta, 2024-07-29:
**Describe the bug**
Freeze charge from the plan used to work very well for agile prices that were reasonably cheap but not worth using for charging. My last successful freeze charge was on 21/7/24.

Since then I have had numerous freeze charge in the plan, none of which have done anything. Manual freeze charge from predate does not work. If I change GivTCP to "idle (paused)" while predbat is freeze charging then predate changes it back to idle.

Charging form the plan and manual charging are both working as expected - just seems to be freeze charging that is not.

**Expected behavior**
freeze charges works as it did previously

**Predbat version**

8.3.1 add on
**Environment details**

Givenergy 3.6kW inverter using the rest api

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Are these expected?

2024-07-29 22:25:26.024015: **Charge window will be disabled as freeze charging is planned**
2024-07-29 22:25:26.024064: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-07-29 22:25:26.024438: Adjust idle time computed idle is 22:25:00-23:59:00
2024-07-29 22:25:26.024502: **Setting ECO mode as no discharge window planned**
2024-07-29 22:25:26.024696: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-07-29 22:25:26.024736: Inverter 0 Turning off scheduled discharge
2024-07-29 22:25:26.024766: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 00:00:00 to 00:00:00 - 00:00:00
2024-07-29 22:25:26.024802: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2024-07-29 22:25:26.025117: Adjust idle time computed idle is 22:25:00-23:59:00
2024-07-29 22:25:26.025228: Inverter 0 Current Target SOC is 100 already at target
2024-07-29 22:25:26.025261: Adjust reserve to default (as set_reserve_enable is true)
2024-07-29 22:25:26.025300: Inverter 0 Current reserve is 4.0 already at target
2024-07-29 22:25:26.247574: Completed run status Idle
2024-07-29 22:25:26.286183: Info: record_status Idle

mpartington, 2024-07-30:
Just to rule some things out, what gen inverter and firmware revision are you using? Have you recently updated the firmware? I think Trefor has the same issue on his gen 1 hybrid with the faster firmware.

I know that the pause command isn't currently controlled by rest and relies on the GivTCP entities.

Are these unhashed in the apps.yaml file? 

If you have a gen 1 then it doesn't set the pause times.

It's all working ok on my AC3, so sounds like something has changed for you 

[image]

naltsta, 2024-07-30:
Gen 1 hybrid

Firmware is all way older than my issue 

Inverter Firmware Version
D0.450-A0.451
Last Update Time
2023-11-26 23:47:43
Battery Firmware Version
3015

I have unhashed the Giv TCP entities in apps.yaml today (but they weren't a week ago when things were working, I presumed through rest) and that doesn't seem to have made a difference. Predate tell me I am free charging right now but my battery is supplying all house load.

GivTCP log since midnight (when freeze charging should have started) is below

2024-07-31 00:00:42,443 - Inv1 - read        -  [CRITICAL] - Midnight, so resetting Day/Night stats...
2024-07-31 00:05:07,213 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 32
2024-07-31 00:05:09,457 - Inv1 - write       -  [INFO    ] - Setting shallow charge 32 was a success
2024-07-31 00:05:34,817 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2024-07-31 00:10:05,391 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 31
2024-07-31 00:10:07,002 - Inv1 - write       -  [INFO    ] - Setting shallow charge 31 was a success
2024-07-31 00:10:13,593 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 31
2024-07-31 00:10:15,203 - Inv1 - write       -  [INFO    ] - Setting shallow charge 31 was a success
2024-07-31 00:10:25,281 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 31
2024-07-31 00:10:26,894 - Inv1 - write       -  [INFO    ] - Setting shallow charge 31 was a success
2024-07-31 00:20:06,761 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 30
2024-07-31 00:20:08,391 - Inv1 - write       -  [INFO    ] - Setting shallow charge 30 was a success
2024-07-31 00:25:05,675 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 28
2024-07-31 00:25:07,274 - Inv1 - write       -  [INFO    ] - Setting shallow charge 28 was a success
2024-07-31 00:25:16,339 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 28
2024-07-31 00:25:17,956 - Inv1 - write       -  [INFO    ] - Setting shallow charge 28 was a success

Thanks for your interest!

naltsta, 2024-07-30:
Predbay planned freeze charging until 00.30 and then idle mode. At 00.33 i started a manual freeze charge. You can see givTCP sending different target SOC for idle or freeze but it doesn't seem to be following them.

In reality it stayed in idle mode the whole time.

2024-07-31 00:25:07,274 - Inv1 - write       -  [INFO    ] - Setting shallow charge 28 was a success
2024-07-31 00:25:16,339 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 28
2024-07-31 00:25:17,956 - Inv1 - write       -  [INFO    ] - Setting shallow charge 28 was a success
2024-07-31 00:30:06,319 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2024-07-31 00:30:07,972 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2024-07-31 00:30:13,497 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of night rate tariff (Dynamic)
2024-07-31 00:33:23,282 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 27
2024-07-31 00:33:24,716 - Inv1 - write       -  [INFO    ] - Setting shallow charge 27 was a success

naltsta, 2024-07-30:
At 00.37 I used givTCP to turn on "eco (paused)" - temporarily the inverter behaved as expected but then reverted to eco mode (presumably directed by predbat)

2024-07-31 00:33:23,282 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 27
2024-07-31 00:33:24,716 - Inv1 - write       -  [INFO    ] - Setting shallow charge 27 was a success
2024-07-31 00:35:06,747 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 26
2024-07-31 00:35:08,357 - Inv1 - write       -  [INFO    ] - Setting shallow charge 26 was a success
2024-07-31 00:37:43,868 - Inv1 - write       -  [INFO    ] - Setting Battery Mode to: Eco (Paused)
2024-07-31 00:37:46,479 - Inv1 - write       -  [INFO    ] - Setting dynamic mode failed: (<class 'AssertionError'>, AssertionError('Unexpected response from remote end: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 44 bytes (0 received)'), <traceback object at 0x7fd5e4bb0980>)
2024-07-31 00:37:48,110 - Inv1 - write       -  [INFO    ] - Setting shallow charge 100 was a success
2024-07-31 00:40:06,508 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 26
2024-07-31 00:40:08,122 - Inv1 - write       -  [INFO    ] - Setting shallow charge 26 was a success

mpartington, 2024-07-31:
Looking at your firmware version, its the older one that doesn't have the new battery pause mode enabled. So you were right to have had the pause entities hashed out in apps.yaml, so I'd change that back

As far as I know, to pause charge on D0.450-A0.451 firmware Predbat would raise the reserve level to the current SoC (it seems to be doing this from the logs)  and also set battery discharge rate to zero. You could check homeassistant to see what happened to these at the time pausecharge should have been set. Otherwise probably best to upload a whole log file that captures when it happens.

I don't think you can directly select ecopause, it is set by a combination of other entity states - not 100% sure which

Only other thing to try is a reboot of the inverter, see if it makes any difference

springfall2008, 2024-08-01:
I think its a bug, normally reserve is used to hold the charge but it wasn't setting it.

Fix on 'main' for testing, sorry I can't try it myself as I'm away.

naltsta, 2024-08-01:
Thanks - I've updated to main. Will see how it gets on when the sun goes down.

springfall2008, 2024-08-06:
Any luck?

naltsta, 2024-08-06:
All seems to be working as expected now - thank you for investigating and finding a fix

closed 2024-08-07 by springfall2008