springfall2008/batpred #1335: REST Disabled

Drewmarkham, 2024-07-30:
Hi - as a newb, I am just starting to understand predbat. I have an issue with REST being disabled. 

From log:

2024-07-30 09:57:39.672530: Error: Exception raised HTTPConnectionPool(host='192.168.1.144', port=6345): Max retries exceeded with url: /readData (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7faa9633d0>: Failed to establish a new connection: [Errno 111] Connection refused'))
2024-07-30 09:57:39.672849: Warn: Inverter 0 unable to read REST data from http://192.168.1.144:6345/readData - REST will be disabled
2024-07-30 09:57:39.713973: Info: record_status Inverter 0 unable to read REST data from http://192.168.1.144:6345/readData - REST will be disabled
2024-07-30 09:57:39.714365: Warn: Inverter control auto restart trigger: REST read failure

From apps.yaml:

  givtcp_rest:
    - 'http://192.168.1.144:6345'

Version info from Log:

-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.7.3
Predbat files are installed correctly for version v8.3.1
**** Starting Standalone Predbat ****
2024-07-30 09:57:29.255543: Loading apps.yaml
 Home Assistant Supervisor: 2024.06.2
 
I have a VPN and have restarted predbat with it off and still errors.

I have looked to see if this has been answered before, but can't seem to find anything.

TIA

springfall2008, 2024-08-01:
Does the readData link above work for you?
Are you using the release version of GivTCP (and not the dev one)?

Drewmarkham, 2024-08-02:
Thx - I have rectified my mistake (typo).

closed 2024-08-04 by gcoan