springfall2008/batpred #1336: Two Car Charging Plans

dandwhelan, 2024-07-30:
**Describe the solution you'd like**
I've two electric cars with two different chargers. One car charges at 7.2Kw and the other charges at 3.6kw.. I would like predbat to have the functionality to independantly control the charging of two different cars, at two different two charging speeds. 

I'd like to see two charging slots available to each car considering the speed they will charge at. Also on the html plan you'd need to add a column for car 1 and car2.

**Describe alternatives you've considered**
Using the normal charging plan and make both switches turn on at the same time. But it wont work well.

Thanks

dandwhelan, 2024-08-31:
I'd still love this feature allowing me to specify the max charging speed of two cars on predbat :) @springfall2008 
PS, love the new web interface. great for troubleshooting.

springfall2008, 2025-11-26:
Now supported AFAIK