springfall2008/batpred #1341: Total Cost Saving Not Incremnetally Increasing daily

Timeflies1980, 2024-08-04:
**Describe the bug**
Total Cost Saving Not Incremnetally Increasing daily 

**Expected behavior**
The graph should increase daily to represent consolidated savings

**Predbat version**
8.3.1 - Addon

**Environment details**

- Givenergy 5kW inverter and 9.5 attery
- HAOS as addon
- Anything else?

**Screenshots**

**Log file**
[predbat.log.1.txt](https://github.com/user-attachments/files/16486877/predbat.log.1.txt)

springfall2008, 2024-08-05:
There's a bunch of connection errors at around midnight. This can't be a co-incidence do you have something set up to restart at midnight e.g. a router or a server?

2024-08-04 00:00:56.858869: Adjust idle time, charge 23:30:00-02:30:00 discharge 00:00:00-00:00:00
2024-08-04 00:00:56.859495: Adjust idle time computed idle is 02:30:00-23:59:00
2024-08-04 00:00:56.859782: Car charging from battery is off, next slot for car 0 is 08-04 00:00:00 - 08-04 00:30:00
2024-08-04 00:00:56.859940: Inverter 0 Current charge limit is 61 % and new target is 59 %
2024-08-04 00:01:08.181930: Inverter 0 charge target 59 via REST successful on retry 0
2024-08-04 00:01:08.600074: Adjust reserve to default as SOC 35 % is above target 59 % or charging active
2024-08-04 00:01:08.600438: Inverter 0 Current reserve is 4.0 already at target
2024-08-04 00:01:08.978391: Completed run status Charging
2024-08-04 00:01:09.042741: Info: record_status Charging target 35%-59%0
2024-08-04 00:01:09.112411: Saved current settings to /config/predbat_config.json
2024-08-04 00:03:03.697948: Warn: Web Socket closed, will try to reconnect in 5 seconds
2024-08-04 00:03:08.704337: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-04 00:03:08.720766: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2024-08-04 00:03:08.728551: Error: Traceback (most recent call last):
  File "/config/ha.py", line 67, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2024-08-04 00:03:08.730174: Warn: Web Socket closed, will try to reconnect in 5 seconds
2024-08-04 00:03:13.736074: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-04 00:03:13.751166: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2024-08-04 00:03:13.752081: Error: Traceback (most recent call last):
  File "/config/ha.py", line 67, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2024-08-04 00:03:13.753566: Warn: Web Socket closed, will try to reconnect in 5 seconds
2024-08-04 00:03:18.759524: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-04 00:03:18.777237: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2024-08-04 00:03:18.778371: Error: Traceback (most recent call last):
  File "/config/ha.py", line 67, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2024-08-04 00:03:18.779898: Warn: Web Socket closed, will try to reconnect in 5 seconds
2024-08-04 00:03:23.786058: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-04 00:03:23.801186: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2024-08-04 00:03:23.802105: Error: Traceback (most recent call last):
  File "/config/ha.py", line 67, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2024-08-04 00:03:23.803372: Warn: Web Socket closed, will try to reconnect in 5 seconds
2024-08-04 00:03:28.807912: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-04 00:03:28.825681: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
2024-08-04 00:03:28.826687: Error: Traceback (most recent call last):
  File "/config/ha.py", line 67, in socketLoop
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
[... cut ...]
    async with session.ws_connect(url) as websocket:
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 1167, in __aenter__
    self._resp = await self._coro
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/client.py", line 821, in _ws_connect
    raise WSServerHandshakeError(
aiohttp.client_exceptions.WSServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

2024-08-04 00:03:33.854427: Warn: Web Socket closed, will try to reconnect in 5 seconds

Timeflies1980, 2024-08-05:
I’ve got an automation to restart HA at 00:03:03 ever night. Guess this might be the issue. I’ve changed it to later to see if this fixes the issue.

springfall2008, 2024-08-06:
> I’ve got an automation to restart HA at 00:03:03 ever night. Guess this might be the issue. I’ve changed it to later to see if this fixes the issue.

Yes that is the problem its actually when the savings update.

Why do you want to restart HA you will end up with gaps in your data every night?

closed 2024-08-07 by springfall2008