springfall2008/batpred #1342: Predbat not limiting Export set in apps.yaml

Drewmarkham, 2024-08-05:
Hi,

I have set the export limit to 3600 in apps.yaml (as reqd by SSEN):

# Export limit is a software limit set on your inverter that prevents exporting above a given level
  # When enabled Predbat will model this limit
  export_limit:
    - 3600
 
Yet predbat is exporting 6000 (see screenshot):

[image]

Do I need to change settings elsewhere?

TIA

gcoan, 2024-08-05:
Predbat doesn't limit the import, it's your inverter that does that. 

The export limit in apps.yaml is to model the inverter export limit. 

If you are exporting too much its a question for your installer

Drewmarkham, 2024-08-05:
I have been told my Inverter export limit was set to 3600 (and confirmed by GivEnergy in line with SSEN requirements) and I thought I should set the export limit on Predbat to the same. So these are now meaningless values?

springfall2008, 2024-08-05:
Looks like you can export at 6000 so you should set apps.yaml accordingly.

Drewmarkham, 2024-08-05:
Unfortunately my DNO (SSEN) has not authorised 6000, but only 3600. I have checked with GivEnergy Support and they have set export to 3600 and discharge to 3198, but predbat ignores these and the apps.yaml setting and is still exporting 6000. I do not wish to get into bother with SSEN.

gcoan, 2024-08-05:
Appreciate that it sounds like this is being bounced back and forth, but to reiterate the point made earlier:

1. Predbat uses the export limit in apps.yaml to model the effects of the inverter export limit on future income from exporting

2. It looks like from the screenshot that Predbat is initiating a full rate (6K) battery discharge and as very little of the battery power is being used by the house, most of the battery discharge plus solar generation is being exported

3. It is the inverter NOT predbat that controls how much export occurs and imposes any DNO limits that are on your G98/G99 approval

TBH its highly unlikely that your DNO would even be aware of you exceeding the export limit as there's no grid power monitoring of individual home import or export.  The limits set by the DNO are conservatively calculated to ensure there isn't undue stress on the grid when you and neighbours also with solar are exporting.

From the evidence given I would say that your inverter hasn't been correctly configured with an export limit, but if GivEnergy say it has then you have to believe them.  Did this export occur at full rate for an extended period of time or did it start and then throttle back?

4. One option you could do is to reduce the maximum battery discharge rate so when Predbat initiates a force discharge it does so at the reduced power level.  See:
https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations
However this will almost certainly 'limit' your inverter (I assume it is an AIO) at 3.6kW max discharge rather than the 6kW it is capable of, so if you have house load above 3.6kW (e.g. ASHP, shower, oven, or several things on at once) you'll end up grid importing.

Drewmarkham, 2024-08-06:
Yes it is an AIO.

I have now double checked with GivEnergy Support and they say I definitely have the export limit set to 3600. They have also set a max discharge to 3198 - that also gets ignored.

Their support person was also satisfied with the configuration of the AIO and Gateway.

I understand the comment about the DNO and lack of monitoring, but I'm not sure I want to be on "the wrong side of the line".

Is it worth me turning Predbat off completely and then manually setting import/export and seeing what occurs?

gcoan, 2024-08-06:
If you turn switch.predbat_set_read_only to on, this will stop predbat from controlling your inverter but still leave it monitoring what is happening.

Yes probably worth trying manual force discharge to see what happens.  I'm leaning towards it being a bug in the AIO firmware that is not taking account of the export limit when discharging the battery, and a manual test should confirm this.

May be worth you upgrading the AIO firmware if you are not on the latest version?  But do look at the givenergy community forum for other feedback on the firmware version beforehand

Drewmarkham, 2024-08-06:
Thx. 

I have set_read_only to On (and left mode to charge & discharge). I have manually set a charge for two hours time (2.5 hours) and set a discharge for an hour after that (1 hour). Max discharge I have reset to 6000. I can't see the Export limit anywhere, but am assuming GivEnergy support are correct and it will export 3600.

Let's see what happens :-)

gcoan, 2024-08-06:
🤞

The export limit isn't visible anywhere in the portal as far as I know.  You can however see when its changed by looking at the remote control history in the portal

Drewmarkham, 2024-08-07:
With read only on. And setting charge/export on the AIO manually. It still peaked at 6000. So I turned Predbat/HA machine off completely and re-ran the test with the same results. Which implies my export limit is not set to 3600 or (as you have stated) the Inverter is ignoring this limit for export.

I then tried to update the firmware, from D06.11 to D06.12 and the update failed. I am going to contact GivEnergy support to understand what/why later this afternoon.

gcoan, 2024-08-07:
At least you've eliminated Predbat as being the cause of it.

Don't let GivEnergy suggest that setting the discharge rate to 3600 is the fix for this, this'll hobble what you can get out of your AIO

I'm lucky enough to not have an export limit imposed by my DNO and can export around 8kW from the solar panels at peak times.  Did your installer apply for a G99 and the DNO imposed that limit, or did they just go for a G98 which doesn't require DNO approval but limits your export output to 3.6kW?

Drewmarkham, 2024-08-08:
Yes, not Predbat. Told to update firmware which succeeded 2nd time, just ran another export still @6000. Will get back to support this pm.

Agreed!

Had "fun" with DNO (SSEN) - raised G99 which was returned (eventually) with 3600 as limit (so same as G98 - which I already had for 13 year old PV system)

Drewmarkham, 2024-08-16:
Think I have confirmation of the solution from GivEnergy Support. The setting "Enable Export Limit (G100)" - has been set to "true" 

My export is now limited to 3600! I will be keeping a close eye on for the next few days.

Thanks for your assistance and if anyone else seems to have a similar issue, maybe this will help.

closed 2024-08-18 by springfall2008