springfall2008/batpred #1348: Thread starvation issue not being trapped and causing a predbat failure

gcoan, 2024-08-06:
**Describe the bug**
Randomly looking at the Predbat HTML plan I noticed that Predbat has stopped running:
[image]

But predbat status and the error attribute doesn't indicate any error:
[image]

Predbat active has been set to on for the last 3 hours:
[image]

Logfiles show thread starvation error
[appdaemon.log.1.log](https://github.com/user-attachments/files/16511131/appdaemon.log.1.log)

Restarting appdaemon got predbat working again:
[appdaemon.log](https://github.com/user-attachments/files/16511220/appdaemon.log)
[image]

**Expected behavior**
Thread error trapped, predbat status changed to something meaningful and error attribute set to True so my error detection automation would have picked up the error

**Predbat version**
8.3.0

**Environment details**
HAOS 12.4 running as a VM
I suspect the error was caused by a second (test) HA VM instance running this morning which accidentally had the max memory set to 12Gb, this caused the host PC to go very slow as it ran out of memory.  Nevertheless the error should be trapped by Predbat

gcoan, 2024-08-13:
Same thing happened again today, Predbat stuck, active mode turned on but the plan hasn't been updated all night.

Logfile shows thread error that predbat doesn't trap.  I see from the logfile that octoblock which is also running within Appdaemon continues to run fine every half hour, so issue is in the predbat code.

Logfile sample when it stopped running:
```
2024-08-13 00:20:28.243578 INFO pred_bat: Calculated end_record as 08-14 10:00:00
2024-08-13 00:20:28.245433 INFO pred_bat: Starting second optimisation end_record 08-14 10:00:00 best_price 12.4 best_price_discharge 13.7 lowest_price_charge 8.6 with charge limits [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7, 13.7]
2024-08-13 00:20:28.246742 INFO pred_bat: Optimise price set 13.7 pass freeze price 13.7 start_at_low False best_price 12.4 best_metric -862.77 best_cost -858.04 best_cycle 61.46 best_carbon 0 best_import 14.03
2024-08-13 00:20:29.593185 INFO pred_bat: Best charge windows best_metric -589.59 best_cost -658.86 best_carbon 0 best_import 15.63 metric_keep 0 end_record 08-14 10:00:00 windows [ 13-08 10:00:00 - 13-08 10:30:00 @ 11.25p 100%, 13-08 10:30:00 - 13-08 11:00:00 @ 9.48p 100%, 13-08 11:00:00 - 13-08 11:30:00 @ 9.9p 100%, 13-08 11:30:00 - 13-08 12:00:00 @ 7.8p 100%, 13-08 12:00:00 - 13-08 12:30:00 @ 9.6p 100%, 13-08 12:30:00 - 13-08 13:00:00 @ 8.72p 100%, 13-08 13:00:00 - 13-08 13:30:00 @ 10.58p 100%, 13-08 13:30:00 - 13-08 14:00:00 @ 9.6p 100%, 13-08 14:00:00 - 13-08 14:30:00 @ 10.98p 100%, 14-08 10:00:00 - 14-08 10:30:00 @ 11.25p 100%, 14-08 10:30:00 - 14-08 11:00:00 @ 9.48p 100%, 14-08 11:00:00 - 14-08 11:30:00 @ 9.9p 100%, 14-08 11:30:00 - 14-08 12:00:00 @ 7.8p 100%, 14-08 12:00:00 - 14-08 12:30:00 @ 9.6p 100%, 14-08 12:30:00 - 14-08 13:00:00 @ 8.72p 100%, 14-08 13:00:00 - 14-08 13:30:00 @ 10.58p 100%, 14-08 13:30:00 - 14-08 14:00:00 @ 9.6p 100%, 14-08 14:00:00 - 14-08 14:30:00 @ 10.98p 100%, 14-08 14:30:00 - 14-08 15:00:00 @ 12.38p 100%, 14-08 15:00:00 - 14-08 15:30:00 @ 14.55p 100%, 14-08 15:30:00 - 14-08 16:00:00 @ 16.32p 100%, 14-08 16:00:00 - 14-08 16:30:00 @ 30.05p 100%, 14-08 16:30:00 - 14-08 17:00:00 @ 29.18p 100%, 14-08 17:00:00 - 14-08 17:30:00 @ 31.01p 100%, 14-08 17:30:00 - 14-08 18:00:00 @ 34.09p 100%, 14-08 19:00:00 - 14-08 19:30:00 @ 21.41p 100%, 14-08 19:30:00 - 14-08 20:00:00 @ 21.41p 100%, 14-08 20:00:00 - 14-08 20:30:00 @ 20.81p 100%, 14-08 20:30:00 - 14-08 21:00:00 @ 20.81p 100%, 14-08 21:00:00 - 14-08 21:30:00 @ 23.94p 100%, 14-08 21:30:00 - 14-08 22:00:00 @ 21.17p 100%, 14-08 22:00:00 - 14-08 22:30:00 @ 21.87p 100%, 14-08 22:30:00 - 14-08 23:00:00 @ 18.52p 100%, 14-08 23:00:00 - 14-08 23:30:00 @ 18.52p 100%, 14-08 23:30:00 - 15-08 00:00:00 @ 18.52p 100%, 15-08 00:00:00 - 15-08 00:30:00 @ 17.3p 100% ]
2024-08-13 00:20:29.594721 INFO pred_bat: Best discharge windows best_metric -589.59 best_cost -658.86 best_carbon 0 best_import 15.63 metric_keep 0 end_record 08-14 10:00:00 windows [ 14-08 07:30:00 - 14-08 08:00:00 @ 15.0p 99%, 14-08 08:00:00 - 14-08 08:30:00 @ 15.0p 99%, 14-08 08:30:00 - 14-08 09:00:00 @ 15.0p 99%, 14-08 09:00:00 - 14-08 09:30:00 @ 15.0p 99%, 14-08 09:30:00 - 14-08 10:00:00 @ 15.0p 99% ]
2024-08-13 00:20:29.596055 INFO pred_bat: Optimise price set 13.7 pass normal price 13.7 start_at_low False best_price 12.4 best_metric -589.59 best_cost -658.86 best_cycle 35.71 best_carbon 0 best_import 15.63
2024-08-13 00:29:59.821253 WARNING AppDaemon: Queue size for thread thread-2 is 51, callback is 'run_time_loop() in pred_bat' called at 2024-08-13 00:20:00+01:00 - possible thread starvation
2024-08-13 00:29:59.821621 INFO AppDaemon: --------------------------------------------------
2024-08-13 00:29:59.821764 INFO AppDaemon: Threads
2024-08-13 00:29:59.821925 INFO AppDaemon: --------------------------------------------------
2024-08-13 00:29:59.822317 INFO AppDaemon: Currently busy threads: 1
2024-08-13 00:29:59.822466 INFO AppDaemon: Most used threads: 2 at 2024-08-12 16:30:00+01:00
2024-08-13 00:29:59.822712 INFO AppDaemon: Last activity: 2024-08-12T16:30:00+01:00
2024-08-13 00:29:59.823190 INFO AppDaemon: Total Q Entries: 51
2024-08-13 00:29:59.823344 INFO AppDaemon: --------------------------------------------------
2024-08-13 00:29:59.824034 INFO AppDaemon: thread-0 - qsize: 0 | current callback: idle | since never, | alive: True, | pinned apps: ['hello_world']
2024-08-13 00:29:59.824204 INFO AppDaemon: thread-1 - qsize: 0 | current callback: idle | since 2024-08-13T00:00:01+01:00, | alive: True, | pinned apps: ['octoblock']
2024-08-13 00:29:59.824360 INFO AppDaemon: thread-2 - qsize: 51 | current callback: run_time_loop() in pred_bat | since 2024-08-13T00:20:00+01:00, | alive: True, | pinned apps: ['pred_bat']
2024-08-13 00:29:59.824482 INFO AppDaemon: --------------------------------------------------
2024-08-13 00:30:00.927295 INFO octoblock: Lowest average price for 1 hour block is: 8.7 p/kWh
2024-08-13 00:30:00.929125 INFO octoblock: Best priced 1 hour period starts at: 2024-08-13T11:30:00BST
2024-08-13 00:30:01.004238 INFO octoblock: Lowest average price for 1.5 hour block is: 8.7 p/kWh
[... cut ...]
2024-08-13 00:33:09.255115 WARNING AppDaemon: Queue size for thread thread-2 is 68, callback is 'run_time_loop() in pred_bat' called at 2024-08-13 00:20:00+01:00 - possible thread starvation
```

Full logfile:
[appdaemon.log](https://github.com/user-attachments/files/16596849/appdaemon.log)

This time I wasn't running any extra VM's so shouldn't have been a host machine issue.

apps.yaml is set to auto threads:
```
  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto
```

No corresponding messages in appdaemon, host, supervisor or HA core logs.

Restarting appdaemon got predbat working again

gcoan, 2024-09-27:
Looks like the same thing happened again last night at 21:20, similar thread queue size messages coming from appdaemon and predbat stops executing

[appdaemon.log.6.log](https://github.com/user-attachments/files/17168214/appdaemon.log.6.log)

Restarting appdaemon fixed the issue and predbat started running again.

You mentioned on #1469 that you weren't sure how you could fix this.

I have in mind a way to enhance the automatic predbat error detector that's included in the documentation https://springfall2008.github.io/batpred/output-data/#predbat-error-monitor to detect predbat no longer executing and to restart predbat or the appdaemon add-on to try to cure the 'stuck' execution.  Its not ideal but I think it would work if you don't think you can detect this in code

gcoan, 2024-09-30:
I've identified and tested a workaround for this issue #2e68c114 - enhancing the 'predbat error monitor' so that it now detects 'stalled execution issues' by triggering if status.last_updated has not changed for 20 minutes, and now it auto-restarts the Predbat add-on

springfall2008, 2024-10-06:
BTW: I would count AppDaemon as depreciated now. Is there a reason you still use it?

gcoan, 2024-10-06:
There were two reasons I was remaining on AppDaemon, one was that I used Octoblock in AppDaemon to calculate cheapest overnight slots for the washing machine, etc.  I have swapped to using the Octopus integration target rate sensors and now ceased using Octoblock so that reason is closed off.

And number two is #1457 I currently use my own shell scripts to write powerup and octopus free electricity events as import rate overrides to apps.yaml and due to Home Assistant's protected storage model, I don't have access to /addon_config/<predbat>/apps.yaml from the command line and need an API to enable me to inject these electricity overrides into Predbat.

Would love to swap to the new Web UI with Predbat but as I am getting a power up event most weeks, sometimes every few days, I don't want to backtrack and have to manually edit them in apps.yaml.  Hence stuck with AppDaemon until #1457 FR happens

Cheers

closed 2024-10-08 by gcoan