springfall2008/batpred #1349: Carbon enable causing discharge during charge window.

johnwb87, 2024-08-06:
**Describe the bug**
With carbon enable on and any non zero value for metric_carbon the plan often discharges during a low rate charge window. This is with Calculate Discharge on charge slots turned off in expert mode and combine charge slots off.

**Expected behavior**
With carbon enabled, Calculate Discharge on charge slots turned off (in expert mode) and combine charge slots off, I hoped that predbat would use carbon metric to influence the plan during cheap rate charge period, hold charge in some situations but not discharge during charge slots. So a middle ground of optimised carbon without brown export during a cheap import rate.

**Predbat version**

8.3.2

**Environment details**

GivEnergy Gen3 Hybrid 5.0 and 9.5kWh
Installed as Add-on

**Screenshots**
[image]
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/16512995/predbat.log)

johnwb87, 2024-08-07:
This is the sort of plan I hope for, currently what is being set for tonight. That first period hold charge is a no cost, no regret choice to get a small carbon benefit.
[image]

springfall2008, 2024-08-07:
Have you tried smaller values, 20p/kG is still fairly large? That said I don't see any problem with the plan its gaining some money exporting at 15p while importing is at 7p

If you turn on combine charge then the discharge during the night period will certainly go away. The reason is actually that the 7p charge slot used for export was never selected for charge in the first place and hence its free for discharge.

johnwb87, 2024-08-07:
Thanks Trefor,

Documentation has the carbon_metric suggestion between 10-200. So I felt 20 was modest. Will experiment with lower.

Interesting insight into the logic. So the slot wasn't chosen for charge, possibly because of the carbon metric, which then opens it up for a discharge, and is then selected with influence from carbon metric. So does imply the sensitivity to the carbon metric.

In evolving the Carbon optimisation perhaps it creates a need for a Calculate Discharge on Low Rate switch. Which would neutralise the potential knife edge, prevent brown export during a low rate and allow Charge to be optimised for carbon(which combining charge slots neutralises)

closed 2025-01-01 by springfall2008