springfall2008/batpred #1352: Huawei Luna2000 charge discharge loop

JohanAlvedal, 2024-08-07:
**Describe the bug**
When is charging to a specified % and  the value Battery discharge = number.battery_maximum_discharging_power is set to 4000W 
So the battery charges as it should, but as soon as it reaches the specified %, it starts to discharge again if  the value Battery discharge = number.battery_maximum_discharging_power is set to 4000W. 
Huawei Luna200 otherwise it will keep charging and then discharge charge in an endless loop.
This is not noticeable as long as the sun is generating, but it is when it is dark outside.

**Expected behavior**
When the battery should charge to a specified %, it should set Battery discharge = number.battery_maximum_discharging_power must be set to 0  
Only when the battery is expected to discharge using force or the house's consumption, should it be 4000W in Batteries Maximum discharging power = number.battery_maximum_discharging_power

**Predbat version**
8.3.2

**Environment details**
- Huawei Luna200
- Standard HAOS 

**Screenshots**
Here you can clearly see how it looked last night when it charged to 100% and then discharged until predbat wanted to charge it again
[image]

**Log file**
No problem in log.

JohanAlvedal, 2024-08-07:
I'm seeing the opposite problem now with the battery draining. charge must be set to the value 0
 number.battery_maximum_charging_power
[image]

JohanAlvedal, 2024-08-08:
Today 

[image]

springfall2008, 2024-08-08:
This is different from other inverters, I'll try to add an option for this behaviour soon

springfall2008, 2024-08-10:
I've added a change to 'main' to enable a new inverter option called 'charge_discharge_with_rate' which when set to True should do what you want.

e.g:

inverter:
   charge_discharge_with_rate: True

Can you please give it a try, I don't want to change the default for HU until it works?

JohanAlvedal, 2024-08-10:
Ok trying now but something is wrong.

[image]
[image]

JohanAlvedal, 2024-08-10:
Now that it went to idle it is correct.

[image]

JohanAlvedal, 2024-08-10:
https://justpaste.it/dzonh

springfall2008, 2024-08-10:
What is in your apps.yaml?

JohanAlvedal, 2024-08-10:
device_id: 6893ee838a0972bd3e50d9e0faa0ba5a  # my battery

  # Inverter max AC limit (one per inverter). E.g for a 3.6kw inverter set to 3600
  # If you have a second inverter for PV only please add the two values together
  inverter_limit:
   - 5000

  # Set the maximum charge/discharge rate of the battery
  # I created a number input field so I can change its value
  battery_rate_max:
   - input_number.pb_battery_rate_max

  inverter:
    charge_discharge_with_rate: True

  # Export limit is a software limit set on your inverter that prevents exporting above a given level
  # When enabled Predbat will model this limit
  #export_limit:
  # - 36

springfall2008, 2024-08-10:
I don't really understand the charging power thing, can you show these two on the chart?

    - number.battery_maximum_charging_power
    - number.battery_maximum_discharging_power

JohanAlvedal, 2024-08-10:
> I don't really understand the charging power thing, can you show these two on the chart?
> 
> ```
> - number.battery_maximum_charging_power
> - number.battery_maximum_discharging_power
> ```

and I don't understand what you mean by chart?
Maybe you and I would have a video meeting on messenger so you can see how my inverter behaves.

JohanAlvedal, 2024-08-11:
Ok, I think I know how they work.
When the inverter has no "forcible charge" or "forcible discharge," we shall call it "idle."
If there is excess power from the solar cells and number.battery_maximum_charging_power has the value 0, it exports everything. But if it has a positive value, say 2000W, then it charges the battery with 2000W, and the rest, if any, is exported.

If number.battery_maximum_discharging_power has the value 0 and the solar cells produce 0W while the house has a power demand, then the power is taken from the grid.

But if number.battery_maximum_discharging_power has a value of 2000W, then power is taken from the battery up to a maximum of 2000W, with the remainder taken from the grid.
This means that as long as you do not run any force action, you can control the charging with number.battery_maximum_charging_power if there is power from the solar cells.

So, if you run a "forcible charge" of 2500W to 80% or for 60 minutes, then number.battery_maximum_charging_power must also be changed to that value (2500W); otherwise, nothing happens. Even if you stop the action, it continues to charge or discharge if the solar cells have excess power.
At the same time, the equivalent setting is number.battery_maximum_discharging_power when discharging the battery.

So then maybe you understand the consequence of - number.battery maximum charging power has a value (say 2000W) at night or when it has forcible charge and it reaches its target of say 80% then it discharges the battery immediately.

JohanAlvedal, 2024-08-15:
I have reverted to V8.3.2 and disabled services. Then, I created an automation that controls my inverter based on the Predbat status and the Predbat charge/discharge sensors. 
It works, but it has become quite complex. Before this, it wasn’t possible to run it properly without spamming my inverter. 

I have also created package files for my Huawei inverter to get the right sensors, even one more for Predbat, along with a complete automation to enable car charging. I will also add the entire automation to control the battery, etc., once I’ve debugged it for a few days. So, a request could be to have an inverter template that doesn’t require direct control but is instead managed via external automations.

closed 2025-08-21 by JohanAlvedal