springfall2008/batpred #1353: Solcast shading feature

paulmower, 2024-08-07:
I found a solcast integration on github where you could add a figure for shading effect by hour. It was  at https://github.com/BJReplay/ha-solcast-solar.

gcoan, 2024-08-07:
This is a fork and expansion of the original oziee solcast integration.  It still works perfectly with Predbat, just set the appropriate sensors in apps.yaml.

After the Oziee integration was nuked by the author I swapped over to the BJreplay version and have continued to be involved in its development.  It offers a number of advantages over the direct Predbat Solcast calls:
- populates the HA energy dashboard with solar predictions
- has shading scaling feature as you highlight
- auto retry when Solcast is busy and gives 429 errors
- caching of data between Solcast polls to avoid Solcast busy errors
- smooth interpolation of predicted power now/power remaining today
- forecast breakdown for individual arrays (e.g. different panels on different roof alignments)
- ability to customise what forecasts are populated in the Solcast sensors
- custom 'forecast next X hours'

I personally don't think Predbat should incorporate all these features, it leads to feature bloat and duplication of function.  If people want the more advanced features then use the separate Solcast integration which Predbat works great with, or for simple demands the Predbat retrieval of Solcast data reduces the number of integrations you need to install.

springfall2008, 2024-08-10:
> I personally don't think Predbat should incorporate all these features, it leads to feature bloat and duplication of function. If people want the more advanced features then use the separate Solcast [...]

Good point, I don't really want to keep replicating another development while it continues to be supported.

>populates the HA energy dashboard with solar predictions

I never figure out how to make this work, support groups in HA are pretty useless for questions too.

>has shading scaling feature as you highlight

This is useful to customise your forecasts

>auto retry when Solcast is busy and gives 429 errors
>caching of data between Solcast polls to avoid Solcast busy errors

Not sure I have seen these issues.

>smooth interpolation of predicted power now/power remaining today

This is done differently via Predbat, you do get the data but not in the same form

>forecast breakdown for individual arrays (e.g. different panels on different roof alignments)

I've tried to avoid that as its not related to Predbat.

>ability to customise what forecasts are populated in the Solcast sensors

Predbat reports 10%/50%/90% already.

>custom 'forecast next X hours'

I think out of scope for Predbat.