springfall2008/batpred #1359: Exclude car kWh from previous load

deanjesper, 2024-08-11:
**Is your feature request related to a problem? Please describe.**

Looking at my Predbat plan today, car charging that happened last Sunday from 9pm appears as predicted load for today at 9pm when in my case, on Intelligent Octopus Go, all car charging is off-peak so taking into account car charging into previous load is wrong.

[image]

**Describe the solution you'd like**

Exclude car kWh from predicted load if on Intelligent Octopus Go

**Describe alternatives you've considered**

N/A

**Additional context**

N/A

gcoan, 2024-08-11:
Are you making use of the separate car charging kWh sensor in apps.yaml,  this should automatically exclude any car charging from the load prediction

See https://springfall2008.github.io/batpred/apps-yaml/#car-charging-filtering

Set switch.predbat_car_charging_hold to on and car_charging_energy in apps.yaml to point to the sensor that records your car charging kWh

deanjesper, 2024-08-12:
Yeah I'm using the separate car charging sensor in apps.yaml but I didn't have Car charging hold on. I thought that held the battery for the car, DOH! I've enabled that now and it's working, thank you! Maybe that toggle should have a better name?

gcoan, 2024-08-12:
Yeah, probably could be named better, but of course that would be a breaking change.

Glad you're sorted now.  Clever stuff this Predbat

closed 2024-08-14 by deanjesper