springfall2008/batpred #1362: Solcast GetUserUsageAllowance endpoint has been deprecated

arturmalinowski, 2024-08-14:
Apparently Solcast's GetUserUsageAllowance endpoint has been officially removed yesterday and this is resulting in a bit of spam in Predbat logs:

`Warn: Error downloading data from url https://api.solcast.com.au//json/reply/GetUserUsageAllowance, code 405`

Not sure if there is an alternative way of getting the usage allowance.

gcoan, 2024-08-14:
@arturmalinowski this seems a bit short sighted, how are you meant to know how many API calls you still have left if this API is withdrawn?

Where did you find that its officially deprecated by Solcast?  I couldn't find anything on their website/blog
Equally I can't see this API covered in the API documentation either https://docs.solcast.com.au/?_gl=1*1ns3eeq*_gcl_au*NjQ3MDg4NzEyLjE3MjM2MzIwMDU.*_ga*MTEwNzMyMDE0LjE3MTU3MjQ4MDk.*_ga_BYH3TC3R79*MTcyMzYzMjAwNS42LjAuMTcyMzYzMjAwOS4wLjAuMA..&_ga=2.160971837.1616858681.1723632005-110732014.1715724809#0556b413-88f7-4856-bf16-24b878a6edaf

gcoan, 2024-08-14:
Just saw the response on GivEnergy Forum

This is the reply I have received from Solcast. I don't really understand what it means and I have replied to Solcast asking for more details. I am accessing Solcast directly from the predbat apps.yaml and am not using the integration. That would explain why this message is coming from Solcast every 5 minutes.

Hi Rob,

The endpoint has been deprecated for quite some time now. However I can see from our logs it looks users were still able to make successful API calls till yesterday for some reason. We will need to look at why that is the case, as they shouldn't have been able to.

If you have any further questions, please let me know.

Kind Regards,
Billy

SwiftRR, 2024-08-14:
I am chasing this with Solcast Support by email. 
As Australia is many hours adrift from us, I will not get a reply now until about 3 am Thurs GMT.
On the Solcast site, My tally of API requests has disappeared from the top banner. This is what I used to see (I have copied this from the Solcast Hobbyist guide: https://kb.solcast.com.au/user-guide-for-solcast-hobbyist-toolkit-account).
[image]
See if this has disappeared for you also.

Rob

SwiftRR, 2024-08-14:
The apps.yaml code for directly accessing Solcast cloud is as outlined in the documentation.:
```
  solcast_host: 'https://api.solcast.com.au/'
  solcast_api_key: 'xxxxxxx'
  solcast_poll_hours: 8
```
Here, the code is asking for polling every 8 hours but predbat refreshes every 5 min. It looks as if Solcast is getting inundated with requests from predbat for the number of API requests every 5 min also (when it isn't needed).

Is there a log for the Solcast integration? Although, as the polling is initiated via an automation, I guess that there will not be the 5 min requests for this info from the integration.

A can of worms ....

Rob

arturmalinowski, 2024-08-14:
I'm willing to bet Solcast website was also caught off guard by this change, you can see in Chrome it makes a request to GetUserUsageAllowance and fails in the same way as Predbat:

[image]

gcoan, 2024-08-14:
> Is there a log for the Solcast integration?

The solcast integration writes its log to the HA core log (menu option system/logs).  Reports when sites fail to load., API limit exceeded, Solcast is busy and it pauses for retry, etc.

Discussion over of this issue on the Solcast integration is here: https://github.com/BJReplay/ha-solcast-solar/discussions/128#discussioncomment-10335517

SwiftRR, 2024-08-15:
For info, this is the reply that I received from Solcast overnight:

> Hi Rob,
> Thanks for forwarding that through.
> As mentioned on the github page, we're looking into alternative (more efficient) solutions at providing this information.
> If you have any further questions, please let me know.
> Kind Regards,
> Billy
> Product Operation Specialist, Solcast

The GitHub page that Billy refers to is the Solcast integration: https://github.com/BJReplay/ha-solcast-solar
that I had referred to this in my previous reply to Solcast. This is affecting both Predbat's direct method and the integration.
For me, I am still receiving the solar predictions in my predbat plan and future days and, looking at my predbat.log, they are coming through every 8 hours as in my yaml code.

So the ball is in Solcast's court. They seem to be trying to come up with a solution.

Do tell me of any points that you might want me to include in a reply to Solcast.

Rob

springfall2008, 2024-08-16:
Oh dear, it looks like the rooftop sites API has gone too:

2024-08-16 08:44:16.788401: Warn: Error downloading data from url https://api.solcast.com.au/rooftop_sites, code 404
2024-08-16 08:44:16.788898: Warn: Solcast sites could not be downloaded, check your Solcast cloud settings

SwiftRR, 2024-08-16:
Seems OK for me on predbat add-on. Extract from my recent predbat.log:
```
2024-08-16 10:25:06.444631: Obtaining solar forecast from Solcast API
2024-08-16 10:25:06.447699: Return cached data for https://api.solcast.com.au/rooftop_sites age 270.0 minutes
2024-08-16 10:25:06.447823: Fetch data for resource id XXXXXXXXX
2024-08-16 10:25:06.449627: Return cached data for https://api.solcast.com.au/rooftop_sites/XXXXXXXX/forecasts age 270.0 minutes
2024-08-16 10:25:06.481573: Fetch data for resource id YYYYYYYY
2024-08-16 10:25:06.483204: Return cached data for https://api.solcast.com.au/rooftop_sites/YYYYYYYYY/forecasts age 269.9 minutes
2024-08-16 10:25:06.572287: Solcast returned 348 data points
2024-08-16 10:25:06.599555: PV Forecast for today is 43.13 (21.66 10% 46.8 90%) kWh and left today is 32.2 (16.64 10% 35.01 90%) kWh
2024-08-16 10:25:06.640160: PV Forecast for day tomorrow is 38.58 (23.33 10% 45.8 90%) kWh
2024-08-16 10:25:06.681977: PV Forecast for day d2 is 33.96 (9.85 10% 45.83 90%) kWh
2024-08-16 10:25:06.724100: PV Forecast for day d3 is 36.25 (14.4 10% 43.04 90%) kWh
2024-08-16 10:25:06.786220: PV Forecast for day d4 is 34.51 (7.64 10% 46.49 90%) kWh
2024-08-16 10:25:06.830347: PV Forecast for day d5 is 25.16 (1.95 10% 45.82 90%) kWh
2024-08-16 10:25:06.872195: PV Forecast for day d6 is 16.87 (1.03 10% 44.47 90%) kWh
2024-08-16 10:25:07.081738: Today's load divergence 108.68 % in-day adjustment 108.24 % damping 0.95x
```

I will see what happens once I reach 480 min (I have my 'poll hours' set to 8 hours = 480 min)

Rob

gcoan, 2024-08-16:
> 2024-08-16 08:44:16.788401: Warn: Error downloading data from url https://api.solcast.com.au/rooftop_sites, code 404

I had an error 405 from the sites retrieval last night in the solcast integration so it used the cached copy, but after that all the subsequent retrievals have been fine.
Probably a transient error

springfall2008, 2024-08-16:
Seems to have cleared now

closed 2024-08-16 by springfall2008