springfall2008/batpred #1363: Octopus Free Electricity session - reverse saving session

johnwb87, 2024-08-14:
Just had email from Octopus about this.

It sounds like it is essentially a reverse savings session as it is only the additional usage that is free.

My first thought is to set a force charge during the window nicely in advance and see what predbat decides to do with that.

Any other thoughts?

CJDumbleton, 2024-08-14:
More background information about free electricity sessions from Octopus here: [https://octopus.energy/free-electricity/](https://octopus.energy/free-electricity/).

gcoan, 2024-08-14:
I suggest the best way is to treat it as a zero price session, then Predbat will decide based on your solar forecast and predicted load as to whether it should discharge the battery or not.

In apps.yaml, add the following:
```
  rates_import_override:
    -  date: '2024-08-15'
       start: '13:00:00'
       end: '14:00:00'
       rate: 0
       load_scaling: 1.0
```

Increase  load scaling if you expect to put the washing on, etc.

Unfortunately until Octopus provide an API for this, there isn't much automation predbat can do.  I have a semi-automated solution which updates apps.yaml for you based on you setting the date and times in Home Assistant - see https://community.givenergy.cloud/d/3968-automating-octopus-power-up-events-into-predbat

chris-y, 2024-08-23:
For Power-Ups there's a third party API here: https://github.com/8none1/octopus_powerups

They are working on one for Free Electricity. 

Would be nice to get these integrated until Octopus have their own API.

gcoan, 2024-08-23:
Interesting, I've asked what region he is in as to whether its the same as me.  I'll set the sensor up and see how it compares to the sessions I get notified of

chris-y, 2024-11-04:
Looks like there's a binary sensor coming for these...

https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/issues/973#issuecomment-2455370370

gcoan, 2024-11-04:
I'm using the 8none1 rest sensor that gives me date and time of the free electricity session, and a separate sensor for power up events.

A series of automations and scripts (copies of which are on the givenergy community forum mentioned earlier) means I:
- get an alert when a new power up or free energy session is available
- can invoke a script to copy the date and time details of the event into Home Assistant helper entities
- then invoke a script that uses the Predbat manual API to instruct Predbat to treat the event as a zero cost price override

So basically two clicks on my dashboard and it all pops up automatically in my Predbat plan.

gcoan, 2024-12-11:
Trefor has added support to Predbat for free electricity sessions based on the Octopus Integration, or screen scraping the Octopus website.
Can close this now

closed 2024-12-11 by gcoan