springfall2008/batpred #1365: solis overlapping times

ncains, 2024-08-15:
I will focus on gathering some specific logs shorty - but the below fragment of issue 1294 still exists for me. modbus sends these overlapping discharge times which are invalid to the solis - but not prevented from being updated. the solis then just does nothing until I manually fix the bad state by hand. the overlaps I have seen always apeear to be 1 minute past the hour.

              another thing I have noticed is that the charge and discharge times are sometimes overlapping, but this is not allowed by Solis. both the webapp and the cloud site have controls to prevent it before submitting it:

```
2024-07-19 14:15:12.264066 INFO pred_bat: Found 1 inverters totals: min reserve 0.28 current reserve 0.28 soc_max 7.1 soc 7.1 charge rate 3.5999999999999996 kW discharge rate 3.5999999999999996 kW battery_rate_min 100.0 w ac limit 4.6 export limit 4.6 kW loss charge 3 % loss discharge 3 % inverter loss 4 %
2024-07-19 14:15:12.330415 INFO pred_bat: Base charge    window [ 19-07 23:30:00 - 20-07 05:30:00 @ 0p 87%, 20-07 23:30:00 - 21-07 05:30:00 @ 0p 87% ] 
2024-07-19 14:15:12.331553 INFO pred_bat: Base discharge window [ 19-07 00:00:00 - 19-07 15:31:00 @ 0p 0.0%, 20-07 00:00:00 - 20-07 15:31:00 @ 0p 0.0% ]
2024-07-19 14:15:12.332487 INFO pred_bat: Charge window will be: 2024-07-19 23:30:00+01:00 - 2024-07-20 05:30:00+01:00 - current soc 100 target 100
2024-07-19 14:15:12.333630 INFO pred_bat: Not setting charging window yet as not within the window (now 07-19 14:15:00 target set_window_minutes 30 charge start time 07-19 23:30:00)
2024-07-19 14:15:12.334883 INFO pred_bat: Next discharge window will be: 2024-07-19 14:20:00+01:00 - 2024-07-19 15:31:00+01:00 at reserve 88
2024-07-19 14:15:16.359657 INFO pred_bat: Inverter 0 Wrote Eco to inverter_mode, successfully now Eco
2024-07-19 14:15:16.361186 INFO pred_bat: Inverter 0 set force discharge to False
2024-07-19 14:15:20.401254 INFO pred_bat: Switch sensor.predbat_GS_0_scheduled_discharge_enable is now off
2024-07-19 14:15:20.406364 INFO pred_bat: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
2024-07-19 14:15:20.410730 INFO pred_bat: Inverter 0 Turning off scheduled discharge 
2024-07-19 14:15:20.415154 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 00:00:00 - 15:31:00 to 00:00:00 - 00:00:00
2024-07-19 14:15:20.419793 INFO pred_bat: Adjust idle time, charge 23:30:00-05:30:00 discharge 00:00:00-00:00:00
2024-07-19 14:15:20.424912 INFO pred_bat: Adjust idle time computed idle is 14:15:00-23:30:00
2024-07-19 14:15:20.430461 INFO pred_bat: Inverter 0 Set new end time to 00:00:00 was 15:31:00

2024-07-19 14:15:24.487102 INFO pred_bat: Inverter 0 Wrote 00:00:00 to discharge_end_time, successfully now 00:00:00
2024-07-19 14:15:28.528276 INFO pred_bat: Inverter 0 Wrote 0 to discharge_end_hour, successfully now 0
2024-07-19 14:15:32.592889 INFO pred_bat: Inverter 0 Wrote 0 to discharge_end_minute, successfully now 0
2024-07-19 14:16:05.677399 INFO pred_bat: Warn: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete 
2024-07-19 14:16:05.728972 INFO pred_bat: Info: record_status Warn: Inverter 0 Trying to press button.solis_modbus_update_charge_discharge_times didn't complete
2024-07-19 14:16:05.734433 INFO pred_bat: Solis Energy Control Switch setting 35 Self-Use unchanged for eco True                                                                                                                                                                                                                      2024-07-19 14:16:05.922303 INFO pred_bat: Completed run status Idle
2024-07-19 14:16:06.300465 INFO pred_bat: Info: record_status Idle 
2024-07-19 14:16:06.344492 INFO pred_bat: Saved current settings to /config/predbat_config.json
```

_Originally posted by @marcoaddario in https://github.com/springfall2008/batpred/issues/1294#issuecomment-2239216293_

springfall2008, 2024-08-16:
I only just released the fix for this yesterday, were you testing on 'main'?

ncains, 2024-08-16:
No sorry, I thought previous release covered it. I will update later today & retest

[Allegis Group]<http://www.allegisgroup.com/>
[Linked In]<https://www.linkedin.com/company/2153>      [Twitter] <https://twitter.com/Allegis_Group>   [Facebook] <https://www.facebook.com/AllegisGroup/>     [Google+] <https://plus.google.com/100595160159631607529>
Neil Cains
CTO Innovation Lab
[Connected]

Futuristic | Strategic | Activator | Ideation | Competition​

***@***.***?anonymous&ep=bwmEmailSignature>         Book time to meet with ***@***.***?anonymous&ep=bwmEmailSignature>

This electronic mail (including any attachments) may contain information that is privileged, confidential, and/or otherwise protected from disclosure to anyone other than its intended recipient(s). Any dissemination or use of this electronic mail or its contents (including any attachments) by persons other than the intended recipient(s) is strictly prohibited. If you have received this message in error, please notify us immediately by reply e-mail so that we may correct our internal records. Please then delete the original message (including any attachments) in its entirety. Thank you

springfall2008, 2024-08-24:
Did it work?

closed 2024-08-25 by springfall2008