springfall2008/batpred #1370: 8.3.4 running in container crashing at startup

sammort, 2024-08-16:
**Describe the bug**
I'm running both Predbat and Home Assistant Core as containers.

The two most recent releases cause predbat to crash on initialisation when running in a container and **not** as an addon.

Reverting to v8.3.2 solves the initialisation crash.

**Expected behavior**
Predbat runs succesfully as a container when not installed as an addon in Home Assistant

**Predbat version**

8.3.4

**Log file**
Batpred logs:

```
2024-08-16 09:52:57.552542: --------------- PredBat - update at 2024-08-16 09:50:00+01:00 with clock skew 0 minutes, minutes now 590
2024-08-16 09:52:57.552834: Config root is /config
2024-08-16 09:52:57.556374: Warn: Failed to decode response <Response [404]> from http://homeassistant:8123/core/api/
2024-08-16 09:52:57.556617: Warn: Unable to connect directly to Home Assistant at http://homeassistant:8123, please check your configuration of ha_url/ha_key
2024-08-16 09:52:57.556763: Config root is /config and printable config_root_p is now /config
2024-08-16 09:52:57.556856: Sanity check:
2024-08-16 09:52:57.557294: Sanity scan files in '/config' : ['apps.yaml', '__pycache__', 'predbat_config.json', 'predbat_save', 'predbat_dashboard.yaml', 'hass.py', 'predbat.log.1', 'predbat.log', 'config.py', 'download.py', 'ha.py', 'inverter.py', 'predbat.py', 'prediction.py', 'unit_test.py', 'utils.py']
2024-08-16 09:52:57.557564: Warn: unable to find /config/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-16 09:52:57.557670: Error: Exception raised 'PredBat' object has no attribute 'get_state'
2024-08-16 09:52:57.568972: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10870, in initialize
    self.auto_config()
  File "/config/predbat.py", line 10709, in auto_config
    states = self.get_state_wrapper()
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/predbat.py", line 937, in get_state_wrapper
    return self.ha_interface.get_state(entity_id=entity_id, default=default, attribute=attribute, refresh=refresh)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 171, in get_state
    return self.base.get_state(entity_id=entity_id, default=default, attribute=attribute)
           ^^^^^^^^^^^^^^^^^^^
AttributeError: 'PredBat' object has no attribute 'get_state'

2024-08-16 09:52:57.583346: Stopping Predbat
2024-08-16 09:52:57.583506: Predbat terminating
2024-08-16 09:52:57.583673: Predbat terminated
```

springfall2008, 2024-08-18:
The error with 8.3.5 isn't correct as it does not match the source code for this version, can you please update predbat.py to 8.3.5 ?

sammort, 2024-08-18:
> The error with 8.3.5 isn't correct as it does not match the source code for this version, can you please update predbat.py to 8.3.5 ?

Yeah you're right, sorry about the false report @springfall2008, I'll close this issue.

Updated to 8.3.5 and needed to restart the homeassistant container (it's not usual that I need to restart home assistant to do a predbat update, but for this release it seems I did).

closed 2024-08-18 by sammort