springfall2008/batpred #1372: Select Manual Charge / Discharge slots for tomorrow, with option to run once, or daily.

Fayrewood, 2024-08-17:
I would like to be able to set a manual charge / discharge slot for tomorrow, but at 16:00hrs, I can only set a slot up to 10:00 the next day.  It would be great to be able to set a slot up to the closest period to the time now.

If possible, also being able to set an option to have the slot as a one time only execution - for example, for Power Up sessions, run once.

gcoan, 2024-08-17:
> If possible, also being able to set an option to have the slot as a one time only execution - for example, for Power Up sessions, run once.

All predbat slot commands are one-time only, once this time is passed it doesn't repeat in any future time slot.
If you want to repeat something you can create an automation to set the predbat manual select controls every day

Fayrewood, 2024-08-17:
Thanks! I didn't know that & that's really helpful.

I find the ability to select a time tomorrow, beyond the current limit, limiting, so if possible, I would like to see a longer list of time slots in the future.

chalfontchubby, 2026-09-18:
Increasing plan_forecast_hours would increase how much you see/plan of the future. Defaults to 30 hours. 
Beyond that, automations are your friend.

closed 2026-09-18 by chalfontchubby