springfall2008/batpred #1373: 8.3.5 has broken #1229 (config pathnames wrong in the logfile)

gcoan, 2024-08-18:
**Describe the bug**
PR #1296 introduced changes to determine the slugname that Predbat was running under so that the pathnames predbat was using could be correctly written in the logfile.

Without this predbat logs that the dashboard, json config, etc are all written to /config which is incorrect.

With Predbat 8.3.3 this is what appears in the logfile when predbat starts up:

```
2024-08-16 12:00:56.839169 INFO pred_bat: --------------- PredBat - update at 2024-08-16 12:00:00+01:00 with clock skew 0 minutes, minutes now 720
2024-08-16 12:00:56.840579 INFO pred_bat: Config root is /config
2024-08-16 12:00:56.868759 INFO pred_bat: Info: Connected to Home Assistant at http://supervisor
2024-08-16 12:00:56.871298 INFO pred_bat: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-16 12:00:56.875611 INFO pred_bat: Info: Web Socket task started
2024-08-16 12:00:56.898148 INFO pred_bat: Config root is /config and printable config_root_p is now /addon_configs/a0d7b954_appdaemon
```

then for example the printable root is shown for example:

2024-08-16 12:01:04.761413 INFO pred_bat: Saved current settings to /addon_configs/a0d7b954_appdaemon/predbat_config.json

With 8.3.5 this now errors:

```
2024-08-18 19:58:10.202466 INFO pred_bat: Config root is /config
2024-08-18 19:58:10.226363 INFO pred_bat: Info: Connected to Home Assistant at http://supervisor/core
2024-08-18 19:58:10.277289 INFO pred_bat: Warn: Failed to decode response <Response [403]> from http://supervisor/core/addons/self/info
2024-08-18 19:58:10.290429 INFO pred_bat: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-18 19:58:10.316719 INFO pred_bat: Info: Web Socket task started
2024-08-18 19:58:10.319563 INFO pred_bat: Config root is /config and printable config_root_p is now /config
```

and subsequent log messages are now incorrect again, e.g.:

2024-08-18 19:58:15.355441 INFO pred_bat: Saved current settings to /config/predbat_config.json

**Expected behavior**
Code introduced to fix 1229 works in both Predbat working in docker or non-docker installs.

**Predbat version**
8.3.5

**Environment details**
HAOSS
Predbat running under appdaemon but same behaviour will occur with predbat add on

**suggested fix**
I think the fix introduced in 8.3.5 to get Predbat working with docker is wrong, in ha.py 8.3.4 the ha_url (if not set in apps.yaml) used to default to "http://supervisor" and this has been changed in 8.3.5 to "http://supervisor/core"

this then breaks the call to determine the slug name around line 53 which appends /addons/self/info to ha_url and makes an API call to find the addon slug name.

In 8.3.4 the API call is being made to http://supervisor/addons/self/info (which works) but in 8.3.5 the call is being made to http://supervisor/core/addons/self/info which isn't the correct API name and so produces error 403.

The fix I believe is to:
1. revert the changes made in 8.3.5 so ha_url defaults to http://supervisor
2. then a change of the ha_url in apps.yaml to point to the correct pathname to HA core.  At the moment in the template is:
```
  # If you are using Predbat outside of HA then set the HA URL or IP address and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'
```
I think ha_url possibly should have been 'http://homeassistant.local:8123/core' but I don't run docker so can't test this.

SwiftRR, 2024-08-18:
I have just restarted predbat
This would explain the error message that I receive in the predbat add-on log linked to supervisor/core.
```
Bootstrap Predbat
Startup
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.6.4
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
Predbat files are installed correctly for version v8.3.5
s6-rc: info: service fix-attrs successfully started
[22:20:57] INFO: Configuring timezone (Europe/London)...
**** Starting Standalone Predbat ****
2024-08-18 22:20:57.615142: Loading apps.yaml
2024-08-18 22:20:57.932686: Info: Connected to Home Assistant at http://supervisor/core
2024-08-18 22:20:57.946010: Warn: Failed to decode response <Response [404]> from http://supervisor/core/addons/self/info
2024-08-18 22:20:57.946689: Creating task: <coroutine object HAInterface.socketLoop at 0x7f914b7680>
2024-08-18 22:20:57.948503: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-18 22:20:57.951855: Info: Web Socket task started
2024-08-18 22:20:57.952634: Warn: unable to find /config/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-18 22:20:58.011531: Info: Web Socket active
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
```
I remember seeing the correct pathname
`/addon_configs/a0d7b954_appdaemon/predbat_config.json`
I can't remember whether this reverted to the /config pathname in v3.8.4 but /config is certainly back for me in v8.3.5:
`2024-08-18 22:25:15.300734: Saved current settings to /config/predbat_config.json`

Rob

gcoan, 2024-08-19:
Hi Nick

Looking at the changelog from 8.3.4 to 8.3.5 https://github.com/springfall2008/batpred/compare/v8.3.4...v8.3.5 there's very little that's @springfall2008 has changed:

- change the default path where HA is expected to be running from http://supervisor to http://supervisor/core
- change the pathnames for HA API calls from /core/api to /api
- introduce ability to configure solcast site id's directly in apps.yaml
- fix car charging filename entry in mkdocs

There were changes made in 8.3.4 from 8.3.3 https://github.com/springfall2008/batpred/compare/v8.3.3...v8.3.4 to move the code I wrote that determines what the slug name is.  This got moved from predbat.py to ha.py and would require a restart of the container/add-on to work (I had to restart my add on as first time through after upgrading I got an error)

Not sure if this is what caused your problem but is the only thing I can think of

SwiftRR, 2024-08-19:
I have just downgraded from v8.3.5 to v8.3.4.
This is the predbat add on log. Compare with the log I posted for v8.3.5 two posts back. The supervisor/core warning was not there in v8.3.4. 
I did have to restart the predbat add-on after downgrading.
```
Bootstrap Predbat
Startup
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.6.4
Predbat files are installed correctly for version v8.3.4
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
**** Starting Standalone Predbat ****
2024-08-19 11:32:31.063931: Loading apps.yaml
s6-rc: info: service fix-attrs successfully started
[11:32:31] INFO: Configuring timezone (Europe/London)...
2024-08-19 11:32:31.373361: Info: Connected to Home Assistant at http://supervisor
2024-08-19 11:32:31.387089: Info: Add-on slug is 6adb4f0d_predbat
2024-08-19 11:32:31.387391: Creating task: <coroutine object HAInterface.socketLoop at 0x7f85173530>
2024-08-19 11:32:31.390151: Info: Web Socket task started
2024-08-19 11:32:31.390997: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-19 11:32:31.395124: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-19 11:32:31.465382: Info: Web Socket active
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
2024-08-19 11:34:15.163840: Info: record_status Idle
```
After my downgrade to v8.3.4, these are the last lines in my predbat.log which look correct for the slug.
With v8.3.5, these lines finished with /config/predbat_config.json (I think!)
```
2024-08-19 11:42:06.527586: Completed run status Idle
2024-08-19 11:42:06.578277: Info: record_status Idle
2024-08-19 11:42:06.623420: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```
So it looks as if v8.3.4 to v8.3.5 has broken things for my predbat add-on on HAOS.

Rob

nipar4, 2024-08-19:
@gcoan if its all that's changed then possibly but this requires a restart every time an upgrade or downgrade is performed, I've tested 8.2.2 => 8.2.3 which used to work OK as well as upgrading to 8.3.4 => 8.3.5 and vice versa

@SwiftRR So if you have to reboot to upgrade maybe its not just a docker problem?

These are the logs I get when upgrading in both environments

First one is where the Predbat restart fails and requires a container restart

2024-08-19 11:53:16.992302: select_event: update, select.predbat_update = v8.3.5 Fix Broken core API when using HA URL
2024-08-19 11:53:16.992424: Calling update service for v8.3.5 Fix Broken core API when using HA URL
2024-08-19 11:53:16.994583: Update Predbat to version v8.3.5 Fix Broken core API when using HA URL
2024-08-19 11:53:18.685080: Kill current threads before update
2024-08-19 11:53:19.584900: Perform the update.....
2024-08-19 11:53:19.602321: Update to version v8.3.5 Fix Broken core API when using HA URL completed
2024-08-19 11:53:19.605148: Info: Web socket stopping
2024-08-19 11:53:19.609518: Info: Web socket stopping
2024-08-19 11:53:28.182845: Refreshing Predbat configuration
2024-08-19 11:53:28.400848: Saved current settings to /config/predbat_config.json

Second one works successfully when predbat is run within /Config

2024-08-19 13:08:24.222473: select_event: update, select.predbat_update = v8.3.5 Fix Broken core API when using HA URL
2024-08-19 13:08:24.222608: Calling update service for v8.3.5 Fix Broken core API when using HA URL
2024-08-19 13:08:24.224947: Update Predbat to version v8.3.5 Fix Broken core API when using HA URL
2024-08-19 13:08:26.492814: Kill current threads before update
2024-08-19 13:08:27.592724: Perform the update.....
2024-08-19 13:08:27.609599: Update to version v8.3.5 Fix Broken core API when using HA URL completed
2024-08-19 13:08:27.612079: Info: Web socket stopping
2024-08-19 13:08:27.631802: Info: Web socket stopping
2024-08-19 13:08:27.653909: Stopping Predbat
2024-08-19 13:08:27.654036: Predbat terminating
2024-08-19 13:08:27.654201: Predbat terminated
2024-08-19 13:09:03.353653: Loading apps.yaml
2024-08-19 13:09:03.367070: Predbat: Startup predbat

Notice the last few lines terminating predbat are not in the first log

Cheers
Nic

SwiftRR, 2024-08-19:
For the completed picture, I have just updated again from v8.3.4 to v8.3.5. 
add-on log below **No restart needed.**
```
Startup
Predbat files are installed correctly for version v8.3.5
**** Starting Standalone Predbat ****
2024-08-19 13:39:37.029681: Loading apps.yaml
2024-08-19 13:39:37.220030: Info: Connected to Home Assistant at http://supervisor/core
2024-08-19 13:39:37.231183: Warn: Failed to decode response <Response [404]> from http://supervisor/core/addons/self/info
2024-08-19 13:39:37.231795: Creating task: <coroutine object HAInterface.socketLoop at 0x7f9bcc7680>
2024-08-19 13:39:37.232777: Info: Web Socket task started
2024-08-19 13:39:37.233391: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-19 13:39:37.236523: Warn: unable to find /config/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-19 13:39:37.275160: Info: Web Socket active
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
2024-08-19 13:41:15.197278: Info: record_status Idle
2024-08-19 13:41:30.211233: Info: record_status Idle
```
With v8.3.5, these are last lines in my predbat.log. Compare with these lines for v8.3.4 which has the slug in giving: /addon_configs/6adb4f0d_predbat/predbat_config.json
```
2024-08-19 13:41:30.165247: Completed run status Idle
2024-08-19 13:41:30.211233: Info: record_status Idle
2024-08-19 13:41:30.255861: Saved current settings to /config/predbat_config.json
```
I am now going to downgrade again. See next post.

Rob

SwiftRR, 2024-08-19:
And back to v8.3.4 from v8.3.5.
add-on log below. **No restart needed**. _I must have got impatient before!_
```
Startup
Predbat files are installed correctly for version v8.3.4
**** Starting Standalone Predbat ****
2024-08-19 13:49:52.192874: Loading apps.yaml
2024-08-19 13:49:52.367880: Info: Connected to Home Assistant at http://supervisor
2024-08-19 13:49:52.377946: Info: Add-on slug is 6adb4f0d_predbat
2024-08-19 13:49:52.378161: Creating task: <coroutine object HAInterface.socketLoop at 0x7fb7e73530>
2024-08-19 13:49:52.379156: Info: Web Socket task started
2024-08-19 13:49:52.379825: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-19 13:49:52.384213: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-19 13:49:52.419063: Info: Web Socket active
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
2024-08-19 13:51:32.385885: Info: record_status Idle
2024-08-19 13:51:47.173669: Info: record_status Idle
```
And with v8.3.4, my slug is back in predbat.log
```
2024-08-19 13:55:14.954923: Completed run status Idle
2024-08-19 13:55:15.003363: Info: record_status Idle
2024-08-19 13:55:15.047884: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```
Rob

SwiftRR, 2024-08-19:
@gcoan I don't know enough about the coding here (one of my projects is to be able to understand why is going on.)
But looking at the changelog 8.3.4 to 8.35, line 56 from 8.3.4 has been blanked out in 8.3.5.
Is this the reason for the issue or am I trying to run before I can walk .......

Rob

gcoan, 2024-08-19:
> line 56 from 8.3.4 has been blanked out in 8.3.5.

the line you are referring to is:
                    self.config_root_p = "/addon_configs/" + res["data"]["slug"]

that's actually OK, it was a cut and paste error introduced in 8.3.4 when Trefor moved the logic that works out the printable config path name from predbat.py to a function get_slug of ha.py.  The line is setting the printable version of the config root and it didn't need to be the predbat init module as it was already in line 10869 of predbat.py

If you look at the compare of 8.3.3 to 8.3.5 the code is fine https://github.com/springfall2008/batpred/compare/v8.3.3...v8.3.5

The problem is line 53 where after connecting to HA it makes an API call to HA:

SwiftRR, 2024-08-19:
@gcoan Thanks for the explanation although how much I understood is debatable!
If nothing else, I have trialled the standalone predbat add-on through v8.3.3, v8.3.4 and v8.3.5.
This confirms your comments in your original post: 
> **Environment details**
> HAOSS
> Predbat running under appdaemon but same behaviour will occur with predbat add on.

I am now of v8.3.4 and I get the correct path: /addon_configs/6adb4f0d_predbat/predbat_config.json

I will now allow greater minds than mine to move this forwards and I look forwards to v8.3.6 

Rob

gcoan, 2024-08-19:
> Thanks for the explanation although how much I understood is debatable!

😁

> I will now allow greater minds than mine to move this forwards to v8.3.6 ..........

Since I don't have docker I wanted Trefor's input rather than me changing it wrongly

springfall2008, 2024-08-20:
> > line 56 from 8.3.4 has been blanked out in 8.3.5.
> 
> the line you are referring to is: self.config_root_p = "/addon_configs/" + res["data"]["slug"]
> 
> that's actually OK, it was a cut and paste error introduced in 8.3.4 when Trefor moved the logic that works out the printable config path name from predbat.py to a function get_slug of ha.py. The line is setting the printable version of the config root and it didn't need to be the predbat init module as it was already in line 10869 of predbat.py
> 
> If you look at the compare of 8.3.3 to 8.3.5 the code is fine [v8.3.3...v8.3.5](https://github.com/springfall2008/batpred/compare/v8.3.3...v8.3.5)
> 
> The problem is line 53 where after connecting to HA it makes an API call to HA: res = self.api_call("/addons/self/info")
> 
> In the api_call function:
> 
> ```
>     def api_call(self, endpoint, data_in=None, post=False):
>         """
>         Make an API call to Home Assistant.
> 
>         :param endpoint: The API endpoint to call.
>         :param data_in: The data to send in the body of the request.
>         :param post: True if this is a POST request, False for GET.
>         :return: The response from the API.
>         """
>         url = self.ha_url + endpoint
>         headers = {
>             "Authorization": "Bearer " + self.ha_key,
>             "Content-Type": "application/json",
>             "Accept": "application/json",
>         }
>         if post:
>             if data_in:
>                 response = requests.post(url, headers=headers, json=data_in, timeout=TIMEOUT)
>             else:
>                 response = requests.post(url, headers=headers, timeout=TIMEOUT)
>         else:
>             if data_in:
>                 response = requests.get(url, headers=headers, params=data_in, timeout=TIMEOUT)
>             else:
>                 response = requests.get(url, headers=headers, timeout=TIMEOUT)
>         try:
>             data = response.json()
> ```
> 
> the url for the API call is built as: url = self.ha_url + endpoint
> 
> ha_url and then the endpoint you're trying to call.
> 
> ha_url (if not set in apps.yaml) now defaults to 'http://supervisor/core' and when trying to find the slugname the api endpoint is '/addons/self/info' so the API path is built as 'http://supervisor/core/addons/self/info' whereas it should be 'http://supervisor/addons/self/info'

Hmm okay this is odd as it worked for me, I'll take another look.

The issue was someone (maybe gcoan) removed the 'core' bit from the default URL which broken the addon method (non-docker). If you set ha_url then it has no impact. 

What will be the path to addons/self/info with a HA URL set?

gcoan, 2024-08-20:
> The issue was someone (maybe gcoan) removed the 'core' bit from the default URL which broken the addon method (non-docker). If you set ha_url then it has no impact.

I removed the 'core' bit of the default HA URL and added it to all the API calls being made to HA as otherwise I couldn't call /addons/self/info.  This works fine with predbat add-on, appdaemon or appdaemon-predbat add-on's.

Its worth noting that I'm trying to make a supervisor API call not a HA API call and the full list of supervisor API calls is at https://developers.home-assistant.io/docs/api/supervisor/endpoints/ 

> What will be the path to addons/self/info with a HA URL set?

The full path should be http://supervisor/addons/self/info

So to my thinking 'http://supervisor' is the HA (supervisor) URL

And '/addons/self/info' is the API path to call, in the same way '/core/api/' is the call for a HA core API call.

I think the problem is that for docker instances the ha_url in apps.yaml was set in a way that the code wasn't expecting and so the API calls in predbat were no longer working under docker.  As I said I don't have docker so no easy way for me to test this.  The fix being to either change apps.yaml or change the way predbat makes API calls so it can distinguish between supervisor and HA API calls.
When running under docker where there isn't a HAOS supervisor running I expect that the supervisor API call to /addons/self/info will fail, but that's OK, it will still leave config_root_p defaulting to /config, but want it still to work for add-on configurations

springfall2008, 2024-08-21:
If I set ha_url I get this:

2024-08-21 19:53:28.418052: Warn: Failed to decode response <Response [404]> from http://homeassistant.local:8123/addons/self/info

What should the path be if you set the HA URL?

gcoan, 2024-08-21:
> If I set ha_url I get this:
> 
> 2024-08-21 19:53:28.418052: Warn: Failed to decode response <Response [404]> from http://homeassistant.local:8123/addons/self/info
> 
> What should the path be if you set the HA URL?

I suspect if you are running HASS then ha_url should be set to http://supervisor so the above API call can be made to find out the add-on slug name from the supervisor and any HA API calls will be to /core/api/... and so be proxied by the supervisor to HA,

If you are running HA as a docker container then it should be the path to the docker environment minus the 'core' bit, so http://homeassistant.local:8123 ?

(all of which assumes that the /core path gets added back in to Predbat, reverting the 8.3.4 pathname change)

springfall2008, 2024-08-21:
Okay Fix on main for testing

gcoan, 2024-08-21:
@springfall2008 it looks to be producing the same result (I don't have ha_url set BTW):

2024-08-21 21:42:22.039416 INFO pred_bat: Info: Connected to Home Assistant at http://supervisor/core
2024-08-21 21:42:22.048538 INFO pred_bat: Warn: Failed to decode response <Response [403]> from http://supervisor/core/addons/self/info
2024-08-21 21:42:22.051226 INFO pred_bat: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-21 21:42:22.058510 INFO pred_bat: Info: Web Socket task started
2024-08-21 21:42:22.059500 INFO pred_bat: Config root is /config and printable config_root_p is now /config

gcoan, 2024-08-21:
I was thinking about something similar to your fix #1382, distinguishing between the API calls that need to be made to the supervisor and those that need to be made to HA, thanks

SwiftRR, 2024-08-21:
I have installed main using standalone predbat add-on.
Here is my predbat add-on log. Looks consistent with [gcoan](https://github.com/gcoan) 
I have then shown the end of my predbat.log and results look good. the same as with v8.3.4.
**predbat add-on log**
```
[22:38:19] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: 2fea88973da165c8983b3c772daf4a9072d8fa44759281865433a9cda05c211605696cdfd92ae9b53c39a009fb8e2842fa78cef01bcb9787
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.3.6
**** Starting Standalone Predbat ****
2024-08-21 22:38:21.375973: Loading apps.yaml
2024-08-21 22:38:21.560986: Info: Connected to Home Assistant at http://supervisor/core
2024-08-21 22:38:21.571574: Info: Add-on slug is 6adb4f0d_predbat
2024-08-21 22:38:21.571773: Creating task: <coroutine object HAInterface.socketLoop at 0x7f88697680>
2024-08-21 22:38:21.573079: Info: Web Socket task started
2024-08-21 22:38:21.573926: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-21 22:38:21.576987: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-21 22:38:21.613401: Info: Web Socket active
Watching ['./config.py', './hass.py', './inverter.py', './utils.py', './download.py', './prediction.py', './predbat.py', './ha.py', './unit_test.py', './apps.yaml'] for changes
2024-08-21 22:40:16.382415: Info: record_status Idle
2024-08-21 22:40:31.035921: Info: record_status Idle
```
**End of predbat.log**
```2024-08-21 22:40:30.988995: Completed run status Idle
2024-08-21 22:40:31.035921: Info: record_status Idle
2024-08-21 22:40:31.081645: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```

Rob

nipar4, 2024-08-22:
I'm no expert with ha and I don't pretend to understand exactly what your trying to do but I am not sure if this will work in a docker environment.  I think the API for HA in a container is very different to the HASOS Supervisor API plus from what I've seen the supervisor API is only available in the local home assistant environment and seeing as most docker instals will either have ha installed in its own docker subnet (not on the same network as predbat) or bound to the docker host depending on the skills of the user it will be difficult to know what we're dealing with to make it work for all situations

gcoan, 2024-08-22:
> I'm no expert with ha and I don't pretend to understand exactly what your trying to do but I am not sure if this will work in a docker environment.

I think the code changes Trefor has introduce in the main branch should be OK for docker.  Assuming that 8.3.5 works on docker then this will work just as well as the only change is to distinguish between the 'set printable config path' bit that calls the HASS supervisor to find the slug name. In non-HASS environments this will fail but defaults the printable config path (shown in the logfile output) to /config.
Users have to set ha_url in apps.yaml to point to the URL that HA is installed on, that all still works the same.

If you have a docker install, can you try the main branch?

nipar4, 2024-08-22:
My docker install of predbat seems to be working fine with the main branch, I have been running it most of the morning

I am getting this error, which I have seen over the last few versions or so but have ignored up until now as I assumed it was just for HA addons but if this also is needed for Docker in some way maybe I need to understand more

2024-08-22 10:05:09.357163: Info: Connected to Home Assistant at http://192.168.1.2:8123
2024-08-22 10:05:09.361590: Warn: Failed to decode response <Response [404]> from http://192.168.1.2:8123/addons/self/info

From what I read from this thread this is predbat trying to identify the path to the config directory under the HA addons structure using what would be the supervisor API. what I don't really understand is how it falls back to /config to cater for Docker

gcoan, 2024-08-22:
Great that it works under docker, thanks for confirming.

The error is as you say about Predbat trying to find the add-on slug name (i.e. app name) that Predbat is running under, as a docker container this will fail, but the fail can be ignored.

Maybe it should be trapped with a better message that makes clear it is just a warning, but in essence the code is:

config_path = "/config"
config_path_p = config_path

ha_url = [whatever is set in apps.yaml for docker users], or default to "http://supervisor/core"

add_on_api_call = (ha_url - "/core") + "/addons/self/info"

make api call(url=add_on_api_call)
if no error
    config_path_p = "/addon_configs/" + [slug name from API call]
else
    report error from api call

The code now has two paths, config_path which is used when constructing file names that predbat needs to write to, e.g. the dashboard.yaml or the predbat config,  and config_path_p is the printable version of the pathname which is the actual pathname that predbat is writing to.

All of this comes about because when you run an add-on in HA, the /config directory presented to the add-on is actually a symbolic link to the real directory under /addon_configs, e.g /addon_configs/6adb4f0d_predbat for the predbat addon

This meant that the pathname written to the logfile was wrong as it said the files were written to /config when actually they weren't.
With docker you don't have this mucking about.

nipar4, 2024-08-22:
OK I'm following you so far, so the 

else
report error from api call

Should be a better error to indicate that its not an addon but a docker container or something

Is config_path_p just used for logging information or is it used to tell predbat where to put the logs etc. as well?

gcoan, 2024-08-22:
> Should be a better error to indicate that its not an addon but a docker container or something

Yes I think the error should say something like 'Warning: Unable to determine HASS add-on name Predbat is running under, check ha_url or ignore if Predbat is running under docker'

> Edit: thought about this and you said in the addon its under the /config directory same as docker so the questions not valid

Predbat writes all its files to /config regardless of how Predbat is running.

If Predbat is running under a HASS add-on then the add-on maps (symbolically links) /config to /addon_configs/[add-on slug name] so the code thinks it is writing to /config but in reality the files are written to the addon_configs subdirectory.  I tried to see if I could retrieve the drive mapping table from python code to work out where /config was actually mounted, but couldn't.  This reading the addon name (HA calls it the slug name) is the only way I could find to determine what the true pathname is

> Is config_path_p just used for logging information or is it used to tell predbat where to put the logs etc. as well?

config_path_p is only used for logging information, its not the actual log path which remains constant at /config.  Rather than having to introduce a whole parallel set of printable pathnames I did try changing config_path to point to /addon_configs/[slug] and get predbat to write directly in there, but it didn't have permissions to do so! So hence the solution adopted

springfall2008, 2024-08-22:
>is ha_url used for anything else? I assumed it was how predbat talked to Ha to update its entities etc.

If you are using Docker or Predbat standalone its used to communicate with HA.

springfall2008, 2024-08-22:
New release has been made

gcoan, 2024-08-22:
Installed 8.3.6 and is working fine.  Thanks Trefor

closed 2024-08-22 by gcoan