springfall2008/batpred #1375: Solis inverter via Solis Cloud - getting  'NoneType' object has no attribute 'split'

chris-y, 2024-08-19:
**Describe the bug**
I'm trying to set up predbat for my Solis inverter but using Solis Cloud rather than modbus.
Monitoring is working great, but I'm not able to do any control.  If I test any control service nothing happens, and I'm seeing ` 'NoneType' object has no attribute 'split'` in the log and status.

I have set read-only to off and set a forced charge, then waited for the start of the forced charge to see if the charge_start_service triggers - it doesn't - and the only thing which looks relevant in the log is this NoneType error.

**Expected behavior**
I expect my service to be called.

**Predbat version**

1.1.8

**Environment details**

- Solis inverter via Solis Cloud
Monitoring using https://github.com/hultenvp/solis-sensor
Control using (when I get it working) https://github.com/stevegal/solis_control
- HAOS Pi image

**Screenshots**
[image]
[image]

**Log file**
```
    self.set_current_from_power(direction, 0)
  File "/config/inverter.py", line 1789, in set_current_from_power
    self.write_and_poll_value(f"timed_{direction}_current", self.base.get_arg(f"timed_{direction}_current", indirect=False, index=self.id), new_current, fuzzy=1)
  File "/config/inverter.py", line 1233, in write_and_poll_value
    domain, entity_name = entity_id.split(".")
AttributeError: 'NoneType' object has no attribute 'split'
```

**Config**

The charge_start_service stuff is what I was trying out.  The first one runs a script which has 'power' as an input and just notifies me.  This never triggers.  The second runs the pyscript which controls the inverter, but I'll probably set this up as a script which takes the parameters to get the config out of here.  For now though I was just trying to get it to trigger anything.

```
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # XXX: This is a configuration template, delete this line once you edit your configuration
  #template: True

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Timezone to work in
  timezone: Europe/London

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #

  inverter_type: MINE
  inverter:
    name: "Solis Inverter"
    has_rest_api: False
    has_mqtt_api: False
    has_service_api: True
    output_charge_control: "current"
    current_dp: 0
    has_charge_enable_time: False
[... 229 lines cut ...]
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
    - name: 'large'
      minutes: 60
      energy: 1.0
    - name: 'small'
      minutes: 15
      energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0
```

springfall2008, 2024-08-21:
You appear to be missing settings for charge/discharge current as below:

  timed_charge_current:
    - number.solis_timed_charge_current
  timed_discharge_current:
    - number.solis_timed_discharge_current

chris-y, 2024-08-21:
Is that literally what I need? Or do I need a helper or value or something to plug into it? 

As I'm still getting what looks like the same error: 

```
2024-08-21 20:57:10.365633: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9820, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/predbat.py", line 8394, in execute_plan
    inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)
  File "/config/inverter.py", line 1985, in adjust_charge_window
    self.enable_charge_discharge_with_time_current("charge", True)
  File "/config/inverter.py", line 1773, in enable_charge_discharge_with_time_current
    self.set_current_from_power(direction, power)
  File "/config/inverter.py", line 1789, in set_current_from_power
    self.write_and_poll_value(f"timed_{direction}_current", self.base.get_arg(f"timed_{direction}_current", indirect=False, index=self.id), new_current, fuzzy=1)
  File "/config/inverter.py", line 1233, in write_and_poll_value
    domain, entity_name = entity_id.split(".")
AttributeError: 'NoneType' object has no attribute 'split'

2024-08-21 20:57:10.409585: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'split'
2024-08-21 20:57:10.409814: Error: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/config/hass.py", line 173, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 10962, in update_time_loop
    raise e
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9820, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/predbat.py", line 8394, in execute_plan
    inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)
  File "/config/inverter.py", line 1985, in adjust_charge_window
    self.enable_charge_discharge_with_time_current("charge", True)
  File "/config/inverter.py", line 1773, in enable_charge_discharge_with_time_current
    self.set_current_from_power(direction, power)
  File "/config/inverter.py", line 1789, in set_current_from_power
    self.write_and_poll_value(f"timed_{direction}_current", self.base.get_arg(f"timed_{direction}_current", indirect=False, index=self.id), new_current, fuzzy=1)
  File "/config/inverter.py", line 1233, in write_and_poll_value
    domain, entity_name = entity_id.split(".")
AttributeError: 'NoneType' object has no attribute 'split'
```

chris-y, 2024-08-21:
OK, I worked that out - I needed a helper created, and this then pokes the value into that. Discharge always seems to be 0, even when I set a forced discharge, is this correct?

I also worked out why my service wasn't called - I'd set it at the wrong level. 

Question: Is the {power} always in watts? Is there a way of getting this in amps? Do I just use the current helper I specified in timed_charge_current?

springfall2008, 2024-08-22:
These are in AMPS (current) rather than watts. The timed_discharge_current should not be zero during discharge.

chris-y, 2024-08-23:
I tried a live test last night, it was supposed to charge the battery a little but when I woke up it had charged to 100%.  It then had planned a discharge but the discharge start/end time hadn't been set.  I fixed that by setting `charge_discharge_with_rate: False`.

So I think the only outstanding issue is that neither the charge/discharge rate in Watts or Amps is changing.
[image]
[image]

The charge rate is a bit weird as it looks like it did change overnight yesterday, but not last night:
[image]

chris-y, 2024-08-23:
Further update.  The charging current is now updating - not sure what I changed to kick it into action!  Discharge current is still stuck on 0.

However, it looks like the charge_start_service etc aren't called after the rate changes, so the rate sent to the inverter is prior to the change.  I observed this earlier when it was in charge mode, and now it's gone to hold but hasn't updated with 0A:

Plan is on HOLD:
[image]

Charge rate has dropped to 0A:
[image]

The last call of the service has happened before the drop: (this screenshot taken after the two above)
[image]

Also I notice there's a 1 minute overlap on the discharge stop and charge start times.

eddysteurs, 2024-10-31:
Chris-y : I managed to get my predbat also working with soliscloud but it seems to call  the STOP CHARGE service every 5 minutes and as such sending a command to the inverter (while not needed as there is no status change), do you have this aswell?

chris-y, 2024-10-31:
Yes, both the start and stop are called every five minutes (sometimes twice) depending on whether it is active or not.
I'm not sure how to stop this, might be a Predbat bug?

eddysteurs, 2024-11-02:
I have requested a predbat update for this , tested it and it seems to work, will come in 8.5.4 version

Fayrewood, 2024-11-03:
Hi @eddysteurs  and @chris-y ,

Could you help me out please?

I am getting an error in HA when creating a simple script based on Eddy's example above of "Message malformed: extra keys not allowed @ data['sequence'][0]['fields']"  I've checked indentation etc but can't see the issue.  I have gone into HA / Scripts and added a blank script and pasted from Eddy's example and added my API keys and times.

I have Solis Cloud installed and connected to the cloud servers and reporting the inverter information OK and have installed the Solis Control .py file in the pyscripts folder. 

I am using a simple set of times to test connectivity for now - can you see what I may be doing wrong?

```
alias: Set Solis Inverter Config (Predbat)
sequence:
  - metadata: {}
    data:
      days:
        - chargeCurrent: "50"
          dischargeCurrent: "50"
          chargeStartTime: "01:05"
          chargeEndTime: "15:55"
          dischargeStartTime: "16:05"
          dischargeEndTime: "18:55"
        - chargeCurrent: "50"
          dischargeCurrent: "50"
          chargeStartTime: "00:00"
          chargeEndTime: "00:00"
          dischargeStartTime: "00:00"
          dischargeEndTime: "00:00"
        - chargeCurrent: "50"
          dischargeCurrent: "50"
          chargeStartTime: "00:00"
          chargeEndTime: "00:00"
          dischargeStartTime: "00:00"
          dischargeEndTime: "00:00"
      config:
        secret: "cc8165edfredacted"
        key_id: "1300386redacted"
        username: "ben@redacted"
        password: "redacted"
        plantId: "redacted"
    action: pyscript.solis_control
fields: {}
mode: single
icon: mdi:home-battery
description: ""

```

eddysteurs, 2024-11-03:
> Hi @eddysteurs and @chris-y ,
> 
> Could you help me out please?
> 
> I am getting an error in HA when creating a simple script based on Eddy's example above of "Message malformed: extra keys not allowed @ data['sequence'][0]['fields']" I've checked indentation etc but can't see the issue. I have gone into HA / Scripts and added a blank script and pasted from Eddy's example and added my API keys and times.
> 
> I have Solis Cloud installed and connected to the cloud servers and reporting the inverter information OK and have installed the Solis Control .py file in the pyscripts folder.
> 
> I am using a simple set of times to test connectivity for now - can you see what I may be doing wrong? If I use the data and config elements directly in Dev Tools / Call Service, it works for me, but I can't get my head around the yaml for the script.
> 
> ```
> alias: Set Solis Inverter Config (Predbat)
> sequence:
>   - metadata: {}
>     data:
>       days:
>         - chargeCurrent: "50"
>           dischargeCurrent: "50"
>           chargeStartTime: "01:05"
>           chargeEndTime: "15:55"
>           dischargeStartTime: "16:05"
>           dischargeEndTime: "18:55"
>         - chargeCurrent: "50"
>           dischargeCurrent: "50"
>           chargeStartTime: "00:00"
>           chargeEndTime: "00:00"
>           dischargeStartTime: "00:00"
>           dischargeEndTime: "00:00"
>         - chargeCurrent: "50"
>           dischargeCurrent: "50"
>           chargeStartTime: "00:00"
>           chargeEndTime: "00:00"
>           dischargeStartTime: "00:00"
>           dischargeEndTime: "00:00"
>       config:
>         secret: "cc8165edfredacted"
>         key_id: "1300386redacted"
>         username: "ben@redacted"
>         password: "redacted"
>         plantId: "redacted"
>     action: pyscript.solis_control
> fields: {}
> mode: single
> icon: mdi:home-battery
> description: ""
> ```
> 
> I get the following in the HA Logs if I run the script:
> 
> ```
> Logger: custom_components.pyscript.function
> Source: custom_components/pyscript/function.py:453
> integration: Pyscript Python scripting ([documentation](https://github.com/custom-components/pyscript), [issues](https://github.com/custom-components/pyscript/issues))
> First occurred: 12:22:22 (7 occurrences)
> Last logged: 12:52:03
> 
> run_coro: got exception Traceback (most recent call last): File "/config/custom_components/pyscript/eval.py", line 755, in call raise TypeError(f"{self.name}() called with unexpected keyword arguments: {unexpected}") TypeError: solis_control() called with unexpected keyword arguments: data
> run_coro: got exception Traceback (most recent call last): File "/config/custom_components/pyscript/eval.py", line 755, in call raise TypeError(f"{self.name}() called with unexpected keyword arguments: {unexpected}") TypeError: solis_control() called with unexpected keyword arguments: action, data, metadata
> ```
> 
> Many thanks! Pete

eddysteurs, 2024-11-03:
i copied and pasted your script in my HA and i works fine, no errors, 
created a new SCRIPT , put it in yaml and pasted your content, so I have no clue what the problem is, sorry

chris-y, 2024-11-03:
Looks OK to me too. Be aware that for your script you need to edit the entire script in yaml, not just the sequence, as you only get part of the yaml then. 

Is the pyscript installed correctly? Did you enable the global variables etc that the script requires?

Fayrewood, 2024-11-03:
Thanks for the prompt responses chaps.  Yes, both of the global variables are enabled.   Now, that's bloody weird.  I copied the code from the script (in yaml mode) and created a new one, and hey presto, it's working!.  Thanks!!

Fayrewood, 2024-11-04:
@eddysteurs  / @chris-y  - I have created a Number helper for input_number.solis_battery_charge_rate_predbat but I am getting errors.  What helpers did you guys create and were they number or text helpers?  Did you have to set an initial value, as mine is currently at zero.

My errors:

```ValueError: Template error: int got invalid input 'unknown' when rendering template '0{{ states('input_number.solis_battery_charge_rate_predbat') | int | string }}' but no default was specified
```
and
```
Set Solis Inverter Config (Predbat): Error executing script. Error for call_service at pos 1: Error rendering data template: ValueError: Template error: int got invalid input 'unknown' when rendering template '0{{ states('input_number.solis_battery_charge_rate_predbat') | int | string }}' but no default was specified
```

```
PredBat Logs

38171 | 2024-11-04 08:40:09.599798: Error: float() argument must be a string or a real number, not 'NoneType'
-- | --
38170 | 2024-11-04 08:40:09.599304: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
38168 | TypeError: float() argument must be a string or a real number, not 'NoneType'
38151 | 2024-11-04 08:40:09.561165: Error: Traceback (most recent call last):
38150 | 2024-11-04 08:40:09.560009: Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
```

Thanks,
Pete

eddysteurs, 2024-11-04:
Normal number helper
[image]
my script
[image]

and in predbat apps.yaml :

  charge_start_service: 
    - service: script.set_solis_inverter_config_predbat_start_charge
  discharge_start_service: 
    - service: script.set_solis_inverter_config_predbat_start_discharge
  charge_stop_service: 
    - service: script.set_solis_inverter_config_predbat_stop_charge
  discharge_stop_service: 
    - service: script.set_solis_inverter_config_predbat_stop_discharge

  timed_charge_current:
    - input_number.solis_load_from_grid_ampere_predbat
  timed_discharge_current:
    - input_number.solis_load_to_grid_ampere_predbat

Fayrewood, 2024-11-04:
Thanks @eddysteurs.  I'll have a play with the helpers.

I have the same service call for each of the charge & discharge services, following the templates near the top of this thread.  I see that you have different service scripts for each.  Are they all duplicates of one script, with the same parameters as per your screenshot?

I'm not familiar with Solis inverters, other than trying to get this set up for a friend, but I noticed that your discharge current is "001", which is different from the "80" in @chris-y 's template.  Any pointers on what the current should be?  It's a 5kw hybrid inverter.

Here's my current apps.yaml.

```
charge_start_service:
    - service: script.set_solis_inverter_config_predbat
  discharge_start_service:
    - service: script.set_solis_inverter_config_predbat
  charge_stop_service:
    - service: script.set_solis_inverter_config_predbat
  discharge_stop_service:
    - service: script.set_solis_inverter_config_predbat

  timed_charge_current:
    - input_number.solis_battery_charge_rate_predbat
  timed_discharge_current:
    - input_number.solis_battery_discharge_rate_predbat
```

chris-y, 2024-11-04:
My charge and discharge scripts are the same - it has to send all this information at once and if you split them and provide just the charge or just the discharge values it will overwrite the other - so it's easier to keep them as the same script and set both.
My "stop" script I'm not sure is necessary, I only added it afterwards to ensure the inverter was stopping when Predbat thought it should, as I was getting weird spikes in current as it stopped.

It should be picking the discharge current up from the helper - Predbat will set this for you.  It looks like @eddysteurs may have split it into a charge and discharge script but as per above I don't see any need to do that.  Mine was hardcoded to 80 as that's the maximum it will export (generally; I think it can do 100A but there's a risk of things tripping apparently, so I set it to 80 to be safe), and the discharge current helper wasn't working correctly at that time.  That was fixed so I've since changed it to `dischargeCurrent: "  0{{ states('input_number.solis_battery_discharge_rate_predbat') | int | string }} "`, and Predbat tends to pick 72A.

closed 2026-08-28 by chalfontchubby