springfall2008/batpred #1378: Iboost Today Attribute Remains At Zero

nbullus, 2024-08-20:
**Describe the bug**
The **predbat.iboost_best** entity contains the total predicted iBoost charge across the whole Predbat plan (36 hours in my case). The **today** attribute on this sensor should hold just the predicted iBoost total for today, but remains at zero.

**Expected behavior**
 The **today** attribute on this sensor should update at midnight and hold the predicted iBoost total for today. In the example screenshots below, this should be set as 6.5 kWh for today and update to 9.75 kWh at midnight. The attribute should also be updated if the the predicated iBoost charge changes with any plan changes.

**Predbat version**

8.3.5

**Environment details**

- GivEnergy AIO
- Standard HAOS installer

**Screenshots**

[image]

[image]

[image]

**Log file**

Not needed. Trefor has already established the root cause.

springfall2008, 2024-08-24:
Fixed on main for the next release

nbullus, 2024-08-25:
Tested on main and planned iBoost over next 36 hrs shows correct figures in Today attribute.

springfall2008, 2024-08-25:
Release as v8.3.7

closed 2024-08-25 by springfall2008