springfall2008/batpred #1380: Exception raised 'Battery_Charge_Rate'

LeeFarm11, 2024-08-20:
**Describe the bug**
Fresh install. First run. Giving error in Log File 
Inverter_limit set to 11500, but I now don't think it is relevant.

2024-08-20 22:01:06.652055: Reserve min: 4.0% Battery_min:4.0%
2024-08-20 22:01:06.652288: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 11000.0 w charge rate 11.0 kW discharge rate 11.0 kW battery_rate_min 200.0 w ac limit 11.52 kW export limit 11.52 kW reserve 4.0 % current_reserve 4.0 %
2024-08-20 22:01:06.720213: Error: Exception raised 'Battery_Charge_Rate'
2024-08-20 22:01:06.727637: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9800, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9311, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 746, in update_status
    self.charge_rate_now = self.rest_data["Control"]["Battery_Charge_Rate"] / MINUTE_WATT
KeyError: 'Battery_Charge_Rate'

2024-08-20 22:01:06.761569: Info: record_status Error: Exception raised 'Battery_Charge_Rate'
2024-08-20 22:01:06.761805: Error: 'Battery_Charge_Rate'
2024-08-20 22:05:00.230715: --------------- PredBat - update at 2024-08-20 22:05:00+01:00 with clock skew 0 minutes, minutes now 1325

**Expected behavior**
Not sure why I am getting this error. Is it a problem in predbat or in GivTCP?

**Predbat version**
1.1.8

**Environment details**
3 phase 11kw Gen 3 Inverter
6 batteries total 20.4kwh
18.7kW Panels

**Screenshots**
Here is an example of the battery being charged at over 11kw.
[image]

**Log file**
[predbat(2).log](https://github.com/user-attachments/files/16682053/predbat.2.log)
[apps.yaml.zip](https://github.com/user-attachments/files/16682056/apps.yaml.zip)

gcoan, 2024-08-20:
Are you running the beta version of givtcp?

There still seems to be work in progress changes for givtcp to support 3 phase inverters
Eg see https://github.com/britkat1980/giv_tcp/issues/186

LeeFarm11, 2024-08-21:
Thanks Geoff. Now raised with britkat1980.

gcoan, 2024-08-21:
@LeeFarm11 

Just had an idea, try setting charge rate manually in apps.yaml

AFAIK Predbat only uses the charge rate from the inverter when setting the inverter to max charge so if you over-wrote it in apps.yaml it should over-ride the non-existence of this entity from GivTCP.

But still would be good to get it fixed in givtcp

LeeFarm11, 2024-08-21:
Like this?

[image]

gcoan, 2024-08-21:
yes.  does it work?

or create a number helper and set that to 11000

LeeFarm11, 2024-08-21:
Unfortunately no. 

`-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on: predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.1.8
 You are running the latest version of this add-on.
Predbat files are installed correctly for version v8.3.5
**** Starting Standalone Predbat ****
2024-08-21 12:55:07.526836: Loading apps.yaml
2024-08-21 12:55:07.827884: Info: Connected to Home Assistant at http://supervisor/core
2024-08-21 12:55:07.839082: Warn: Failed to decode response <Response [404]> from http://supervisor/core/addons/self/info
2024-08-21 12:55:07.840473: Creating task: <coroutine object HAInterface.socketLoop at 0x7f1a8a83f610>
2024-08-21 12:55:07.842647: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-21 12:55:07.843612: Info: Web Socket task started
2024-08-21 12:55:07.844287: Warn: unable to find /config/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-21 12:55:07.883652: Info: Web Socket active
2024-08-21 12:55:08.023190: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
2024-08-21 12:55:08.029784: Warn: Regular expression argument: pv_forecast_today unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_today), now will disable
2024-08-21 12:55:08.035827: Warn: Regular expression argument: pv_forecast_tomorrow unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow), now will disable
2024-08-21 12:55:08.041883: Warn: Regular expression argument: pv_forecast_d3 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3)), now will disable
2024-08-21 12:55:08.048267: Warn: Regular expression argument: pv_forecast_d4 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4)), now will disable
2024-08-21 12:55:08.054826: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-08-21 12:55:08.062159: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-08-21 12:55:08.067171: Warn: Regular expression argument: octopus_intelligent_slot unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_intelligent_dispatching), now will disable
2024-08-21 12:55:08.072823: Warn: Regular expression argument: octopus_ready_time unable to match re:(time.octopus_energy([0-9a-z_]+|)_intelligent_ready_time), now will disable
2024-08-21 12:55:08.078385: Warn: Regular expression argument: octopus_charge_limit unable to match re:(number.octopus_energy([0-9a-z_]+|)_intelligent_charge_limit), now will disable
2024-08-21 12:55:08.084929: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-08-21 12:55:08.090520: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
2024-08-21 12:55:08.097132: Warn: Regular expression argument: metric_octopus_import unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate), now will disable
2024-08-21 12:55:08.103233: Warn: Regular expression argument: metric_octopus_export unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate), now will disable
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.8.1
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[12:55:09] INFO: Configuring timezone (Europe/London)...
Watching ['./prediction.py', './hass.py', './apps.yaml', './config.py', './download.py', './predbat.py', './utils.py', './ha.py', './unit_test.py', './inverter.py'] for changes
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-08-21 12:55:19.341223: Warn: No solar data has been configured.
2024-08-21 12:55:19.691554: Error: Exception raised 'Battery_Charge_Rate'
2024-08-21 12:55:19.698649: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9800, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9311, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 746, in update_status
    self.charge_rate_now = self.rest_data["Control"]["Battery_Charge_Rate"] / MINUTE_WATT
KeyError: 'Battery_Charge_Rate'

[... cut ...]
  File "/config/predbat.py", line 10962, in update_time_loop
    raise e
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9800, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9311, in fetch_inverter_data
    inverter.update_status(self.minutes_now)
  File "/config/inverter.py", line 746, in update_status
    self.charge_rate_now = self.rest_data["Control"]["Battery_Charge_Rate"] / MINUTE_WATT
KeyError: 'Battery_Charge_Rate'`

gcoan, 2024-08-21:
I've looked at the code:

```
        if self.rest_data:
            self.charge_enable_time = self.rest_data["Control"]["Enable_Charge_Schedule"] == "enable"
            self.discharge_enable_time = self.rest_data["Control"]["Enable_Discharge_Schedule"] == "enable"
            self.charge_rate_now = self.rest_data["Control"]["Battery_Charge_Rate"] / MINUTE_WATT
            self.discharge_rate_now = self.rest_data["Control"]["Battery_Discharge_Rate"] / MINUTE_WATT
```

its the next to last line it is failing on, its using REST to communicate with GivTCP, not using the Home Assistant controls (so the changes you made earlier to apps.yaml would have had no benefit), and ti appears that GivTCP isn't providing a battery_charge_rate.

Continue pursuing the GivTCP change via brikkat, but in the meantime, comment out the rest lines in apps.yaml and try using the home assistant controls instead.

I did notice from the full list of warnings that predbat says you don't have the octopus integration setup nor the solcast.  Have you configured your import and export rates manually in apps.yaml (in which case you can comment the octopus energy regular expressions out of apps.yaml), and are you intending to call solcast directly from predbat (in which case you can comment out the solcast lines in apps.yaml).  
You will need import/export rates and a solar forecast for predbat to meaningfully work

LeeFarm11, 2024-08-21:
Thanks Geoff. 

I will comment out the REST section.

Yes I have configured import and export rates manually. I'm on Economy 10 with Outgoing Fixed.
Yes I do intend to use Solcast in Predbat, but haven't reached that point working through the instructions yet. Because of the ongoing development in GivTCP, I'm trying to go one step at a time.

gcoan, 2024-08-21:
That makes sense.  You can just comment out the regular expressions in apps.yaml that point to the octopus integration and solcast integration.  It'll just stop the warnings when you load Predbat

LeeFarm11, 2024-08-21:
I commented out the 2 lines in apps.yaml relating to REST.
`  # If using Docker then change homeassistant.local to the Docker IP address
  #givtcp_rest:
  #  - 'http://192.168.1.204:6345'
`
Restarted and I think it is looking better in the log file. I will continue with the instructions to add Solcast and dashboards. I'll leave the Solcast and Octopus setting alone for now.

I tried the Open Web UI and I only get error message saying connection to 192.168.1.204:5052 cannot be established. Is that indicating a problem?

LeeFarm11, 2024-08-21:
Am I doing something wrong with the code markers?

gcoan, 2024-08-21:
> I tried the Open Web UI and I only get error message saying connection to 192.168.1.204:5052 cannot be established. Is that indicating a problem?

I think that's part of GivTCP isn't it? Am sure I saw something earlier about it not working.

I use Power Flow Plus HACS front end component, works very well

LeeFarm11, 2024-08-21:
No this is Predbat -
[image]

gcoan, 2024-08-21:
Ah ok. That should show the predbat log file. Never heard of it not working. 

Maybe try restarting the add on?

LeeFarm11, 2024-08-21:
No luck with the Web UI. 
But now Solcast is set up using the built in 'version', so no Solcast Addon.
Here is one of the 5 minute loops. It has Solcast prediction, but now showing some issue with **discharge_rate** rather than charge_rate......
Any ideas? Thanks.

2024-08-21 18:10:00.487262: --------------- PredBat - update at 2024-08-21 18:10:00+01:00 with clock skew 0 minutes, minutes now 1090
2024-08-21 18:10:00.488149: Saved current settings to /config/predbat_config.json
2024-08-21 18:10:00.521658: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 2.0 minutes
2024-08-21 18:10:00.521764: Predbat /config/predbat.py version v8.3.5 currently running, latest version is v8.3.5 latest beta v8.3.5
2024-08-21 18:10:00.521834: Autoupdate: Currently up to date
2024-08-21 18:10:00.603090: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-08-21 18:10:00.605770: Predbat mode is set to Monitor
2024-08-21 18:10:01.478035: Found 5761 load_today datapoints going back 4 days
2024-08-21 18:10:03.093761: Car charging hold True threshold 6.0
2024-08-21 18:10:03.093875: Current data so far today: load 24.5 kWh import 0.0 kWh export 43.4 kWh pv 71.5 kWh
2024-08-21 18:10:03.093918: Fetching futurerate data from None
2024-08-21 18:10:03.095226: Adding rate import: {'start': '01:00:00', 'end': '06:00:00', 'rate': 11.84} => 08-21 01:00:00 to 08-21 06:00:00 @ 11.84 date None increment False
2024-08-21 18:10:03.096389: Adding rate import: {'start': '06:00:00', 'end': '14:00:00', 'rate': 26.55} => 08-21 06:00:00 to 08-21 14:00:00 @ 26.55 date None increment False
2024-08-21 18:10:03.097982: Adding rate import: {'start': '14:00:00', 'end': '17:00:00', 'rate': 11.84} => 08-21 14:00:00 to 08-21 17:00:00 @ 11.84 date None increment False
2024-08-21 18:10:03.098788: Adding rate import: {'start': '17:00:00', 'end': '21:00:00', 'rate': 26.55} => 08-21 17:00:00 to 08-21 21:00:00 @ 26.55 date None increment False
2024-08-21 18:10:03.099703: Adding rate import: {'start': '21:00:00', 'end': '23:00:00', 'rate': 11.84} => 08-21 21:00:00 to 08-21 23:00:00 @ 11.84 date None increment False
2024-08-21 18:10:03.100318: Adding rate import: {'start': '23:00:00', 'end': '01:00:00', 'rate': 26.55} => 08-21 23:00:00 to 08-21 01:00:00 @ 26.55 date None increment False
2024-08-21 18:10:03.101854: Adding rate export: {'rate': 15} => 08-21 00:00:00 to 08-21 00:00:00 @ 15.0 date None increment False
2024-08-21 18:10:03.105706: Standing charge is set to 53.66 p
2024-08-21 18:10:05.860096: Rate min forward looking: now 11.84 at end of forecast 11.84
2024-08-21 18:10:05.860609: Import rates min 11.84 max 26.55 average 20.42
2024-08-21 18:10:05.923665: Export rates min 15.0 max 15.0 average 15.0
2024-08-21 18:10:05.923793: Rate thresholds (for charge/discharge) are import 26.05p (0.0) export 15.0p (0.0)
2024-08-21 18:10:05.947928: High export rate found rates in range 15.0 to 15.0
2024-08-21 18:10:05.961947: Low Import rate found rates in range 11.84 to 11.84
2024-08-21 18:10:06.086544: Today's energy import 0.0 kWh export 43.4 kWh cost -597.34 p import 53.66 p export -651.0 p carbon 0.0 kg
2024-08-21 18:10:06.086676: Obtaining solar forecast from Solcast API
2024-08-21 18:10:06.090728: Return cached data for https://api.solcast.com.au/rooftop_sites age 8.5 minutes
2024-08-21 18:10:06.090834: Fetch data for resource id e13e-97a0-2e0b-490e
2024-08-21 18:10:06.093947: Return cached data for https://api.solcast.com.au/rooftop_sites/e13e-97a0-2e0b-490e/forecasts age 8.4 minutes
2024-08-21 18:10:06.213960: Solcast returned 337 data points
2024-08-21 18:10:06.249616: PV Forecast for today is 1.53 (1.2 10% 1.91 90%) kWh and left today is 0.56 (0.31 10% 0.89 90%) kWh
2024-08-21 18:10:06.277684: PV Forecast for day tomorrow is 35.28 (18.18 10% 52.03 90%) kWh
2024-08-21 18:10:06.308038: PV Forecast for day d2 is 87.66 (46.31 10% 99.98 90%) kWh
2024-08-21 18:10:06.339205: PV Forecast for day d3 is 51.03 (2.85 10% 102.35 90%) kWh
2024-08-21 18:10:06.371800: PV Forecast for day d4 is 96.78 (41.67 10% 103.86 90%) kWh
2024-08-21 18:10:06.402248: PV Forecast for day d5 is 74.93 (19.36 10% 102.14 90%) kWh
2024-08-21 18:10:06.431553: PV Forecast for day d6 is 89.39 (27.93 10% 99.73 90%) kWh
2024-08-21 18:10:06.656343: Historical data totals for days [1, 2, 3] are [29.86, 33.63, 31.24] - min 29.86
2024-08-21 18:10:06.656441: Model filter enabled - Discarding day 1 as it is the lowest of the 3 datapoints
2024-08-21 18:10:06.657495: Historical days now [2, 3] weight [0.75, 0.5]
2024-08-21 18:10:06.753198: Today's load divergence 111.34 % in-day adjustment 110.78 % damping 0.95x
2024-08-21 18:10:06.753284: Today's predicted so far 19.96 kWh with 0.0 kWh car/iBoost excluded and 0 kWh import ignored and 0 forecast extra.
2024-08-21 18:10:06.753329: Today's actual load so far 23.4 kWh with 1.0 kWh Car/iBoost excluded and 0 kWh import ignored.
2024-08-21 18:10:07.085292: Inverter 0: Type GE GivEnergy
2024-08-21 18:10:07.085495: Inverter 0 has timed pause support enabled
2024-08-21 18:10:07.086083: Invertor time 2024-08-21 17:08:12+00:00, Predbat computer time 2024-08-21 18:10:07.085786+01:00, difference -1.93 minutes
2024-08-21 18:10:07.086196: Reserve min: 4.0% Battery_min:4.0%
2024-08-21 18:10:07.086293: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 11000.0 w charge rate 11.0 kW discharge rate 11.0 kW battery_rate_min 200.0 w ac limit 11.5 kW export limit 11. [... cut ...]
2024-08-21 18:11:13.562691: Info: record_status Warn: Inverter 0 write to discharge_rate failed
2024-08-21 18:11:13.634788: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-08-21 18:11:13.783328: Completed run status Idle with Errors reported (check log)
2024-08-21 18:11:13.826923: Saved current settings to /config/predbat_config.json
2024-08-21 18:15:01.081481: --------------- PredBat - update at 2024-08-21 18:15:00+01:00 with clock skew 0 minutes, minutes now 1095

LeeFarm11, 2024-08-21:
And this line says Battery is 10kWh. But it is really 20.4kWh.
Maybe that is another mis-read from GivTCP? But I don't see any relevant Entities or Helpers from GivTCP.

2024-08-21 18:10:07.086293: Inverter 0 with soc_max 10.0 kWh nominal_capacity 10.0 kWh battery rate raw 11000.0 w charge rate 11.0 kW discharge rate 11.0 kW battery_rate_min 200.0 w ac limit 11.5 kW export limit 11.5 kW reserve 4.0 % current_reserve 4.0 %

gcoan, 2024-08-21:
Hi @LeeFarm11 some progress then, good

Based on what you shared earlier, the discharge rate was set correctly in apps.yaml to point to number.givtcp_SERIALNO_battery_discharge_rate but this is currently set to zero and then predbat errors when trying to set it to max discharge rate (11000W).

I also noticed the current SoC isn't being reported correctly:

2024-08-21 18:10:07.088260: Found 1 inverters totals: min reserve 0.4 current reserve 0.4 soc_max 10.0 soc 0.0 charge rate 11.0 kW discharge rate 0.2 kW battery_rate_min 200.0 w ac limit 11.5 export limit 11.5 kW loss charge 3 % loss discharge 3 % inverter loss 4 %

Coming up as zero, should be more than that.

I'm afraid these are questions for GivTCP.  Must be something wrong in the mapping of modbus registers to home assistant controls.

LeeFarm11, 2024-08-21:
Thanks @gcoan 

Will post in https://github.com/britkat1980/giv_tcp/issues/186

crg-n, 2024-08-27:
Hi, I am having the same issue. It looks like you are making progress with the 3 phase issues in [GivTCP issue 186](https://github.com/britkat1980/giv_tcp/issues/186) but I suspect the missing "Battery_Charge_Rate" remains unaddressed in the latest versions of GivTCP and so I have opened issue [#215 ](https://github.com/britkat1980/giv_tcp/issues/215).

I believe it is solely this problem that is upsetting Predbat for me as if I do a nasty workaround and manually set the "Battery_Charge_Rate" in inverter.py, PredBat is able to run without any other errors with GivTCP.

britkat1980, 2024-08-27:
This is an error in Dev Version of GivTCP, fixed in 716 (to be tested). I think it only happened when charge rate is set to Zero. Install 716 and confirm fix

crg-n, 2024-08-27:
Fix confirmed working on my end. Many thanks! This issue could likely be closed @LeeFarm11

LeeFarm11, 2024-08-27:
Hi @crg-n 

Now I'm running GivTCP Dev720 and latest update predbat. But I get this error (now relating battery size from REST). Did you make some other changes, or am I missing something?
Thanks

Startup
Predbat files are installed correctly for version v8.3.5
**** Starting Standalone Predbat ****
2024-08-27 22:41:53.830337: Loading apps.yaml
2024-08-27 22:41:54.094625: Info: Connected to Home Assistant at http://supervisor/core
2024-08-27 22:41:54.103079: Warn: Failed to decode response <Response [404]> from http://supervisor/core/addons/self/info
2024-08-27 22:41:54.104206: Creating task: <coroutine object HAInterface.socketLoop at 0x7f94624ff5a0>
2024-08-27 22:41:54.104932: Info: Web Socket task started
2024-08-27 22:41:54.106154: Warn: unable to find /config/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-08-27 22:41:54.109437: Info: Start socket for url http://supervisor/core/api/websocket
2024-08-27 22:41:54.138099: Info: Web Socket active
2024-08-27 22:41:54.242338: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
2024-08-27 22:41:54.247954: Warn: Regular expression argument: pv_forecast_today unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_today), now will disable
2024-08-27 22:41:54.253020: Warn: Regular expression argument: pv_forecast_tomorrow unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow), now will disable
2024-08-27 22:41:54.258875: Warn: Regular expression argument: pv_forecast_d3 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3)), now will disable
2024-08-27 22:41:54.264114: Warn: Regular expression argument: pv_forecast_d4 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4)), now will disable
2024-08-27 22:41:54.268942: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-08-27 22:41:54.273898: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-08-27 22:41:54.278364: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-08-27 22:41:54.282279: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
2024-08-27 22:41:54.286566: Warn: Regular expression argument: metric_octopus_import unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate), now will disable
2024-08-27 22:41:54.290690: Warn: Regular expression argument: metric_octopus_export unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate), now will disable
Watching ['./prediction.py', './hass.py', './apps.yaml', './config.py', './download.py', './predbat.py', './utils.py', './ha.py', './unit_test.py', './inverter.py'] for changes
2024-08-27 22:42:03.744582: Warn: Historical day 2 has 5 minutes of gap in the data, filled from 16.05 kWh to make new average 16.11 kWh (percent 100%)
**2024-08-27 22:42:04.024602: Error: Reported battery size from REST is 0.0, but it must be >0**
2024-08-27 22:42:04.024665: Error: Exception raised 
2024-08-27 22:42:04.032517: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9800, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9310, in fetch_inverter_data
    inverter = Inverter(self, id)
  File "/config/inverter.py", line 281, in __init__
    raise ValueError
ValueError

2024-08-27 22:42:04.072690: Info: record_status Error: Exception raised 
2024-08-27 22:42:04.072787: Error: 
Traceback (most recent call last):
  File "/config/hass.py", line 173, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 10962, in update_time_loop
    raise e
  File "/config/predbat.py", line 10956, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9800, in update_pred
    self.fetch_inverter_data()
  File "/config/predbat.py", line 9310, in fetch_inverter_data
    inverter = Inverter(self, id)
  File "/config/inverter.py", line 281, in __init__
    raise ValueError
ValueError

LeeFarm11, 2024-08-27:
Ah. @gcoan  and @britkat1980 
I see crg-n has AIO, not 3 phase, so maybe not the same issues.
But I do still have the REST section commented out, so why is it searching for REST information?

apps.yaml -
  #
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  #givtcp_rest:
  #  - 'http://192.168.1.204:6345'

crg-n, 2024-08-27:
Hi Lee. No, my apologies - I was referring to the "Battery_Charge_Rate" exception from your title/first post, which BritKat has now fixed. It looks like some of your 3 phase issues are persisting but hopefully you'll get those sorted in the GivTCP issue. I am on an AIO and it's all working through REST for me now. I think gcoan previously got you to disable REST as a workaround for the "Battery_Charge_Rate" being missing so now that it is fixed maybe you could try REST on again?

This probably won't help your issue but one thing I found that might be useful to you in the future was connecting to the REST api directly to see what values it was making available. I believe this is what predbat calls so you can use this to figure out if the issue is with Predbat or GivTCP. For me it is at: http://homeassistant.local:6345/readData

Good luck!

LeeFarm11, 2024-08-27:
Thanks for the guidance. I will try to take a look.
But I have REST commented out......... so why do I have REST errors?

[image]

gcoan, 2024-08-27:
Hi @LeeFarm11 

Yes I did advise to try commenting out the apps.yaml rest configuration lines and to use the givtcp home assistant sensors instead, to see if that fixed your original issue.

I don't know why you are getting that error when you've commented REST out, but it's hard to verify apps.yaml with all the weird formatting.
What's happening is that github is parsing the apps.yaml content and trying to format it as HTML!

To stop this, precede the apps.yaml text with three backwards quotes "```" on a line on their own,  then put the same after the apps.yaml fragment.  This will cause github to treat the pasted file section as being code and not interpret it.

You could try REST again, but equally the sensors should work.  I think you shared them before, but have a look at the sensor defined as the charge rate in apps.yaml, does it have a zero value as predbat complains?   (I know the error refers to REST, it may be that the error message is misleading!)

Looking at your logfile I am slightly confused by the error

2024-08-27 22:41:54.103079: Warn: Failed to decode response <Response [404]> from http://supervisor/core/addons/self/info

as this should have been fixed in 8.3.5.  Are you running predbat under docker or as the predbat add-on?
It doesn't impact on this issue but I am surprised to see it in your log

britkat1980, 2024-08-27:
I believe that predbat uses rest for reading data and that the rest config line you commented out was for control…?

gcoan, 2024-08-27:
> I believe that predbat uses rest for reading data and that the rest config line you commented out was for control…?

Maybe that Predbat does.  Learn something new every day !

Thanks

LeeFarm11, 2024-08-28:
I took a look at the REST page. http://homeassistant.local:6345/readData I think this might be the relevant section causing the battery size issue -

"TD2343G049": { **"Battery_Capacity_kWh": 0.0, "Battery_Capacity_kWh_calc": 21.28**, "Battery_Priority": "Load", "Battery_Type": "Lithium", "Boost_Temperature": 42.7, "Buck_Boost_Temperature": 43.1, "DC_Status": "Normal", "Inverter_Temperature": 46.2, "Invertor_Firmware": "ZAAA040404", "Invertor_Max_Bat_Rate": 11000, "Invertor_Max_Inv_Rate": 11000, "Invertor_Serial_Number": "TD2343G049", "Invertor_Software": "ZA1.0", "Invertor_Time": "2024-08-28T07:30:53+01:00", "Invertor_Type": "Gen 3 - Hybrid_3ph", "Power_Factor": 0, "Start_Delay_Time": 3, "System_Mode": "Grid_tied", "status": "Normal" }

So is this predbat reading the inverter, not related to GivTCP?

LeeFarm11, 2024-08-28:
For info here is the entire record -

[REST.txt](https://github.com/user-attachments/files/16777439/REST.txt)

gcoan, 2024-08-28:
> I took a look at the REST page. http://homeassistant.local:6345/readData I think this might be the relevant section causing the battery size issue -
>
> "TD2343G049": { **"Battery_Capacity_kWh": 0.0, [...]

I believe the sequence is something like:

1. GivTCP interrogates the modbus registers on the inverter and presents the information to MQTT
2. MQTT presents the queued data as a REST interface
3. MQTT also populates Home Assistant entities with the data queued via REST

4. Updating a HA GivTCP entity calls a REST interface to update MQTT
5. Calling a MQTT REST interface causes GivTCP to update the inverter modbus registers

Predbat reads the inverter data via 1 and then 2

Predbat can write inverter data via 4 and then 5, or 5 directly.  When you comment out the rest interface in apps.yaml, Predbat is doing 4 which then calls 5.

I should make this more clear in the documentation, I do need to add the instructions on installing MQTT etc so I'll include it in there

So the problem with the missing data via REST is back to GivTCP

For comparison here's the start of the REST data for my inverter/batteries (Gen 1 hybrid)

{ "Battery_Details": { "DX2327G139": { "Battery_Capacity": 182.13, "Battery_Cell_10_Voltage": 3.351, "Battery_Cell_11_Voltage": 3.349, "Battery_Cell_12_Voltage": 3.349, "Battery_Cell_13_Voltage": 3.361, "Battery_Cell_14_Voltage": 3.351, "Battery_Cell_15_Voltage": 3.353, "Battery_Cell_16_Voltage": 3.351, "Battery_Cell_1_Temperature": 22.2, "Battery_Cell_1_Voltage": 3.349, "Battery_Cell_2_Temperature": 22.8, "Battery_Cell_2_Voltage": 3.345, "Battery_Cell_3_Temperature": 23.2, "Battery_Cell_3_Voltage": 3.343, "Battery_Cell_4_Temperature": 21.4, "Battery_Cell_4_Voltage": 3.346, "Battery_Cell_5_Voltage": 3.358, "Battery_Cell_6_Voltage": 3.348, "Battery_Cell_7_Voltage": 3.349, "Battery_Cell_8_Voltage": 3.361, "Battery_Cell_9_Voltage": 3.348, "Battery_Cells": 16, "Battery_Cycles": 208, "Battery_Design_Capacity": 186.0, "Battery_Firmware_Version": 3015, "Battery_Remaining_Capacity": 63.72, "Battery_SOC": 35, "Battery_Serial_Number": "DX2327G139", "Battery_Temperature": 27.2, "Battery_USB_present": true, "Battery_Voltage": 53.608 } }, "Control": { "Active_Power_Rate": 100, "Battery_Charge_Rate": 2600, "Battery_Discharge_Rate": 2600, "Battery_Power_Cutoff": 4, "Battery_Power_Mode": "enable", "Battery_Power_Reserve": 4, "Battery_pause_mode": "Disabled", "Enable_Charge_Schedule": "enable", "Enable_Discharge": "enable", "Enable_Discharge_Schedule": "disable", "Force_Charge": "Normal", "Force_Export": "Normal", "Local_control_mode": "Load", "Mode": "Eco", "PV_input_mode": "Independent", "Reboot_Addon": "disable", "Reboot_Invertor": "disable", "Target_SOC": 98, "Temp_Pause_Charge": "Normal", "Temp_Pause_Discharge": "Normal" }, "Energy": { "Rates": { "Battery_Value": 0.319, "Battery_ppkwh": 0.096, "Current_Rate": 0.279, "Current_Rate_Type": "Day", "Day_Cost": 0.0, "Day_Energy_Total_kWh": 0, "Day_Energy_kWh": 0.0, "Day_Rate": 0.279, "Day_Start_Energy_kWh": 13483.9, "Export_Rate": 0.169, "Import_ppkwh_Today": 0.225, "Night_Cost": 0.0, "Night_Energy_Total_kWh": 0, "Night_Energy_kWh": 0.0, "Night_Rate": 0.168, "Night_Start_Energy_kWh": 13483.9 }, "Today": { "AC_Charge_Energy_Today_kWh": 0.4,

britkat1980, 2024-08-28:
I think the problem is the three phase battery capacity has a different name (_calc) suffix. I’ll align names now that were agreed the right calculation!

gcoan, 2024-09-20:
is this resolved in GivTCP now?

gcoan, 2024-09-20:
Ok thanks for the update @LeeFarm11 

Keep this open for any residual predbat issues 

Cheers

closed 2025-01-01 by springfall2008