springfall2008/batpred #1381: Huawei charge - stop

JohanAlvedal, 2024-08-21:
**Describe the bug**
My inverter that stops when it reaches the target (%) which is quite narrow ie is reached quickly. it might be better to set a time on the time slot 30 min instead of %. 

**Expected behavior**
Instead of charging Force charge in % when the target is reached too quickly. Force charge in time (30min) instead of %.
not sure if it's enough though but might be worth a try.

action: huawei_solar.forcible_charge
data:
  duration: 30
  power: "3500"
  device_id: 6893ee838a0972bd3e50d9e0faa0ba5a

**Predbat version**
8.3.5

**Environment details**
- Huawei Luna200 10kwh battery
- Standard HAOS installer

**Screenshots**
[image]

**Log file**
[https://justpaste.it/cege1](url)

springfall2008, 2024-08-22:
Good point, although 30 minute maybe too long?

springfall2008, 2025-04-19:
Did you ever find a solution here?

JohanAlvedal, 2025-04-19:
My solution to the whole thing with Huawei Luna 2000. Is to handle it with automation in combination with automations and scripts, packages file etc. with a special one in the config file. see here:

https://github.com/JohanAlvedal/Homeautomation/tree/main/Predbat

closed 2025-08-21 by JohanAlvedal