springfall2008/batpred #1384: "Cost" keeps carrying previous slots cost over to next slot. Total also behaving strangely.

grey-area001, 2024-08-22:
**Describe the bug**

Twice after a negative pricing and heavy load period I've had a situation in which the Predbat Plan shows a decrease in price over a slot. When the plan ticks to the next slot the decrease is carried over and either increased or decreased but doesn't go away entirely until (I think) the next day.

**Expected behavior**
The cost and running total operate as normal

**Predbat version**

Present on 8.3.5

**Environment details**

GivEnergy gen 3 3.5kWh inverter and 9.5kw battery 
Standard HAOS installer

**Screenshots**

Previous incident (no logs available)

[image]

Current incident (with logs)

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/16714497/predbat.log)

springfall2008, 2024-08-24:
Looking at the table IBOOST total goes down and EXPORT goes up, meaning somehow Predbat things IBOOST is running in reverse - bonkers.
Will check...

2024-08-22 16:52:17.069712 INFO pred_bat:  EXPORT: [     6.9,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58,    11.58]
2024-08-22 16:52:17.074193 INFO pred_bat:  IBOOST: [   13.01,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,        0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      0.0,      1.2,      1.2,      2.4,      2.4,      3.6,      4.4,      5.6,      5.6,      6.8,      6.8,      6.8,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,      8.0,        0,      0.0,      0.0,      0.0,      0.0,      0.0]
2024-08-22 16:52:17.079269 INFO pred_bat:  METRIC: [ -145.12,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -215.35,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.04,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.07,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -163.08,  -110.77,  -110.56,  -108.68,  -106.99,   -105.8]

springfall2008, 2024-08-24:
Fix on main for testing

grey-area001, 2024-08-25:
This was occurring again today. I installed the main branch and the issue seems resolved.

[predbat.log](https://github.com/user-attachments/files/16739698/predbat.log)

springfall2008, 2024-08-25:
Release as v8.3.7

closed 2024-08-25 by springfall2008