springfall2008/batpred #1385: Stuck on Loading apps.yaml

letsstartbeinganon, 2024-08-23:
**Describe the bug**
I'm trying to set up Batpred with my Fox inverter. At some point a change I made in the config means Batpred isn't loading and just says 'Loading apps.yaml' in the config file. Earlier in the process, before I'd entered (what I thought was correctly) all the entity IDs etc into apps.yaml, it appears to have been working.

**Expected behavior**
I expected it to start and begin its calculations.

**Predbat version**

Latest version

**Environment details**

Fox ESS, here is my apps.yaml

https://pastebin.com/eJ2Ym5C6

I am getting an error message that says

bad indentation of a mapping entry (257:5)

 254 |   # Auto matches Zappi and Wallbox, ...
 255 |   # One entry per car
 256 |   # car_charging_planned:
 257 |     - 're:(sensor.wallbox_portal_st ...
-----------^
 258 |  

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**

Log file below showing Predbat seeming to work, and then endless 'Loading apps.yaml'

https://pastebin.com/JG85xxfV

letsstartbeinganon, 2024-08-23:
I have managed to fix this and get Predbat running by just removing the lines relating to car charging (apart from saying that I don't have cars)

But now I am getting the following warnings:

changed to Warn: can not resolve watch_list value {octopus_charge_limit}
09:11:27 - 3 minutes ago - Supervisor
changed to Warn: can not resolve watch_list value {octopus_ready_time}
09:11:27 - 3 minutes ago - Supervisor

letsstartbeinganon, 2024-08-23:
This was fixed by adding a space after the # in the lines relating to Octopus intelligent. I'm not on Octopus Intelligent so I commented them out (put a hash in front of them).

gcoan, 2024-08-23:
> 256 | # car_charging_planned:
> 257 | - 're:(sensor.wallbox_portal_st ...

The first error is you commented out the sensor name (car charging planned) but didn't comment out the regular expression that the sensor is set to,  so the 're' line would be matched with whatever is previous in your apps.yaml which is why you got an intentation value error

Maybe the same with Octopus intelligent, you didn't comment everything out properly?

If you can share your apps.yaml we can take a look at it now

closed 2026-08-20 by chalfontchubby