springfall2008/batpred #1391: Appdaemon Predbet error after today's update

btxluke, 2024-08-26:
**Describe the bug**
Predbat will not start after today's update. 

HA + appdaemon&predbat  preinstall addon

Rebuild add-on gives message:

Failed to rebuild add-on

The command '/bin/bash -o pipefail -c apk add --no-cache --virtual .build-dependencies build-base=0.5-r3 python3-dev=3.11.9-r0 && echo "Step1"' returned a non-zero code: 2

**Predbat version**

8.3.6 | 8.3.7

**Environment details**

- givenergy inverter & battery
- Standard HAOS install

**Log file**
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/config.py Got data, writing to /config/apps/config.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/prediction.py Got data, writing to /config/apps/prediction.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/utils.py Got data, writing to /config/apps/utils.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/inverter.py Got data, writing to /config/apps/inverter.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/ha.py Got data, writing to /config/apps/ha.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/download.py Got data, writing to /config/apps/download.py.v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.6/apps/predbat/unit_test.py Got data, writing to /config/apps/unit_test.py.v8.3.6 Update complete Predbat files are installed correctly for version v8.3.6 s6-rc: info: service legacy-services: stopping s6-rc: info: service legacy-services successfully stopped s6-rc: info: service appdaemon: stopping [19:56:12] INFO: Service AppDaemon exited with code 0 (by signal 0) s6-rc: info: service appdaemon successfully stopped s6-rc: info: service init-appdaemon: stopping s6-rc: info: service init-appdaemon successfully stopped s6-rc: info: service legacy-cont-init: stopping s6-rc: info: service legacy-cont-init successfully stopped s6-rc: info: service fix-attrs: stopping s6-rc: info: service base-addon-log-level: stopping s6-rc: info: service base-addon-log-level successfully stopped s6-rc: info: service fix-attrs successfully stopped s6-rc: info: service base-addon-banner: stopping s6-rc: info: service base-addon-banner successfully stopped s6-rc: info: service s6rc-oneshot-runner: stopping s6-rc: info: service s6rc-oneshot-runner successfully stopped s6-rc: info: service s6rc-oneshot-runner: starting s6-rc: info: service s6rc-oneshot-runner successfully started s6-rc: info: service base-addon-banner: starting ----------------------------------------------------------- Add-on: appdaemon-predbat Predbat pre-install in AppDaemon ----------------------------------------------------------- Add-on version: 1.0.11 You are running the latest version of this add-on. System: Home Assistant OS 13.1 (aarch64 / raspberrypi4-64) Home Assistant Core: 2024.8.2 Home Assistant Supervisor: 2024.08.0 ----------------------------------------------------------- Please, share the above information when looking for help or support in, e.g., GitHub, forums or the Discord chat. ----------------------------------------------------------- s6-rc: info: service base-addon-banner successfully started s6-rc: info: service fix-attrs: starting s6-rc: info: service base-addon-log-level: starting s6-rc: info: service fix-attrs successfully started s6-rc: info: service base-addon-log-level successfully started s6-rc: info: service legacy-cont-init: starting s6-rc: info: service legacy-cont-init successfully started s6-rc: info: service init-appdaemon: starting s6-rc: info: service init-appdaemon successfully started s6-rc: info: service appdaemon: starting s6-rc: info: service appdaemon successfully started s6-rc: info: service legacy-services: starting [19:57:30] INFO: Starting AppDaemon... s6-rc: info: service legacy-services successfully started Predbat files are installed correctly for version v8.3.6 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/predbat.py Got data, writing to /config/apps/predbat.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/config.py Got data, writing to /config/apps/config.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/prediction.py Got data, writing to /config/apps/prediction.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/utils.py Got data, writing to /config/apps/utils.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/inverter.py Got data, writing to /config/apps/inverter.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/ha.py Got data, writing to /config/apps/ha.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/download.py Got data, writing to /config/apps/download.py.v8.3.7 Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.3.7/apps/predbat/unit_test.py Got data, writing to /config/apps/unit_test.py.v8.3.7 Update complete Predbat files are installed correctly for version v8.3.7 Predbat files are installed correctly for version v8.3.7 s6-rc: info: service legacy-services: stopping s6-rc: info: service legacy-services successfully stopped s6-rc: info: service appdaemon: stopping [15:43:27] INFO: Service AppDaemon exited with code 0 (by signal 0) s6-rc: info: service appdaemon successfully stopped s6-rc: info: service init-appdaemon: stopping s6-rc: info: service init-appdaemon successfully stopped s6-rc: info: service legacy-cont-init: stopping s6-rc: info: service legacy-cont-init successfully stopped s6-rc: info: service fix-attrs: stopping s6-rc: info: service base-addon-log-level: stopping s6-rc: info: service fix-attrs successfully stopped s6-rc: info: service base-addon-log-level successfully stopped s6-rc: info: service base-addon-banner: stopping s6-rc: info: service base-addon-banner successfully stopped s6-rc: info: service s6rc-oneshot-runner: stopping s6-rc: info: service s6rc-oneshot-runner successfully stopped

springfall2008, 2024-08-27:
Can you retry installing the latest addon?

SzosszeNET, 2024-08-27:
Getting the same issue trying to install the latest version

[image]

gcoan, 2024-08-27:
Suggest as @springfall2008 says, reinstalling the add-on.

Firstly with your file editor, open the apps.yaml file you configured, copy the contents and paste it somewhere safe.

Then stop and uninstall the appdaemon-predbat add-on

Just to ensure everything is clean, shutdown and restart Home Assistant (and all running add-on's)

Install appdaemon-predbat add-on

You'll probably need to start the add-on to ensure that everything gets downloaded.  You'll get an error that you have a template apps.yaml file.

Open the template apps.yaml file that was installed with the new copy of appdaemon-predbat, select all and paste in the original apps.yaml that you copied away earlier

Restart appdaemon-predbat add-on and hopefully all will be running clean

btxluke, 2024-08-28:
I didn't manage to get this working - and I've just reverted to the seperate addons - documentation here: 

https://springfall2008.github.io/batpred/install/#predbat-installation-into-appdaemon

as @gcoan said - backup your apps.yaml - it just worked copied and pasted over the /homeassistant/appdaemon/apps/batpred/config/apps.yaml

SzosszeNET, 2024-08-28:
@gcoan I get the above error when I'm trying to install the add-on.
Have uninstalled the plugin, deleted appdaemon.
Deleted the appdaemon files from "addon_configs", deleted all appdaemon folders from the config folder
restarted HA
And still fails to install the addon with the screengrab I shared

SzosszeNET, 2024-08-28:
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 19:22:25 (1 occurrences)
Last logged: 19:22:25

Failed to to call /addons/46f69597_appdaemon-predbat/install - The command '/bin/bash -o pipefail -c apk add --no-cache --virtual .build-dependencies build-base=0.5-r3 python3-dev=3.11.9-r0 && echo "Step1"' returned a non-zero code: 2

SzosszeNET, 2024-08-28:
Okay - it's actually a dependency problem on the latest HA?

~ # apk add python3-dev=3.11.9-r0
ERROR: unable to select packages:
  python3-3.12.3-r1:
    breaks: python3-dev-3.12.3-r2[python3=3.12.3-r2]
    satisfies: world[python3=3.12.3-r1] py3-pip-24.0-r2[python3] py3-pip-24.0-r2[python3~3.12] py3-setuptools-69.5.1-r0[python3]
               py3-setuptools-69.5.1-r0[python3~3.12] py3-parsing-3.1.2-r1[python3] py3-parsing-3.1.2-r1[python3~3.12]
               py3-packaging-24.0-r1[python3] py3-packaging-24.0-r1[python3~3.12] py3-awake-1.0-r10[python3]
               py3-awake-1.0-r10[python3~3.12]
  python3-dev-3.12.3-r2:
    breaks: world[python3-dev=3.11.9-r0]

gcoan, 2024-08-28:
Sorry all.

Can you try with the predbat add on rather than the app daemon -predbat add on?

Functionally it's absolutely the same just doesn't use appdaemon as the 'wrapper' to predbat. 

Predbat runs the same, just copy the same apps.yaml file in.

Predbat add on is the preferred install method now. See the install documentation for details

SzosszeNET, 2024-08-28:
Yup - that works. News to me, I believe the documentation doesn't mention that option

springfall2008, 2024-08-30:
Closing as appdeamon-predbat is no longer supported and the addon works for this user

closed 2024-08-30 by springfall2008

btxluke, 2024-08-30:
Thanks @springfall2008 - I think thats the best solution. Could you update the documentation to mention the new preferred install method? Plus the web dashboard is cool!

I'm a little snowed under at the moment, but in the following months I'll be happy to contribute some CSS work for the dashboard, if that can be helpful?